warpem/warp #3: Installation question for HPC computing nodes

rdrighetto, 2024-04-18:
Hello Alister and Dimitry,

First of all, thanks a lot for this long-awaited release! Can't wait to start Warp'in on Linux!

I'm a bit confused by the installation instructions:

1. Do I understand it correctly that if I install from conda-forge I don't need to run the build scripts?

2. On our cluster, the login (head) node has internet access but no GPU (hence no NVIDIA driver). Conversely, the GPU computing nodes don't have access to the internet. So I have to install it from the login node, but how to ensure that it will fetch the correct CUDA stuff in this case?

Thank you!

dtegunov, 2024-04-19:
Hi Ricardo,

The way conda-forge will only give you a CUDA-enabled PyTorch if you run it on a GPU system or inside Nvidia's container is really annoying. The easiest thing to do is to wait for the conda-forge package. If you're feeling more adventurous, you can try to get PyTorch from the pytorch channel instead, with the environment file looking something like this:

```
name: warp_build
channels:
  - pytorch
  - nvidia
  - conda-forge
  - nodefaults
dependencies:
  - torchvision
  - libtiff
  - python
  - pytorch-cuda=11.8
  - pytorch>2
  - fftw
  - cmake
  - cxx-compiler=1.3
  - dotnet=8.0
```

alisterburt, 2024-04-19:
@rdrighetto did this get you up and running?

rdrighetto, 2024-04-20:
Hi!
Thanks for the quick reply @dtegunov and @alisterburt!
I just gave it a try with the template `warp_build.yml` you suggested above, creating the conda environment works fine but upon running `./build-native-linux.sh` I get a couple errors, see log here:
https://drive.switch.ch/index.php/s/ihxHT5aqnhtyC8r

This is running under
`Linux login12 5.15.0-91-generic #101-Ubuntu SMP Tue Nov 14 13:30:08 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux`

Please let me know if you need any other information to help debugging!

ps: I'm traveling until Monday evening so might be slow to reply until then
Thanks again!

cdienem, 2024-04-20:
Hi @rdrighetto 

one of the first errors indicates that your CUDA is not found properly (`fatal error: cuda_runtime.h: No such file or directory`). I had something similar at the first try and providing the following environment variables made it work:
```
CUDA_ROOT=/opt/cuda/cuda-11.8

# Add CUDA to path
export PATH=$CUDA_ROOT/bin:$PATH

# Add CUDA libraries to CPATH
export CPATH=$CUDA_ROOT/include:$CPATH

# Add CUDA libraries to LD_LIBRARY_PATH
export LD_LIBRARY_PATH=$CUDA_ROOT/lib64:$LD_LIBRARY_PATH
```

If you check your environment with `env`before installing to make sure these are correct. Hopefully this will also fix your downstream issues.

Chris

rdrighetto, 2024-04-20:
Hi @cdienem!
Thanks for the tip and sorry for not being more attentive to the error messages at first.
Indeed, loading the `CUDA/11.8.0` module (and only this module) provided by our HPC sets these environment variables and solves these first errors.
However, when it comes to compiling `LibTorchSharp` I then run into many other errors which boil down to:
`#error You need C++17 to compile PyTorch`

Any hints are appreciated!

dtegunov, 2024-04-20:
Was that a 214 MB log file 😳
The new error sounds like maybe the C++ compiler installed in the conda environment gets overriden by something else on that system (e.g. when you load the CUDA module). What is the output from CMake if you just run the configuration part without compiling:
```
cd LibTorchSharp
mkdir build
cd build
cmake -DCMAKE_PREFIX_PATH=`python -c 'import torch;print(torch.utils.cmake_prefix_path)'` ..
```

Alternative: The conda-forge pytorch Nvidia driver shenanigans only apply to PyTorch >= 2. You could use this environment.yml:
```
name: warp_build
channels:
  - conda-forge
  - nodefaults
dependencies:
  - torchvision
  - libtiff
  - python
  - pytorch=1.13.1
  - pytorch-gpu
  - fftw
  - cmake
  - cxx-compiler=1.3
  - dotnet=8.0
```
And modify line 19 in LibTorchSharp/LibTorchSharp/THSAutograd.cpp like this:
`c10::AutogradState::set_tls_state(c10::AutogradState(!enabled, enabled, !enabled));`

dtegunov, 2024-04-20:
Actually, let's remove the need for an external CUDA module entirely:
```
name: warp_build
channels:
  - conda-forge
  - nodefaults
dependencies:
  - torchvision
  - libtiff
  - python
  - pytorch=1.13.1
  - pytorch-gpu
  - fftw
  - cmake
  - cxx-compiler=1.3
  - dotnet=8.0
  - cudatoolkit=11.4
  - cudatoolkit-dev
  - cudnn
```
This still requires you to modify THSAutograd.cpp as described in my previous reply.
Also make sure to remove the build directories in NativeAcceleration and LibTorchSharp after unsuccessful build attempts.

rdrighetto, 2024-04-20:
> Was that a 214 MB log file 😳

sorry 😬 

Summarizing the issue so far:
I tried a fresh install following https://github.com/warpem/warp/issues/3#issuecomment-2067701153 strictly with the [first environment suggestion](https://github.com/warpem/warp/issues/3#issuecomment-2065895443) to avoid that loading our CUDA module would put something weird in the way, but the `#error You need C++17 to compile PyTorch` persisted.

Then I tried another clean install following the [latest environment suggestion](https://github.com/warpem/warp/issues/3#issuecomment-2067740795)  but then the environment can't even be created due to the conflicts listed below.

Let's leave this aside for now, I can wait for the official conda package. Thanks a lot for your help guys, really appreciated!

```
Channels:
 - conda-forge
Platform: linux-64
Collecting package metadata (repodata.json): ...working... done
Solving environment: ...working... failed
Channels:
 - conda-forge
Platform: linux-64
Collecting package metadata (repodata.json): ...working... done
Solving environment: ...working... failed

LibMambaUnsatisfiableError: Encountered problems while solving:
  - nothing provides __cuda needed by pytorch-1.13.1-cuda112py310he33e0d6_200

Could not solve for environment specs
The following packages are incompatible
├─ cudatoolkit 11.4**  is requested and can be installed;
├─ pytorch-gpu is installable with the potential options
│  ├─ pytorch-gpu [1.10.0|1.10.1|...|1.9.1] would require
│  │  └─ pytorch [1.10.0 cuda102py37h689c94d_1|1.10.0 cuda102py37h98b7ee3_0|...|1.9.1 cuda102py39h2fcd037_3], which requires
│  │     └─ cudatoolkit 10.2|10.2.* , which conflicts with any installable versions previously reported;
│  ├─ pytorch-gpu [1.10.0|1.10.1|...|1.9.1] would require
│  │  └─ pytorch [1.10.0 cuda110py37h4a33b93_0|1.10.0 cuda110py37h7b7832c_1|...|1.9.1 cuda110py39h5cf7045_3], which requires
│  │     └─ cudatoolkit 11.0|11.0.* , which conflicts with any installable versions previously reported;
│  ├─ pytorch-gpu [1.10.0|1.10.1|...|1.9.1] would require
│  │  └─ pytorch [1.10.0 cuda111py37h07fa5b8_1|1.10.0 cuda111py37he371307_0|...|1.9.1 cuda111py39hb4a4491_3], which requires
│  │     └─ cudatoolkit 11.1|11.1.* , which conflicts with any installable versions previously reported;
│  ├─ pytorch-gpu [1.10.0|1.10.1|...|1.9.1] would require
│  │  └─ pytorch [1.10.0 cuda112py37h3bec1eb_0|1.10.0 cuda112py37haf94430_1|...|1.9.1 cuda112py37h3bec1eb_3], which requires
│  │     ├─ python >=3.7,<3.8.0a0  with the potential options
│  │     │  ├─ python [3.7.0|3.7.1|3.7.2|3.7.3|3.7.6] would require
│  │     │  │  └─ python_abi * *_cp37m, which can be installed;
│  │     │  ├─ python [3.7.10|3.7.12|3.7.9], which can be installed;
│  │     │  └─ python [3.7.10|3.7.12|...|3.7.9] would require
│  │     │     └─ python_abi 3.7.* *_cp37m, which can be installed;
│  │     └─ python_abi 3.7.* *_cp37m, which can be installed;
│  ├─ pytorch-gpu [1.10.0|1.10.1|...|1.9.1] would require
│  │  └─ pytorch [1.10.0 cuda112py38h4f2a933_0|1.10.0 cuda112py38had345c2_1|...|1.9.1 cuda112py38h4f2a933_3], which requires
│  │     ├─ python >=3.8,<3.9.0a0  but there are no viable options
│  │     │  ├─ python [3.8.0|3.8.1] would require
│  │     │  │  └─ python_abi * *_cp38, which conflicts with any installable versions previously reported;
│  │     │  ├─ python [3.8.10|3.8.12|...|3.8.8] would require
│  │     │  │  └─ python_abi 3.8.* *_cp38, which conflicts with any installable versions previously reported;
│  │     │  ├─ python [3.8.12|3.8.13|3.8.16|3.8.5] conflicts with any installable versions previously reported;
│  │     │  └─ python 3.8.19 would require
│  │     │     ├─ ncurses >=6.4.20240210,<7.0a0 , which can be installed;
│  │     │     └─ python_abi 3.8.* *_cp38, which conflicts with any installable versions previously reported;
│  │     └─ python_abi 3.8.* *_cp38, which conflicts with any installable versions previously reported;
│  ├─ pytorch-gpu [1.10.0|1.10.1|...|1.9.1] would require
│  │  └─ pytorch [1.10.0 cuda112py39h3ad47f5_1|1.10.0 cuda112py39h4e14dd4_0|...|1.9.1 cuda112py39h4e14dd4_3], which requires
│  │     ├─ python >=3.9,<3.10.0a0  but there are no viable options
│  │     │  ├─ python [3.9.0|3.9.1|...|3.9.9] would require
│  │     │  │  └─ python_abi 3.9.* *_cp39, which conflicts with any installable versions previously reported;
│  │     │  ├─ python [3.9.10|3.9.12|3.9.16|3.9.17|3.9.18] conflicts with any installable versions previously reported;
│  │     │  └─ python 3.9.19 would require
│  │     │     ├─ ncurses >=6.4.20240210,<7.0a0 , which can be installed;
│  │     │     └─ python_abi 3.9.* *_cp39, which conflicts with any installable versions previously reported;
│  │     └─ python_abi 3.9.* *_cp39, which conflicts with any installable versions previously reported;
[... 79 lines cut ...]
   ├─ pytorch 1.13.1 would require
   │  └─ python >=3.11,<3.12.0a0  but there are no viable options
   │     ├─ python [3.11.0|3.11.1|...|3.11.8] would require
   │     │  └─ python_abi 3.11.* *_cp311, which conflicts with any installable versions previously reported;
   │     └─ python 3.11.9 would require
   │        ├─ ncurses >=6.4.20240210,<7.0a0 , which can be installed;
   │        └─ python_abi 3.11.* *_cp311, which conflicts with any installable versions previously reported;
   ├─ pytorch 1.13.1 would require
   │  ├─ python >=3.8,<3.9.0a0  but there are no viable options
[... cut ...]
   │  │  ├─ python [3.9.0|3.9.1|...|3.9.9], which cannot be installed (as previously explained);
   │  │  ├─ python [3.9.10|3.9.12|3.9.16|3.9.17|3.9.18] conflicts with any installable versions previously reported;
   │  │  └─ python 3.9.19, which cannot be installed (as previously explained);
   │  ├─ python_abi 3.9.* *_cp39, which conflicts with any installable versions previously reported;
   │  └─ pytorch-gpu 99999999* , which conflicts with any installable versions previously reported;
   └─ pytorch [1.11.0|1.12.0|...|2.1.2], which cannot be installed (as previously explained).

```

alisterburt, 2024-04-21:
I'm reopening as this is an important issue for us to get a handle on: people will need to create conda environments to run this in (or we provide that env in a container)

conda package is only a partial solution and this exploration is useful! Thanks @rdrighetto and @cdienem ☺️

dtegunov, 2024-04-21:
Sorry, I should have tested it. This works for me for a fresh build (after modifying THSAutograd.cpp):
```
name: warp_build
channels:
  - conda-forge
  - nodefaults
dependencies:
  - torchvision
  - libtiff
  - python
  - pytorch=1.11
  - pytorch-gpu
  - fftw
  - cmake
  - cxx-compiler=1.3
  - dotnet=8.0
  - cudatoolkit
  - cudatoolkit-dev
  - cudnn
```

cdienem, 2024-04-21:
With modiefied THSAutograd.cpp and `warp_build.yml` like this:
```
name: warp_build
channels:
  - conda-forge
  - nodefaults
dependencies:
  - torchvision
  - libtiff
  - python
  - pytorch=1.11
  - pytorch-gpu
  - fftw
  - cmake
  - cxx-compiler=1.3
  - dotnet=8.0
  - cudatoolkit
  - cudatoolkit-dev
  - cudnn
```
everythings builds ok.

However, I discovered that the `warp_build/pkgs/cuda-toolkit` folder is installed with very resticted permissions in my case (as I install everything as root), so when running WarpTools, it fails to load libcudart for NativeAcceleration.

```
Unhandled exception. System.DllNotFoundException: Unable to load shared library 'NativeAcceleration' or one of its dependencies. In order to help diagnose loading problems, consider using a tool like strace. If you're using glibc, consider setting the LD_DEBUG environment variable: 
/opt/warp/warp-2.0.0dev1/Release/linux-x64/publish/NativeAcceleration.so: cannot open shared object file: No such file or directory
libcudart.so.11.0: cannot open shared object file: No such file or directory
libNativeAcceleration.so: cannot open shared object file: No such file or directory
/opt/warp/warp-2.0.0dev1/Release/linux-x64/publish/NativeAcceleration: cannot open shared object file: No such file or directory
/opt/warp/warp-2.0.0dev1/Release/linux-x64/publish/libNativeAcceleration: cannot open shared object file: No such file or directory

   at Warp.GPU.GetDeviceCount()
   at WarpTools.Commands.DistributedOptions.GetWorkers(Boolean attachDebugger) in /opt/warp/warp-2.0.0dev1/WarpTools/Commands/DistributedOptions.cs:line 41
   at WarpTools.Commands.MotionCTFFrameseries.Run(Object options) in /opt/warp/warp-2.0.0dev1/WarpTools/Commands/Frameseries/MotionCTFFrameseries.cs:line 187
   at WarpTools.WarpTools.Run(Object options) in /opt/warp/warp-2.0.0dev1/WarpTools/Program.cs:line 30
   at Warp.Tools.CommandLineParserHelper.ParseAndRun(String[] args, Func`2 run, Type[] verbs, String appName) in /opt/warp/warp-2.0.0dev1/WarpLib/Tools/CommandLineParserHelper.cs:line 26
   at WarpTools.WarpTools.Main(String[] args) in /opt/warp/warp-2.0.0dev1/WarpTools/Program.cs:line 17
   at WarpTools.WarpTools.<Main>(String[] args)
./run_motion_ctf.sh: line 13: 16725 Aborted                 (core dumped) WarpTools fs_motion_and_ctf --settings frameseries.settings --m_grid 6x6x40 --c_grid 6x6x1 --c_range_max 5 --c_defocus_max 4 --out_averages --out_average_halves --perdevice 1
```

Allowing read access for non-root (755) to the `warp_build` environment fixes it. It was not only the access `warp_build/pkgs/cuda-toolkit` as 755 on only that folder gave other errors with CUFFT.

A brief test with `WarpTools` and `Noise2Mic` seems to work.

rdrighetto, 2024-04-23:
Hi there!

Current status of environment installation following https://github.com/warpem/warp/issues/3#issuecomment-2067907834 is:

```
(base) diogori@transfer12:warp$ mamba env create -f warp_build.yml
Channels:
 - conda-forge
Platform: linux-64
Collecting package metadata (repodata.json): done
Solving environment: done

Downloading and Extracting Packages:

Preparing transaction: done
Verifying transaction: done
Executing transaction: - By downloading and using the CUDA Toolkit conda packages, you accept the terms and conditions of the CUDA End User License Agreement (EULA): https://docs.nvidia.com/cuda/eula/index.html

\ By downloading and using the cuDNN conda packages, you accept the terms and conditions of the NVIDIA cuDNN EULA -
  https://docs.nvidia.com/deeplearning/cudnn/sla/index.html

done
ERROR conda.core.link:_execute(949): An error occurred while installing package 'conda-forge::cudatoolkit-dev-11.7.0-h1de0b5d_6'.
Rolling back transaction: done

LinkError: post-link script failed for package conda-forge::cudatoolkit-dev-11.7.0-h1de0b5d_6
location of failed script: /scicore/projects/scicore-p-structsoft/ubuntu/software/Miniconda3/miniconda3_python3.12.1/envs/warp_build/bin/.cudatoolkit-dev-post-link.sh
==> script messages <==
<None>
==> script output <==
stdout: Running Post installation
downloading https://developer.download.nvidia.com/compute/cuda/11.7.0/local_installers/cuda_11.7.0_515.43.04_linux.run to /scicore/projects/scicore-p-structsoft/ubuntu/software/Miniconda3/miniconda3_python3.12.1/envs/warp_build/pkgs/cudatoolkit-dev/cuda_11.7.0_515.43.04_linux.run
Extracting on Linux

stderr: terminate called after throwing an instance of 'boost::filesystem::filesystem_error'
  what():  boost::filesystem::copy_file: No such file or directory: "./builds/cuda_cupti/extras/CUPTI/include/cupti_callbacks.h", "/tmp/tmp1r0xdw8w/extras/CUPTI/include/cupti_callbacks.h"
Aborted (core dumped)
Traceback (most recent call last):
  File "/scicore/projects/scicore-p-structsoft/ubuntu/software/Miniconda3/miniconda3_python3.12.1/envs/warp_build/bin/cudatoolkit-dev-post-install.py", line 311, in <module>
    _main()
  File "/scicore/projects/scicore-p-structsoft/ubuntu/software/Miniconda3/miniconda3_python3.12.1/envs/warp_build/bin/cudatoolkit-dev-post-install.py", line 307, in _main
    extractor.extract()
  File "/scicore/projects/scicore-p-structsoft/ubuntu/software/Miniconda3/miniconda3_python3.12.1/envs/warp_build/bin/cudatoolkit-dev-post-install.py", line 135, in extract
    subprocess.run(cmd, env=os.environ.copy(), check=True)
  File "/scicore/projects/scicore-p-structsoft/ubuntu/software/Miniconda3/miniconda3_python3.12.1/envs/warp_build/lib/python3.10/subprocess.py", line 526, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['/scicore/projects/scicore-p-structsoft/ubuntu/software/Miniconda3/miniconda3_python3.12.1/envs/warp_build/pkgs/cudatoolkit-dev/cuda_11.7.0_515.43.04_linux.run', '--silent', '--toolkit', '--toolkitpath=/tmp/tmp1r0xdw8w', '--override']' returned non-zero exit status 134.

return code: 1

()

```

Upon inspecting `/tmp/cuda-installer.log` I see a bunch of permission-related warnings/errors.
See the log here (this time only 156 KB): https://drive.switch.ch/index.php/s/D7I459UPLsMRPJx
Am I supposed to run this as root? In my view, these conda packages for the CUDA stuff spilling out of the environment path is very problematic.

Related: in a previous attempt, I had errors because the `/tmp/cuda-installer.log` file already existed in the head node and belonged to somebody else (apparently from a cryoSPARC installation attempt), so I had to switch to a different login node where this file did not exist yet. I know this is not a Warp problem per se but as I said above, CUDA stuff can get very weird when it comes to installing via conda.

rdrighetto, 2024-04-23:
ps: even though I used `mamba` above the same things happen when using `conda`.

rdrighetto, 2024-04-23:
@cdienem thanks for the tips! In this case, the problem is not the conda environment itself (I installed this Miniconda instance myself and the environments are located in a path that I definitely have permissions to), but that conda packages like `cudatoolkit` and `cudatoolkit-dev` try to do stuff OUTSIDE the environment, namely in `/tmp/` and `/var/` when installing.

Having pinpointed these guys as the culprits, I managed to build it 🎉 using the following environment:

```
name: warp_build
channels:
  - conda-forge
  - nodefaults
dependencies:
  - torchvision
  - libtiff
  - python
  - pytorch=1.11
  - pytorch-gpu
  - fftw
  - cmake
  - cxx-compiler=1.3
  - dotnet=8.0
  # - cudatoolkit
  # - cudatoolkit-dev
  - cudnn
  ```
  
And since `cudatoolkit` and `cudatoolkit-dev` are commented out, I loaded our own `CUDA/11.8.0` module provided by our HPC team. I used the latest commit from the repo: https://github.com/warpem/warp/commit/d9b13c2c64bd98b0294e2d83a57f6d8d60274cdf

Haven't really tested it yet but `WarpTools --help` works so I call it a victory for today.

Thanks a lot for bearing with me!

alisterburt, 2024-04-23:
Glad you got it built! 🥳 thanks for reporting back

alisterburt, 2024-04-23:
Given that we've merged #17 and your later issues were permissions related I think we can close here - thanks again @rdrighetto !

closed 2024-04-23 by alisterburt