warpem/warp #7: Noise2Mic GPU memory consumption question

cdienem, 2024-04-21:
Hi,

when running 

```
Noise2Mic --processingdir warp
```
using the Relion tutorial data movies (already processed with `WarpTools fs_motion_and_ctf`), I quickly run into GPU memory issues when using `--batchsize` larger than 5 on a 1080 with 8 GB memory.

I agree, this is not the most modern card, but I wonder if there are additional ways to reduce memory consumption except for tuning the batch size (as a batch size of 5 yields a training time of 3+ days for 24 movies). 

The movies are K2 size, compressed tifs at 0.8 A/px and 49 frames.

alisterburt, 2024-04-21:
Hey! This seems excessively long to me but maybe my expectations are wrong - @dtegunov what would you expect on this card?

dtegunov, 2024-04-23:
Hey Chris,

- Have you tried lowering --iterations to get a more reasonable training time? Was the result ok?
- Denoising unbinned 0.8 A/px micrographs will make training unnecessarily hard (too much noise), and inference is going to take a while. I'd scale them down to something like 6 A/px.

cdienem, 2024-04-23:
Hi Dimitry,

I tried a few parameters and made the following observations:

- lowering `--window` to `128` or `256` dramatically decreased training time while still providing reasonable results
- fewer `--iterations` also reduced training times (obviously), I tested 50-100 and they all looked OK (as much as they can be OK with 24 movie examples...)

The more important observation is, that the time remaining that is shown is completely overestimated. 

When running `Noise2Mic --processingdir warp --batchsize 5 --rescale_angpix 6 --lowpass 6 --iterations 200 --window 128`, it starts with an estimation of ~4:00:00 hours, which quickly decreases in a linear manner to zero as denoising progresses (in steps of about 00:02:00). The overall time is then about 00:02:45 (HH:MM:SS). So something is not right there.

So the GPU usage does not seem to be an issue, its the time estimate.

The default parameters are probably meant for tomograms, whereas frame series might need a different set of params for iterations and window. The latter is absolutely fine, just requires documentation (I am happy to provide a few real world tests at some point to find a good set of recommendations).

Thanks for looking into it!

dtegunov, 2024-04-24:
The time estimate is fixed now.

While I was fixing it, I remembered my rationale for the defaults: 512 px window gives more context, which can be helpful sometimes; a batch size of 64 helps the BatchNorm layers because they benefit from larger batches. This combination fits nicely on an 80 GB A100, leaving only 1 GB free 😄 I agree most people won't be running it on an 80 GB card. However, it is unlikely that users with the necessary means would explore higher values for these parameters to improve results if I lower the defaults, whereas users with less VRAM will be forced to lower them anyway. So it's discoverability vs. inconvenience for different groups of users. Not sure I can enable both without complicating the application.

alisterburt, 2024-04-24:
Thanks for the write up @dtegunov - super useful. This is solved with documentation, I'll add it to the docs. We could also add a link to the docs from the help text, what do you think?

closed 2024-05-01 by dtegunov