warpem/warp #11: Suggestion: movie .xml file as text (not binary)

cdienem, 2024-04-22:
Problem: movie xml files are stored in binary format and are complicated to read/modify on unix commandline.

I have come across this already in Warp 1.x. I understand that there are reasons to store things as binary and not plain text. However, it is very hard to access these files on a linux commandline when they are stored as binaries, especially for inexperienced users. So maybe it would be worth switching to plain text if there is no good reason to keep the binary format.

alisterburt, 2024-04-22:
Interesting - I've never had problems but I always interact with these either through VSCode or Python which I guess handle this transparently.

It looks like these files are written from the `SaveMeta` method in unicode with an `XMLTextWriter` class
https://github.com/warpem/warp/blob/30fc4e8b4cbfe5f672cd0df9e65c864ba5a0f14a/WarpLib/Movie.cs#L961-L964

this is provided by `System.Xml`
https://learn.microsoft.com/en-us/dotnet/api/system.xml.xmltextwriter?view=net-8.0

We could change the encoding to Utf-8 easily but I don't have good intuition for how much will break if we do this
https://learn.microsoft.com/en-us/dotnet/api/system.text.encoding?view=net-8.0

I agree that the ability to easily edit these files is important

alisterburt, 2024-04-22:
@dtegunov any objections to going to UTF if unicode is causing @cdienem issues?

cdienem, 2024-04-22:
> Interesting - I've never had problems but I always interact with these either through VSCode or Python which I guess handle this transparently.

Most GUI text editors handle these files well, however, `nano`, `more`, `sed` etc. cannot handle them.

When opening them the simplest way in Python, the way I get the contents as a string is

```
f = open("settingsfile.settings", "rb")
data = f.read()
print(data.decode("utf-16")
```

XML parsers of the standard library, however, seem to handle this under the hood (e.g. `xml.etree.ElementTree.parse`).

alisterburt, 2024-04-22:
makes sense and given that the whole point of WarpTools is to enable users on HPC systems to be effective... we should find a solution

alisterburt, 2024-04-22:
Just had a chat with @dtegunov - he's going to test changing to UTF-8 and see what it breaks/how hard it would be to move to that. If it's easy, that's what we'll do

dtegunov, 2024-04-23:
We anticipated this would be a much harder decision when we thought we'd have to drop emoji support due to UTF-8. However, ChatGPT says that all emoji can be represented. With this major obstacle out of the way, here is 773bf9d.

closed 2024-04-23 by dtegunov

cdienem, 2024-04-23:
Great, thanks for trying and changing this! I like the suggestion of  `.💩` extensions... :laughing: