warpem/warp #13: Export micrograph list as .star

cdienem, 2024-04-22:
Apologies for all the issues I open here :smile: 

Problem: I cannot seem to find a way to export a micrograph list from `WarpTool` processing output.

I guess you have this on your list already. For streamlining and interfacing, it would be nice to be able to export micrograph lists as a star file. This is especially useful as Warp2 does not write CTFFIND-style log-files that could be used to quickly assemble a star file. Additionally, the current output from motion correction (in the xml files) cannot be used for polishing in Relion.

alisterburt, 2024-04-22:
Don't apologise - this is all incredibly useful feedback, keep it coming! 🙂 

I agree this would be ideal. 

Are there multiple 'modes' you want this tool to run in? if so could you provide a list of these modes?

I will be able to implement much more quickly if for each mode you could provide an example of the minimal file contents you require with relevant metadata labels.

alisterburt, 2024-04-23:
@dtegunov we agreed this would be good to have yesterday.

We provide an interface to RELION with particle stacks and we provide polishing tools that outperform it in M. 

Given that, is it a priority to add this now or should we park it for later?

alisterburt, 2024-04-24:
After discussion: this is a priority as it enables the use of Warp for preprocessing and RELION for polishing and has potential to drive user adoption in SPA