warpem/warp #16: task: finish conda-forge recipe so users can install from conda

alisterburt, 2024-04-22:


alisterburt, 2024-04-23:
pipeline trying to build against current version of main d9b13c2

https://dev.azure.com/conda-forge/feedstock-builds/_build/results?buildId=920573&view=results

alisterburt, 2024-04-23:
That build is failing.

When I try to build an environment locally with the conda_build.yml in the repo at d9b13c2

```yaml
name: warp_build
channels:
  - conda-forge
  - nodefaults
dependencies:
  - torchvision
  - libtiff
  - python
  - pytorch=1.11
  - pytorch-gpu
  - fftw
  - cmake
  - cxx-compiler=1.3
  - dotnet=8.0
  - cudatoolkit
  - cudatoolkit-dev
  - cudnn
```

It resolves as follows:
```txt
# packages in environment at /home/burta2/mambaforge/envs/warp_build:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                  2_kmp_llvm    conda-forge
binutils                  2.39                 hdd6e379_1    conda-forge
binutils_impl_linux-64    2.39                 he00db2b_1    conda-forge
binutils_linux-64         2.39                h5fc0e48_13    conda-forge
brotli-python             1.1.0           py310hc6cd4ac_1    conda-forge
bzip2                     1.0.8                hd590300_5    conda-forge
c-ares                    1.28.1               hd590300_0    conda-forge
c-compiler                1.3.0                h7f98852_0    conda-forge
ca-certificates           2024.2.2             hbcca054_0    conda-forge
certifi                   2024.2.2           pyhd8ed1ab_0    conda-forge
cffi                      1.16.0          py310h2fee648_0    conda-forge
charset-normalizer        3.3.2              pyhd8ed1ab_0    conda-forge
cmake                     3.29.2               hcfe8598_0    conda-forge
cuda-version              11.8                 h70ddcb2_3    conda-forge
cudatoolkit               11.8.0              h4ba93d1_13    conda-forge
cudatoolkit-dev           11.7.0               h1de0b5d_6    conda-forge
cudnn                     8.9.7.29             hbc23b4c_3    conda-forge
cxx-compiler              1.3.0                h4bd325d_0    conda-forge
dotnet                    8.0.204              ha770c72_0    conda-forge
dotnet-aspnetcore         8.0.4                hb8a3ed7_0    conda-forge
dotnet-runtime            8.0.4                hb8a3ed7_0    conda-forge
dotnet-sdk                8.0.204              hb8a3ed7_0    conda-forge
fftw                      3.3.10          nompi_hc118613_108    conda-forge
freetype                  2.12.1               h267a509_2    conda-forge
gcc                       9.5.0               h1fea6ba_13    conda-forge
gcc_impl_linux-64         9.5.0               h99780fb_19    conda-forge
gcc_linux-64              9.5.0               h4258300_13    conda-forge
gxx                       9.5.0               h1fea6ba_13    conda-forge
gxx_impl_linux-64         9.5.0               h99780fb_19    conda-forge
gxx_linux-64              9.5.0               h43f449f_13    conda-forge
icu                       73.2                 h59595ed_0    conda-forge
idna                      3.7                pyhd8ed1ab_0    conda-forge
kernel-headers_linux-64   2.6.32              he073ed8_17    conda-forge
keyutils                  1.6.1                h166bdaf_0    conda-forge
krb5                      1.21.2               h659d440_0    conda-forge
lcms2                     2.16                 hb7c19ff_0    conda-forge
ld_impl_linux-64          2.39                 hcc3a1bd_1    conda-forge
lerc                      4.0.0                h27087fc_0    conda-forge
libblas                   3.9.0            16_linux64_mkl    conda-forge
libcblas                  3.9.0            16_linux64_mkl    conda-forge
[... 188 lines cut ...]
2024-04-23T14:33:11.7259684Z │  │  │  ├─ cudnn [8.8.0.121|8.9.1.23|8.9.7.29], which cannot be installed (as previously explained);
2024-04-23T14:33:11.7260671Z │  │  │  ├─ cudnn [8.8.0.121|8.9.1.23|8.9.7.29], which can be installed (as previously explained);
2024-04-23T14:33:11.7261201Z │  │  │  ├─ cudnn 8.9.7.29, which can be installed;
2024-04-23T14:33:11.7261858Z │  │  │  └─ cudnn 8.9.7.29, which can be installed (as previously explained);
2024-04-23T14:33:11.7262508Z │  │  └─ pytorch [2.0 cuda112*|2.0.* cuda112*], which cannot be installed (as previously explained);
2024-04-23T14:33:11.7263264Z │  ├─ torchvision 0.15.2 would require
2024-04-23T14:33:11.7263752Z │  │  └─ pytorch >=2.0.0,<2.1.0a0  but there are no viable options
2024-04-23T14:33:11.7264509Z │  │     ├─ pytorch [1.10.0|1.10.1|...|2.1.2], which cannot be installed (as previously explained);
2024-04-23T14:33:11.7265215Z │  │     ├─ pytorch [1.11.0|1.12.0|...|2.1.2], which cannot be installed (as previously explained);
2024-04-23T14:33:11.7265737Z │  │     ├─ pytorch [1.13.0|1.13.1|2.0.0|2.1.0|2.1.2], which cannot be installed (as previously explained);
2024-04-23T14:33:11.7266424Z │  │     ├─ pytorch 2.0.0 conflicts with any installable versions previously reported;
2024-04-23T14:33:11.7266916Z │  │     ├─ pytorch [2.0.0|2.1.0|2.1.2], which cannot be installed (as previously explained);
2024-04-23T14:33:11.7267443Z │  │     ├─ pytorch [1.12.0|1.12.1|1.13.0|1.13.1|2.0.0], which cannot be installed (as previously explained);
2024-04-23T14:33:11.7267957Z │  │     ├─ pytorch [1.13.0|1.13.1|2.0.0], which cannot be installed (as previously explained);
2024-04-23T14:33:11.7268651Z │  │     ├─ pytorch [2.0.0|2.1.0|2.1.2], which cannot be installed (as previously explained);
2024-04-23T14:33:11.7269419Z │  │     ├─ pytorch [1.12.0|1.12.1|1.13.0|1.13.1|2.0.0], which cannot be installed (as previously explained);
[... cut ...] not be installed (as previously explained);
2024-04-23T14:33:11.7270677Z │  │     ├─ pytorch [2.0.0|2.1.0|2.1.2], which cannot be installed (as previously explained);
2024-04-23T14:33:11.7271279Z │  │     ├─ pytorch [1.12.0|1.12.1|1.13.0|1.13.1|2.0.0], which cannot be installed (as previously explained);
2024-04-23T14:33:11.7271841Z │  │     └─ pytorch [2.0.0|2.1.0|2.1.2], which cannot be installed (as previously explained);
```

There seems to be some ABI incompatibility.

Any ideas @dtegunov ?

alisterburt, 2024-04-25:
After discussing with @dtegunov 
1. we're blocked on conda-forge packaging as mentioned above and are looking into contracting with Quansight for support
2. we're going to look at building/deploying ourselves with github actions and our own channel

Option 1 is preferred as we can leverage their build infrastructure and get guaranteed compatibility with their standard libraries. Option 2 is our clearest path to a simple installation for our users.

alisterburt, 2024-04-25:
Investigating option 2:

relevant conda-forge docs: https://conda.io/projects/conda/en/latest/user-guide/tasks/create-custom-channels.html

A channel seems to be a local or web hosted folder structure containing tarballs. The tarballs can be built with conda-build by you in your own environment.

I haven't yet tried to run conda-build in our infrastructure but setting it up amounts to:
- installing `conda-build`
- specifying a recipe as we have done on conda-forge
- running conda-build 

We can be more flexible with specifying our requirements in a package if we don't have to adhere to conda-forge's rules, this should work

dtegunov, 2024-05-02:
Since we're hosting the package under our own channel now, I think we can drop the conda-forge path. Just need a GitHub pipeline to build it non-locally.

alisterburt, 2024-05-10:
closed by #58 - not conda-forge but conda with our own warpem channel as the CUDA provided by conda-forge proved tricky to work with

closed 2024-05-10 by alisterburt