warpem/warp #8: Which program allows to prerpare training examples for BoxNet?

cdienem, 2024-04-22:
Hi,

in WarpTools, a BoxNet can be trained. However, I could not find from the code, which tool provides the `boxnet_examples_frameseries` functionality mentioned here

https://github.com/warpem/warp/blob/e450ab9a6d46c59523eef04e1ca228b449fe7adc/WarpTools/Commands/Frameseries/BoxNetTrainFrameseries.cs#L26

to prepare the examples. Is this a feature that is still coming in the future?

Thanks,
Chris

alisterburt, 2024-04-22:
Great question @cdienem !

These are generated internally by the Warp desktop GUI when preparing to train BoxNet here
https://github.com/warpem/warp/blob/e450ab9a6d46c59523eef04e1ca228b449fe7adc/Warp/Controls/BoxNet/BoxNetExport.xaml.cs#L153

There is no existing WarpTool for preparing these examples and the structure of this folder is not explicitly written down anywhere.

I don't know that we will provide a GUI tool for annotating the data in the near term. We definitely should provide this in the long term and we should definitely document the structure of the training data so that others can prepare their own training data and use this WarpTool.

If we can provide the structure of the data required here, does this solve your issue? 🙂

cdienem, 2024-04-22:
Hi @alisterburt
yes, documentation of the folder structure and data types would be helpful. No rush, you probably still have a big chunk of documentation to write for more important parts of the tool :smile:

alisterburt, 2024-04-22:
Perfect, I've added an issue to track this so will close here - thanks for your input!

closed 2024-04-22 by alisterburt