warpem/warp #19: Error when Warp compiled on linux

Angeevillor, 2024-04-23:
**Hello, I am trying to compile Warp on a linux(Ubuntu 22.04) system based on cuda11.8. After I configured and activated the conda environment, I ran the "build-native unix.sh" file and got the following error:**

[100%] Linking CXX shared library lib/libNativeAcceleration.so
/home/xxx/.conda/envs/warp_build/bin/../lib/gcc/x86_64-conda-linux-gnu/9.5.0/../../../../x86_64-conda-linux-gnu/bin/ld: cannot find -lfftw3_mpi: No such file or directory
/home/xxx/.conda/envs/warp_build/bin/../lib/gcc/x86_64-conda-linux-gnu/9.5.0/../../../../x86_64-conda-linux-gnu/bin/ld: cannot find -lfftw3f_mpi: No such file or directory
/home/xxx/.conda/envs/warp_build/bin/../lib/gcc/x86_64-conda-linux-gnu/9.5.0/../../../../x86_64-conda-linux-gnu/bin/ld: cannot find -lfftw3l_mpi: No such file or directory
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/NativeAcceleration.dir/build.make:2797: lib/libNativeAcceleration.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/NativeAcceleration.dir/all] Error 2
make: *** [Makefile:91: all] Error 2

**It seems that the absence of the "fftw_mpi" library in my conda environment is causing make to fail. I noticed that the previous conda environment seemed to have fftw installed, but there was no mpi version. How do I install this library in the conda environment?**

Thank you so much!

alisterburt, 2024-04-23:
Hi @Angeevillor - I don't immediately know how to fix your problem but your `$LD_LIBRARY_PATH` might be useful for debugging

Mine, in the environment where everything ran successfully, contains these entries

```txt
/lsf/10.1/linux3.10-glibc2.17-x86_64/lib:/lsf/10.1/linux3.10-glibc2.17-x86_64/lib
/home/burta2/software/warptools-dev/warp/NativeAcceleration/build/lib
/home/burta2/mambaforge/envs/warp_build/lib64
/home/burta2/software/warptools-dev/warp/NativeAcceleration/build/lib
/home/burta2/mambaforge/envs/warp_build/lib64
/home/burta2/software/warptools-dev/warp/NativeAcceleration/build/lib
/home/burta2/mambaforge/envs/warp_build/lib64
```

alisterburt, 2024-04-26:
closing this issue as not enough information provided to help and things are still evolving here, please feel free to reopen if you come back with more information!