warpem/warp #22: CUFFT_INTERNAL_ERROR while running WarpTools

Conecsion, 2024-04-24:
Hi I compiled the program and ran the test data with commands:
`WarpTools fs_motion_and_ctf --settings warp_frameseries.settings --m_grid 1x1x3 --c_grid 2x2x1 --c_range_max 7 --c_defocus_max 8 --c_use_sum --out_averages --out_average_halves --perdevice 4`

But I got errors:
> No alternative input specified, will use input parameters from warp_frameseries.settings
File search will be relative to /data/Linux_warp_test/frames/
328 files found
Parsing previous results for each item, if available...
328/328, previous metadata found for 8                                                                                                                 
Connecting to workers...
Connected to 8 workers
0/328terminate called after throwing an instance of 'std::runtime_error'
  what():  cuFFT error: CUFFT_INTERNAL_ERROR at /home/username/Programs/warp/NativeAcceleration/gtom/src/FFT/FFT.cu:23

I' using Ubuntu 20.04 and have cuda 11.8 installed

$PATH:

> /home/username/Programs/warp/Release/linux-x64/publish:/usr/local/cuda-11.8/bin:/home/username/Programs/anaconda3/condabin

$LD_LIBRARY_PATH

> /home/username/Programs/warp/NativeAcceleration/build/lib:/home/username/Programs/anaconda3/envs/warp_build/lib:/usr/local/cuda-11.8/lib64::/usr/local/lib:/usr/local/lib

alisterburt, 2024-04-24:
Hi @Conecsion ! Thanks for the clear report

There are some log files generated by the workers in <processing_dir>/logs (might not be exactly correct

Could you look at the errors in there? 

It might be a simple out of memory error. Warp algorithms were generally written with 16GB cards in mind and this test script was written for cards which have ~80GB  memory. If you don't have similarly large cards this is the likely cause of your errors

alisterburt, 2024-04-24:
--perdevice is the parameter to change if you need to reduce the memory consumption generally for warp tools

Conecsion, 2024-04-24:
Thanks for your hint.
I built a docker image [https://hub.docker.com/r/conecsion/warp](url)
And it runs smoothly for now, with commands 
`sudo docker run --rm -it --gpus all -v .:/share -w /share conecsion/warp bash`

alisterburt, 2024-04-25:
Excellent, I'll make sure this gets documented properly alongside the rest of WarpTools for future users - thanks!

Re docker: great! Docker is not ideal as a platform for us as lots of our users can't install it as they don't have root in the HPC environments they work in - this is one of the reasons we are targeting conda/conda-forge

We might also explore building a singularity container at some point but this is not a short term priority

closed 2024-04-25 by alisterburt

mpm896, 2024-07-03:
Hi @Conecsion , can I ask how you overcame this issue by making a new Docker image? I'm currently trying to run warptools on an AWS instance within a Docker container, and I'm getting the same error as you while trying to run `fs_motion_and_ctf`. I'm wondering if there are some settings I need to change, but I'm not very familiar with Docker. Thanks!!