warpem/warp #27: Can't complile on linux

thorstenwagner, 2024-04-26:
Hi,

warp is compiling for a while but after it reached 100% I get:
```
-- Detecting CXX compile features - done
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Error at LibTorchSharp/CMakeLists.txt:7 (find_package):
  Could not find a package configuration file provided by "Torch" with any of
  the following names:

    TorchConfig.cmake
    torch-config.cmake

  Add the installation prefix of "Torch" to CMAKE_PREFIX_PATH or set
  "Torch_DIR" to a directory containing one of the above files.  If "Torch"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
```

The file its looking for is here:
```
/mnt/beegfs/software/em/automatic_installs/miniconda3_envs/warp_build/lib/python3.10/site-packages/torch/share/cmake/Torch/TorchConfig.cmake

```

I think the problem is that this is not working:
```
python -c 'import torch;print(torch.utils.cmake_prefix_path)'
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/mnt/beegfs/software/em/automatic_installs/miniconda3_envs/warp_build/lib/python3.10/site-packages/torch/__init__.py", line 199, in <module>
    from torch._C import *  # noqa: F403
ImportError: /cm/local/apps/gcc/7.2.0/lib64/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /mnt/beegfs/software/em/automatic_installs/miniconda3_envs/warp_build/lib/python3.10/site-packages/torch/lib/libtorch_python.so)
```

Might be related to how you install pytorch?
https://discuss.pytorch.org/t/glibcxx-3-4-30-not-found/196875/9

alisterburt, 2024-04-26:
Hi @thorstenwagner - sorry you ran into problems and thanks for reporting here

It looks like something about your local setup means that your local system libraries are found rather than the conda-forge ones against which PyTorch was built. I don't know immediately how to solve this, but maybe adding the path to the lib/lib64 directories in your conda environment to $LD_LIBRARY_PATH would help?

alisterburt, 2024-04-26:
Is it the cuda-version requirement you added above? Yes please, a PR would be great, I'll test in our local infrastructure and merge if it works for us too 🙂

thorstenwagner, 2024-04-26:
Similar to the one above. Btw, which relion version do I need?

alisterburt, 2024-04-26:
testing has been done against relion-devel/main which is the private repository containing the development version of RELION 5. I'm not sure how up to date the public relion/5.0-beta branch is with respect to this

alisterburt, 2024-04-26:
closing this as solved

closed 2024-04-26 by alisterburt