warpem/warp #28: Can't create worker

thorstenwagner, 2024-04-26:
When I run the EMPIAR script, i see the following output:
```
Running command create_settings with:
output = warp_frameseries.settings
folder_processing = warp_frameseries
folder_data = /home/sphire-devel/warptest/frames
recursive = False
extension = *.tif
angpix = 0.7894
bin = null
bin_angpix = null
gain_path = gain_ref.mrc
defects_path = null
gain_flip_x = False
gain_flip_y = True
gain_transpose = False
exposure = 2.64
eer_ngroups = 40
tomo_dimensions = null

Settings saved to /mnt/beegfs/home/sphire-devel/warptest/warp_frameseries.settings
Now trying to find the files referenced by these settings...
328 files found
Running command fs_motion_and_ctf with:
m_range_min = 500
m_range_max = 10
m_bfac = -500
m_grid = 1x1x3
c_window = 512
c_range_min = 30
c_range_max = 7
c_defocus_min = 0.5
c_defocus_max = 8
c_voltage = 300
c_cs = 2.7
c_amplitude = 0.07
c_fit_phase = False
c_use_sum = True
c_grid = 2x2x1
out_averages = True
out_average_halves = True
out_skip_first = 0
out_skip_last = 0
device_list = {  }
perdevice = 4
workers = {  }
settings = warp_frameseries.settings
input_data = {  }
input_data_recursive = False
input_processing = null
output_processing = null

No alternative input specified, will use input parameters from warp_frameseries.settings
File search will be relative to /home/sphire-devel/warptest/frames
328 files found
Parsing previous results for each item, if available...
328/328, previous metadata found for 0                                                                                                                                                                                                    
Connecting to workers...
```

After a while, connecting to workers fails:
```
Unhandled exception. System.Exception: Couldn't connect to newly created worker at localhost:45851 for 20 seconds, something must be wrong
   at Warp.WorkerWrapper..ctor(Int32 deviceID, Boolean silent, Boolean attachDebugger) in /mnt/beegfs/software/em/warp/2.0beta/src/warp/WarpLib/WorkerWrapper.cs:line 104
   at WarpTools.Commands.DistributedOptions.GetWorkers(Boolean attachDebugger) in /mnt/beegfs/software/em/warp/2.0beta/src/warp/WarpTools/Commands/DistributedOptions.cs:line 57
   at WarpTools.Commands.MotionCTFFrameseries.Run(Object options) in /mnt/beegfs/software/em/warp/2.0beta/src/warp/WarpTools/Commands/Frameseries/MotionCTFFrameseries.cs:line 187
   at WarpTools.WarpTools.Run(Object options) in /mnt/beegfs/software/em/warp/2.0beta/src/warp/WarpTools/Program.cs:line 30
   at Warp.Tools.CommandLineParserHelper.ParseAndRun(String[] args, Func`2 run, Type[] verbs, String appName) in /mnt/beegfs/software/em/warp/2.0beta/src/warp/WarpLib/Tools/CommandLineParserHelper.cs:line 26
   at WarpTools.WarpTools.Main(String[] args) in /mnt/beegfs/software/em/warp/2.0beta/src/warp/WarpTools/Program.cs:line 17
   at WarpTools.WarpTools.<Main>(String[] args)

```

Any recommendation what to do?

alisterburt, 2024-04-26:
huh - the workers are subprocesses launched which actually do the computational work, they write logs to `<processing_dir>/logs` - is there any useful info in that folder?

If the workers are not spawning correctly they may not have even had time to write logs - in that case I'd have to defer to @dtegunov

shahpnmlab, 2024-04-30:
I see this also. in my testing so far it fails on the first run. After rerunning the command things get back into action. This happened when working with binaries compiled from the main branch. Tried it again with torch-2 and things are running as expected.

thorstenwagner, 2024-04-30:
I'm using this torch-2 branch and it works now :-) Can be closed!

alisterburt, 2024-04-30:
🥳

alisterburt, 2024-04-30:
actually, going to leave open until that branch is merged if that fixes things and something wasn't working before!

Did anything show up in the logs for either of you @shahpnmlab @thorstenwagner ?

thorstenwagner, 2024-04-30:
The logs were not even created. No log dir.

alisterburt, 2024-04-30:
okay good to know: workers are dying before outputting anything

shahpnmlab, 2024-04-30:
I can confirm Thorsten's observations.

alisterburt, 2024-04-30:
@shahpnmlab you're having problems right now? with no solutions?

alisterburt, 2024-04-30:
I see the torch 2 branch is merged and you indicated above that things were working as expected on that branch?

I'm running tests on our infrastructure and am not having problems with this

shahpnmlab, 2024-04-30:
> @shahpnmlab you're having problems right now? with no way to make things work?

What I meant was that things are working with the torch-2 branch.  In fact the binaries compiled with this updated code were spiffier than the older version... (but I am not sure if this was because of the lateness of the hour and the cluster wasnt being hammered... :P )

alisterburt, 2024-04-30:
@shahpnmlab cool! Closing here if things are working as that branch has now been merged - feel free to reopen if still having issues

shahpnmlab, 2024-05-01:
Reopening this because it happened again. I am not certain what triggers this behaviour... so far it looks a bit random.
```
WarpTools fs_motion_and_ctf --settings previous.settings --m_grid 1x1x5 --c_grid 2x2x1 --c_range_max 7 --c_defocus_max 8 --c_use_sum --out_averages --out_average_halves  --perdevice 4 --c_amplitude 0.1 --c_window 768
Running command fs_motion_and_ctf with:
m_range_min = 500
m_range_max = 10
m_bfac = -500
m_grid = 1x1x5
c_window = 768
c_range_min = 30
c_range_max = 7
c_defocus_min = 0.5
c_defocus_max = 8
c_voltage = 300
c_cs = 2.7
c_amplitude = 0.1
c_fit_phase = False
c_use_sum = True
c_grid = 2x2x1
out_averages = True
out_average_halves = True
out_skip_first = 0
out_skip_last = 0
device_list = {  }
perdevice = 4
workers = {  }
settings = previous.settings
input_data = {  }
input_data_recursive = False
input_processing = null
output_processing = null

No alternative input specified, will use input parameters from previous.settings
File search will be relative to /path/to/frames
259 files found
Parsing previous results for each item, if available...
259/259, previous metadata found for 0                                                                                                                                       
Connecting to workers...
Unhandled exception. System.TimeoutException: Worker process did not connect within the allotted time.
   at Warp.WorkerWrapper.ListenForPort(String pipeName, Int32 timeoutMilliseconds) in /path/to/WarpLib/WorkerWrapper.cs:line 158
   at Warp.WorkerWrapper..ctor(Int32 deviceID, Boolean silent, Boolean attachDebugger) /path/to/WarpLib/WorkerWrapper.cs:line 90
   at WarpTools.Commands.DistributedOptions.GetWorkers(Boolean attachDebugger) in /path/to/WarpTools/Commands/DistributedOptions.cs:line 57
   at WarpTools.Commands.MotionCTFFrameseries.Run(Object options) in /path/to/WarpTools/Commands/Frameseries/MotionCTFFrameseries.cs:line 187
   at WarpTools.WarpTools.Run(Object options) in /path/to/WarpTools/Program.cs:line 30
   at Warp.Tools.CommandLineParserHelper.ParseAndRun(String[] args, Func`2 run, Type[] verbs, String appName) in /exafs1/well/strubi-stuart/users/dce358/softwares_for_all/warp/WarpLib/Tools/CommandLineParserHelper.cs:line 26
   at WarpTools.WarpTools.Main(String[] args) in/path/to/WarpTools/Program.cs:line 17
   at WarpTools.WarpTools.<Main>(String[] args)
Aborted
compg025:$ Unhandled exception. System.InvalidOperationException: Pipe hasn't been connected yet.
   at System.IO.Pipes.PipeStream.CheckWriteOperations()
   at System.IO.StreamWriter.Flush(Boolean flushStream, Boolean flushEncoder)
   at System.IO.StreamWriter.Dispose(Boolean disposing)
   at System.IO.TextWriter.Dispose()
   at WarpWorker.WarpWorker.SendPort(String pipeName, Int32 port, Int32 timeoutMilliseconds) in /path/to/WarpWorker/WarpWorker.cs:line 154
   at WarpWorker.WarpWorker.Main(String[] args) in/path/to/WarpWorker/WarpWorker.cs:line 94
   at WarpWorker.WarpWorker.<Main>(String[] args)
```
It is hard to know what is causes this to happen, because one doesnt expect for it to happen. However, rerunning the same command again, resolves the problem. Sorry I cannot be mor helpful. I will continue trying and perhaps everytime i fail I will learn more about the conditions that triggers this bug.

shahpnmlab, 2024-05-01:
One potential point that comes to mind is that it is likely to be an HPC probem, where the data lives on a file system and takes time to serve the data to the gpu servers...

thorstenwagner, 2024-05-01:
> [Verfolgungsbild][https://github.com/notifications/beacon/AAIP66KHVJUPVRKW6CIMLMLZAEFDTA5CNFSM6AAAAABG2SELIOWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTT4PZZQ4.gif]

I also run it on a HPC system. Works fine so far. Will check it tomorrow

dtegunov, 2024-05-01:
This sounds like the issue should be reopened.
Could you please `export WARP_DEBUG=1` in the environment where you're running this, and check if anything useful ends up in the worker*.out / worker*.err files?

shahpnmlab, 2024-05-02:
Ok I was able to trigger this bug again and i thankfully had the environment var set as well. Below is output for worker0.err
```
Unhandled exception. System.InvalidOperationException: Pipe hasn't been connected yet.
   at System.IO.Pipes.PipeStream.CheckWriteOperations()
   at System.IO.StreamWriter.Flush(Boolean flushStream, Boolean flushEncoder)
   at System.IO.StreamWriter.Dispose(Boolean disposing)
   at System.IO.TextWriter.Dispose()
   at WarpWorker.WarpWorker.SendPort(String pipeName, Int32 port, Int32 timeoutMilliseconds) in /exafs1/well/strubi-stuart/users/dce358/softwares_for_all/warp/WarpWorker/WarpWorker.cs:line 154
   at WarpWorker.WarpWorker.Main(String[] args) in /exafs1/well/strubi-stuart/users/dce358/softwares_for_all/warp/WarpWorker/WarpWorker.cs:line 94
   at WarpWorker.WarpWorker.<Main>(String[] args)
```
I think this likely due to caching of the data on the compute node. When the data is requested at first call, it is not avaiable immediately and the program crashes becuse its time-out value is smaller than the time within which the data can be fetched from the FS. When the program is called a second time around, the data (or parts of the data) are cached and the program can get on with processing it.
This is just a hypothesis though.

dtegunov, 2024-05-02:
The step failing here is the initial master–worker handshake, which comes long before any data are requested. Let me check if it can be made more robust.
As for data requests, most image read operations are set to wait and try 50 times before giving up, so the overall timeout is well over a minute.

dtegunov, 2024-05-18:
This may be fixed by the timeout increase introduced in a393f22. @thorstenwagner @shahpnmlab can you please check?

shahpnmlab, 2024-05-20:
It seems to be working but, there is a new error in gtom/FFT not sure if it is related.
Output of worker0.out
```
Running on GPU #3 (1487 MB free), port 45889

Received "LoadGainRef", with 5 arguments, for GPU #3, 1487 MB free:
Loaded gain reference and defect map: 4096, 4096, 1, normal, real, 1 A/px, ID = -1, False, False, False, 
Execution took 1.005 seconds

Received "LoadStack", with 4 arguments, for GPU #0, 1487 MB free:
Loaded stack: 4096, 4096, 5, normal, real, 1 A/px, ID = -1, 1
Execution took 1.268 seconds

10 seconds without heartbeat, exiting
2 arguments, for GPU #1, 987 MB free:
```

Output of worker0.err
```
terminate called after throwing an instance of 'std::runtime_error'
  what():  cuFFT error: CUFFT_INTERNAL_ERROR at /softwares/warp/NativeAcceleration/gtom/src/FFT/FFT.cu:23
```

Full stacktrace
```
sh warpTools_20240520_123031.sh
Running command fs_motion_and_ctf with:
m_range_min = 500
m_range_max = 10
m_bfac = -500
m_grid = 1x1x10
c_window = 768
c_range_min = 30
c_range_max = 7
c_defocus_min = 0.5
c_defocus_max = 10
c_voltage = 300
c_cs = 2.7
c_amplitude = 0.1
c_fit_phase = False
c_use_sum = True
c_grid = 2x2x1
out_averages = True
out_average_halves = True
out_skip_first = 0
out_skip_last = 0
device_list = { 0, 1, 2, 3 }
perdevice = 2
workers = {  }
settings = warp_frameseries.settings
input_data = {  }
input_data_recursive = False
input_processing = null
output_processing = null

No alternative input specified, will use input parameters from warp_frameseries.settings
File search will be relative to 20240410_MA104_NSP5GFP_RV17hpi_SC_PNMS_stuart/frames
259 files found
Parsing previous results for each item, if available...
259/259, previous metadata found for 8                                                                                                                                     
Connecting to workers...
Connected to 8 workers
Failed to process /20240410_MA104_NSP5GFP_RV17hpi_SC_PNMS_stuart/warp_frameseries/frame_001_01694_-60.0.eer, marked as unselected           
Unhandled exception. bash: line 1: 156902 Aborted                /softwares_for_all/warp/Release/linux-x64/publish/WarpWorker -d 0 -p 0 --pipe WarpWorkerPipe_2_156820_1563074934 > worker0.out 2> worker0.err
System.AggregateException: One or more errors occurred. (Connection refused (localhost:37570))
 ---> System.Net.Http.HttpRequestException: Connection refused (localhost:37570)
 ---> System.Net.Sockets.SocketException (111): Connection refused
   at System.Net.Sockets.Socket.AwaitableSocketAsyncEventArgs.ThrowException(SocketError error, CancellationToken cancellationToken)
   at System.Net.Sockets.Socket.AwaitableSocketAsyncEventArgs.System.Threading.Tasks.Sources.IValueTaskSource.GetResult(Int16 token)
   at System.Net.Sockets.Socket.<ConnectAsync>g__WaitForConnectWithCancellation|285_0(AwaitableSocketAsyncEventArgs saea, ValueTask connectTask, CancellationToken cancellationToken)
   at System.Net.Http.HttpConnectionPool.ConnectToTcpHostAsync(String host, Int32 port, HttpRequestMessage initialRequest, Boolean async, CancellationToken cancellationToken)
   --- End of inner exception stack trace ---
   at System.Net.Http.HttpConnectionPool.ConnectToTcpHostAsync(String host, Int32 port, HttpRequestMessage initialRequest, Boolean async, CancellationToken cancellationToken)
   at System.Net.Http.HttpConnectionPool.ConnectAsync(HttpRequestMessage request, Boolean async, CancellationToken cancellationToken)
   at System.Net.Http.HttpConnectionPool.CreateHttp11ConnectionAsync(HttpRequestMessage request, Boolean async, CancellationToken cancellationToken)
   at System.Net.Http.HttpConnectionPool.AddHttp11ConnectionAsync(QueueItem queueItem)
   at System.Threading.Tasks.TaskCompletionSourceWithCancellation`1.WaitWithCancellationAsync(CancellationToken cancellationToken)
   at System.Net.Http.HttpConnectionPool.SendWithVersionDetectionAndRetryAsync(HttpRequestMessage request, Boolean async, Boolean doRequestAuth, CancellationToken cancellationToken)
   at System.Net.Http.RedirectHandler.SendAsync(HttpRequestMessage request, Boolean async, CancellationToken cancellationToken)
   at System.Net.Http.HttpClient.<SendAsync>g__Core|83_0(HttpRequestMessage request, HttpCompletionOption completionOption, CancellationTokenSource cts, Boolean disposeCts, CancellationTokenSource pendingRequestsCts, CancellationToken originalCancellationToken)
   --- End of inner exception stack trace ---
   at System.Threading.Tasks.Task.ThrowIfExceptional(Boolean includeTaskCanceledExceptions)
   at System.Threading.Tasks.Task`1.GetResultCore(Boolean waitCompletionNotification)
   at Warp.WorkerConsole.SetFileOutput(String path) in /softwares_for_all/warp/WarpLib/WorkerWrapper.cs:line 541
   at WarpTools.Commands.BaseCommand.<>c__DisplayClass1_0.<IterateOverItems>b__0(Int32 iitem, Int32 threadID) in /softwares_for_all/warp/WarpTools/Commands/BaseCommand.cs:line 103
   at Warp.Tools.Helper.<>c__DisplayClass50_0.<ForCPUGreedy>b__0(Object id) in //softwares_for_all/warp/WarpLib/Tools/Helper.cs:line 768
```

shahpnmlab, 2024-05-20:
Yes, i think it appears to be fixed for now. Thanks!

alisterburt, 2024-05-21:
closing, feel free to reopen if you feel its useful!

xingzhe1129, 2024-05-28:
Hello. When I tried to submit my job to the cluster and it always stopped at connecting to worker. Below is the error. Do you know how to fix it? Thanks.
[image]

alisterburt, 2024-05-28:
Hi @xingzhe1129 - at the top of your screenshot it says `data found for 0` - you're probably not in the right directory or something like this once the job is launched

alisterburt, 2024-05-28:
also, please don't post screenshots of tracebacks - instead copy the text

xingzhe1129, 2024-05-28:
Sure. I don't think I am not in the right directory. I can run very well in the standalone workstation but didn't work in the cluster. The following is the full text of output.

Running command fs_motion_and_ctf with:
m_range_min = 500
m_range_max = 10
m_bfac = -500
m_grid = 1x1x3
c_window = 512
c_range_min = 30
c_range_max = 7
c_defocus_min = 0.5
c_defocus_max = 8
c_voltage = 300
c_cs = 2.7
c_amplitude = 0.07
c_fit_phase = False
c_use_sum = True
c_grid = 2x2x1
out_averages = True
out_average_halves = True
out_skip_first = 0
out_skip_last = 0
device_list = {  }
perdevice = 1
workers = {  }
settings = warp_frameseries.settings
input_data = {  }
input_data_recursive = False
input_processing = null
output_processing = null

No alternative input specified, will use input parameters from warp_frameseries.settings
File search will be relative to /lab-share/PCMM-Wu-e2/Public/xip/frames
328 files found
Parsing previous results for each item, if available...
0/328                                                                              10/328, previous metadata found for 0                                                                              20/328, previous metadata found for 0                                                                              30/328, previous metadata found for 0                                                                              40/328, previous metadata found for 0                                                                              50/328, previous metadata found for 0                                                                              60/328, previous metadata found for 0                                                                              70/328, previous metadata found for 0                                                                              80/328, previous metadata found for 0                                                                              90/328, previous metadata found for 0                                                                              100/328, previous metadata found for 0                                                                              110/328, previous metadata found for 0                                                                              120/328, previous metadata found for 0                                                                              130/328, previous metadata found for 0                                                                              140/328, previous metadata found for 0                                                                              150/328, previous metadata found for 0                                                                              160/328, previous metadata found for 0                                                                              170/328, previous metadata found for 0                                                                              180/328, previous metadata found for 0                                                                              190/328, previous metadata found for 0                                                                              200/328, previous metadata found for 0                                                                              210/328, previous metadata found for 0                                                                              220/328, previous metadata found for 0                                                                              230/328, previous metadata found for 0                                                                              240/328, previous metadata found for 0                                                                              250/328, previous metadata found for 0                                                                              260/328, previous metadata found for 0                                                                              270/328, previous metadata found for 0                                                                              280/328, previous metadata found for 0                                                                              290/328, previous metadata found for 0                                                                              300/328, previous metadata found for 0                                                                              310/328, previous metadata found for 0                                                                              320/328, previous metadata found for 0                                                                              328/328, previous metadata found for 0
Connecting to workers...

alisterburt, 2024-05-28:
@xingzhe1129 if things work when you run locally but not on a cluster then something about the way you are running on your cluster is incorrect - we can't really help with your infrastructure, can you talk to your IT?

xingzhe1129, 2024-05-28:
Thank you so much for your quick response. The error is worker process didn't connect within the allotted time. Do you think it is not a bug of the software? I will also try to talk to my IT. Thanks.

alisterburt, 2024-05-28:
@xingzhe1129 if it runs fine on a workstation accessing the same files in the same way its unlikely a problem with WarpTools 🙂

jpellman, 2024-06-28:
Hi @alisterburt / @dtegunov 

I'm one of the IT people at NYSBC/SEMC.  We're running across a similar issue on our SLURM cluster right now, and I wanted to verify that I accurately understand how your application is architected.  

From what I can tell Warp uses a master/worker structure.  In this setup, workers are spawned on a given host where they then bind to a TCP port.  This TCP port is then communicated back to the master using a named pipe.  The master and the workers then communicate over TCP using the loopback interface (`localhost`) via REST API calls.  The named pipe is only used for this initial setup and is used by the master to map worker processes to ports.

Based on the source code it looks like the path for the named pipe is defined [here](https://github.com/warpem/warp/blob/0d420aab96553461c7ad89e2a061941ae4d930e4/WarpLib/WorkerWrapper.cs#L42).  This path name is then passed in to the worker command [here](https://github.com/warpem/warp/blob/0d420aab96553461c7ad89e2a061941ae4d930e4/WarpLib/WorkerWrapper.cs#L64-L74) (at least on Linux systems).  Moreover, the path looks to be relative, rather than absolute, so the named pipe would be created in whichever directory `WorkerWrapper` was invoked in.

Is this interpretation correct?  I ask because the use of a relative path over an absolute path could be related to this issue.  Namely, if Warp is invoked from a working directory that is under a network volume (e.g., NFS, Lustre), `WorkerWrapper` will call the worker command with a relative path that is under the network volume.  Subsequently, the worker will try to create a named pipe on a network volume.  Named pipes don't really work on network volumes (see [here](https://stackoverflow.com/a/1316195) for an explanation of why) and because of this the initial communication / handshake between the master and worker would fail.

Oftentimes, named pipes / FIFOs are created under whatever the value of `TMPDIR` is (which is usually `/tmp`) in Linux.  _/run_ and _/var/run_ are also common locations for FIFOs (although `/var/run` is no longer supposed to be used).  If the potential issue I described is at play here, it might make sense to change the location of the named pipe from a relative path, to an absolute path under one of these locations.

Many thanks,
John

alisterburt, 2024-06-28:
Hi @jpellman thanks for looking at the issue and outlining a potential solution! What you outline about the system is correct and I learned a bunch from your reply 🙂 

I'll check everything works in our systems with the pipes under `TMPDIR` and add it if it seems to work will merge it in here

alisterburt, 2024-06-28:
possibly solved by #166 - @jpellman if you were having a reproducible issue could you please test and report back? 

I will close for now but please reopen if the same issue pops up

ryanfeathers, 2024-09-05:
> possibly solved by #166 - @jpellman if you were having a reproducible issue could you please test and report back?
> 
> I will close for now but please reopen if the same issue pops up

I think I'm experiencing this same issue unfortunately. Weirdly, it seemed intermittent earlier today but it now won't run at all. I'm using a VM with 4 A100s

```
WarpTools ts_ctf --settings warp_tiltseries.settings --defocus_max 8 --device_list 0 1 2 3 --perdevice 4

Unhandled exception. System.TimeoutException: Worker process did not connect within the allotted time.
   at Warp.WorkerWrapper.ListenForPort(String pipeName, Int32 timeoutMilliseconds) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1725371307707/work/WarpLib/WorkerWrapper.cs:line 167
   at Warp.WorkerWrapper..ctor(Int32 deviceID, Boolean silent, Boolean attachDebugger) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1725371307707/work/WarpLib/WorkerWrapper.cs:line 99
   at WarpTools.Commands.DistributedOptions.GetWorkers(Boolean attachDebugger) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1725371307707/work/WarpTools/Commands/DistributedOptions.cs:line 57
   at WarpTools.Commands.CTFTiltseries.Run(Object options) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1725371307707/work/WarpTools/Commands/Tiltseries/CTFTiltseries.cs:line 60
   at WarpTools.WarpTools.Run(Object options) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1725371307707/work/WarpTools/Program.cs:line 30
   at Warp.Tools.CommandLineParserHelper.ParseAndRun(String[] args, Func`2 run, Type[] verbs, String appName) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1725371307707/work/WarpLib/Tools/CommandLineParserHelper.cs:line 26
   at WarpTools.WarpTools.Main(String[] args) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1725371307707/work/WarpTools/Program.cs:line 17
   at WarpTools.WarpTools.<Main>(String[] args)
```

alisterburt, 2024-09-05:
@ryanfeathers hey! Can you try launching a WarpWorker process yourself separately on a specific port then providing it as an argument to your WarpTools call? Would be useful to see what the worker is saying 🙂

ryanfeathers, 2024-09-06:
@alisterburt hmmm maybe I missed something but the work isn't saying anything. Here's what I did:

- Opened two terminals and ran `export WARP_DEBUG=1` in both. 

- Started the worker in the first one
    ```
    WarpWorker -d 1 -p 8000 --debug 
    Running on GPU #1 (80917 MB free), port 8000 

    Debug mode
    ```

- In the other terminal ran 
    ```
    WarpTools ts_ctf --settings warp_tiltseries.settings --defocus_max 8 --perdevice 4 --workers localhost:8000
    ```

- After finding the files and metadata (which is pretty slow on my filesystem), the job fails with this error

    ```
    Unhandled exception. System.Exception: Couldn't connect to worker at localhost:8000 for 20 seconds, something must be wrong
         at Warp.WorkerWrapper..ctor(String host, Int32 port) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1725371307707/work/WarpLib/WorkerWrapper.cs:line 138
         at WarpTools.Commands.DistributedOptions.GetWorkers(Boolean attachDebugger) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1725371307707/work/WarpTools/Commands/DistributedOptions.cs:line 77
         at WarpTools.Commands.CTFTiltseries.Run(Object options) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1725371307707/work/WarpTools/Commands/Tiltseries/CTFTiltseries.cs:line 60
         at WarpTools.WarpTools.Run(Object options) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1725371307707/work/WarpTools/Program.cs:line 30
         at Warp.Tools.CommandLineParserHelper.ParseAndRun(String[] args, Func`2 run, Type[] verbs, String appName) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1725371307707/work/WarpLib/Tools/CommandLineParserHelper.cs:line 26
         at WarpTools.WarpTools.Main(String[] args) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1725371307707/work/WarpTools/Program.cs:line 17
         at WarpTools.WarpTools.<Main>(String[] args
    Aborted
    ```
    
Side note: I noticed when this job fails because something is missing, the files are marked as "unselected" and will be skipped in the future. If the job runs partially but fails, the completed files are not marked "unselected". So when I can get the job to run, it starts from the beginning every time.

alisterburt, 2024-09-06:
@ryanfeathers thanks for the spelunking! I'm not immediately sure what's wrong and can't reproduce on our system so investigation is tough - you mentioned things working intermittently, is that still the case? Do you have any idea what might be different when it works and when it doesn't? Does your infrastructure come with anyone who might be able to help debug?

ryanfeathers, 2024-09-06:
@alisterburt When I say intermittently, I mean **very** intermittently. Maybe 1 out of 25-50 times I start the job it properly connects to the worker. Otherwise, it reads the files then aborts with the error. It's also worth noting, "parsing previous metadata" takes a minimum of 2-3 minutes due to a large number of files and a slower shared network drive. 

I'm not sure I can get much support for debugging the infrastructure without telling them exactly what to look for. 

So far, n=2 it seems more likely to connect in the morning. But also, it failed with the same error yesterday about halfway through the job 🤷‍♂️

alisterburt, 2024-09-06:
@ryanfeathers 2-3 minutes seems excessive - do you see the same connection problems when data is not on the slow shared drive? If so it's possible that the communication of the port number over the named pipe (as nicely explained by @jpellman above) is still failing

I assume you're running a recent version?

ryanfeathers, 2024-09-06:
Yes, running version 2.0.0 and ran update yesterday! It is an excessive number of files... I copied a subset to fast scratch and can confirm it seems to connecting much better!

alisterburt, 2024-09-07:
Cool, glad you've got a workaround at least! Not ideal though - thanks for the report 🙂

Phaips, 2024-10-17:
Hey,

I am observing something similar. When running locally I get:

```
Running command fs_motion_and_ctf with:
m_range_min = 500
m_range_max = 10
m_bfac = -500
m_grid = 1x1x3
c_window = 512
c_range_min = 30
c_range_max = 7
c_defocus_min = 2
c_defocus_max = 8
c_voltage = 300
c_cs = 2.7
c_amplitude = 0.07
c_fit_phase = False
c_use_sum = True
c_grid = 2x2x1
out_averages = True
out_average_halves = True
out_skip_first = 0
out_skip_last = 0
device_list = {  }
perdevice = 1
workers = {  }
settings = tiltseries.settings
input_data = {  }
input_data_recursive = False
input_processing = null
output_processing = null
input_norawdata = False

No alternative input specified, will use input parameters from tiltseries.settings
File search will be relative to /scicore/home/engel0006/vysamu91/warp/raw
2271 files found
Parsing previous results for each item, if available...
2271/2271, previous metadata found for 0                                                                   
Connecting to workers...
bash: line 1: 641297 Killed                  /scicore/projects/scicore-p-structsoft/ubuntu/software/Miniconda3/miniconda3_python3.12.1/envs/warp/bin/WarpWorker -d 0 -p 0 --pipe /scratch/vysamu91/slurm-job.9488216/WarpWorkerPipe_1_641277_1183645883 > worker0.out 2> worker0.err
Unhandled exception. System.Exception: Couldn't connect to newly created worker at localhost:33847 for 20 seconds, something must be wrong
   at Warp.WorkerWrapper..ctor(Int32 deviceID, Boolean silent, Boolean attachDebugger) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1723683627331/work/WarpLib/WorkerWrapper.cs:line 113
   at WarpTools.Commands.DistributedOptions.GetWorkers(Boolean attachDebugger) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1723683627331/work/WarpTools/Commands/DistributedOptions.cs:line 57
   at WarpTools.Commands.MotionCTFFrameseries.Run(Object options) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1723683627331/work/WarpTools/Commands/Frameseries/MotionCTFFrameseries.cs:line 187
   at WarpTools.WarpTools.Run(Object options) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1723683627331/work/WarpTools/Program.cs:line 30
   at Warp.Tools.CommandLineParserHelper.ParseAndRun(String[] args, Func`2 run, Type[] verbs, String appName) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1723683627331/work/WarpLib/Tools/CommandLineParserHelper.cs:line 26
   at WarpTools.WarpTools.Main(String[] args) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1723683627331/work/WarpTools/Program.cs:line 17
   at WarpTools.WarpTools.<Main>(String[] args)
Aborted
```

and when submitting via SLURM I get:

```
No alternative input specified, will use input parameters from tiltseries.settings
File search will be relative to /scicore/home/engel0006/vysamu91/warp/raw
2271 files found
Parsing previous results for each item, if available...
0/2271Unhandled exception. System.ArgumentOutOfRangeException: count ('-2') must be a non-negative value. (Parameter 'count')
Actual value was -2.
   at System.ArgumentOutOfRangeException.ThrowNegative[T](T value, String paramName)
   at System.ArgumentOutOfRangeException.ThrowIfNegative[T](T value, String paramName)
   at System.String.Ctor(Char c, Int32 count)
   at Warp.Tools.VirtualConsole.ClearLastLine() in /usr/share/miniconda/envs/package-build/conda-bld/warp_1723683627331/work/WarpLib/Tools/VirtualConsole.cs:line 77
   at WarpTools.Commands.BaseOptions.<>c__DisplayClass36_0.<Evaluate>b__1(Int32 i, Int32 threadID) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1723683627331/work/WarpTools/Commands/BaseOptions.cs:line 190
   at Warp.Tools.Helper.<>c__DisplayClass51_0.<ForCPU>b__0(Object id) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1723683627331/work/WarpLib/Tools/Helper.cs:line 719
/var/spool/slurm/job9656540/slurm_script: line 30: 2737702 Aborted
```

Compared to other experiences here it does not seem to run intermittently (never so far). I ran `conda update warp` because I think the version was a bit outdated but it did not fix it unfortunately.

alisterburt, 2024-10-17:
Hey @Phaips - are your problems consistent or intermittent? Is there any output in the worker logs at `<processing_dir>/logs` that might be useful?

Can you confirm you see the `System.ArgumentOutOfRangeException: count ('-2') must be a non-negative value.` error when submitting to slurm on the latest version `2.0.0dev29`? I know @dtegunov was improving the `VirtualConsole` recently...

Phaips, 2024-10-18:
Thank you for the quick reply @alisterburt !
The bug appears consistent, there is no logs generated yet, and it also failed running it only on a small subset of the data. As mentioned in #278 I updated the version but it ran everything from `2.0.0dev25`.
I will report back to see if it works with `2.0.0dev29` :)

Phaips, 2024-10-24:
Hey, sorry for the slow reply but running the `fs_motion_and_ctf ` from `2.0.0dev29 ` still fails:

```
2271/2271, previous metadata found for 0
Connecting to workers...
Unhandled exception. System.Exception: Couldn't connect to newly created worker at localhost:38117 for 20 seconds, something must be wrong
   at Warp.WorkerWrapper..ctor(Int32 deviceID, Boolean silent, Boolean attachDebugger) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1728495487215/work/WarpLib/WorkerWrapper.cs:line 113
   at WarpTools.Commands.DistributedOptions.GetWorkers(Boolean attachDebugger) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1728495487215/work/WarpTools/Commands/DistributedOptions.cs:line 57
   at WarpTools.Commands.MotionCTFFrameseries.Run(Object options) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1728495487215/work/WarpTools/Commands/Frameseries/MotionCTFFrameseries.cs:line 200
   at WarpTools.WarpTools.Run(Object options) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1728495487215/work/WarpTools/Program.cs:line 30
   at Warp.Tools.CommandLineParserHelper.ParseAndRun(String[] args, Func`2 run, Type[] verbs, String appName) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1728495487215/work/WarpLib/Tools/CommandLineParserHelper.cs:line 26
   at WarpTools.WarpTools.Main(String[] args) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1728495487215/work/WarpTools/Program.cs:line 17
   at WarpTools.WarpTools.<Main>(String[] args)
/var/spool/slurm/job10452037/slurm_script: line 30: 2033630 Aborted                 (core dumped) WarpTools fs_motion_and_ctf --settings tiltseries.settings --m_grid 1x1x3 --c_grid 2x2x1 --c_range_max 7 --c_defocus_max 5 --c_defocus_min 1 --c_use_sum --out_averages --out_average_halves --perdevice 4
```

at this point I am not sure if the cause is on my side or not. It fails with the same error trying to run locally and via slurm.
Thanks!

alisterburt, 2024-10-24:
Hey @Phaips - thanks for getting back. I think this is an issue related to something on your HPC system and I see two possible causes for this - neither are particularly straightforward to debug and I apologise for that.

1. the communication with the worker process over a port on localhost is blocked by your IT team
2. the named pipe we set up to communicate the port the worker process is running on back to the master process doesn't work well on your filesystem

### Possibility 1
- launch a `WarpWorker` process manually with `WarpWorker.exe --debug --device 0 --port 8787`
- from a different terminal on the same host hit the REST API of the worker with `curl http://localhost:8787/v1/Console/lines/last10`

You should see a response like

```txt
StatusCode        : 200
StatusDescription : OK
Content           : [{"message":"Running on GPU #0 (22902 MB free), port 8787","timestamp":"2024-10-24T20:27:31.8092484Z","type":0},{"message":"","timestamp":"2024-10-24T20:27:31.8095341Z","type":0},{"message":"Debug mod...
RawContent        : HTTP/1.1 200 OK
                    Content-Length: 255
                    Content-Type: application/json; charset=utf-8
                    Date: Thu, 24 Oct 2024 20:55:41 GMT
                    Server: Kestrel

                    [{"message":"Running on GPU #0 (22902 MB free), port 8787",...
Forms             : {}
Headers           : {[Content-Length, 255], [Content-Type, application/json; charset=utf-8], [Date, Thu, 24 Oct 2024 20:55:41 GMT], [Server, Kestrel]}
Images            : {}
InputFields       : {}
Links             : {}
ParsedHtml        : mshtml.HTMLDocumentClass
RawContentLength  : 255```

If this works it confirms that you can communicate with the worker from localhost over the REST API. Move on to possibility 2

### Possibility 2
- verify that either the environment variable `TMPDIR` is set (`echo $TMPRDIR`) or the directory `/tmp` exists on your filesystem
- if this all looks okay, you'll need to work with your IT to figure out what's going on. It would be helpful to know what kind of filesystem the TMPDIR is on in your infrastructure. These code snippets define where the pipe is created and create it: 
https://github.com/warpem/warp/blob/e6ffcfa410ede6b636eaaf45761379465d9651a7/WarpLib/WorkerWrapper.cs#L64-L83
https://github.com/warpem/warp/blob/e6ffcfa410ede6b636eaaf45761379465d9651a7/WarpWorker/WarpWorker.cs#L135-L156

Phaips, 2024-10-25:
Hey @alisterburt thanks so much for the detailed troubleshooting I appreciate that!

Running the debugs revealed that indeed the first possibility seems to be the cause. Running the` WarpWorker` starts and listens on `port 8787`. However, when try accessing it with `curl` the request gets routed through a proxy server (Squid) instead. The very long error in short is: `ERROR: The requested URL could not be retrieved. Access Denied.` So naturally I tried curl in the `--noproxy` mode and managed to get: 
`[{"message":"Running on GPU #0 (12023 MB free), port 8787","timestamp":"2024-10-25T12:51:31.2428245Z","type":0},{"message":"","timestamp":"2024-10-25T12:51:31.2430946Z","type":0},{"message":"Debug mode","timestamp":"2024-10-25T12:51:31.243806Z","type":0}]`

This seems to be the general issue not only for `curl `! Thus, the `WarpTools` processes should also bypass the proxy when communicating over localhost. So I ran my example with:
```
export no_proxy=localhost,127.0.0.1
WarpTools fs_motion_and_ctf --settings tiltseries.settings --m_grid 1x1x3 --c_grid 2x2x1 --c_range_max 7 --c_defocus_max 5 --c_defocus_min 1 --c_use_sum --out_averages --out_average_halves --perdevice 4
```
and it runs now smoothly! 

As a final remark since (in my case) the user environment is configured to use a proxy server for HTTP/HTTPS requests I would just need to check if our IT department allows to modify the proxy settings. They might need to adjust the proxy configuration to allow local traffic. This might also be the case for others using HPC's.

```
echo $http_proxy
http://squid.deploy.int:3128
```

Thanks again for all the help!

Cheers,
Philippe

alisterburt, 2024-10-25:
Thanks for your patience and for working with me to get through this @Phaips - I'm glad we got to a solution 🙂 

For what it's worth, having local traffic routed through a proxy seems pretty overbearing and I'd be surprised if it was intentional... might be a simple misconfiguration on the IT side

closed 2024-10-25 by alisterburt