warpem/warp #31: Individually deselecting tilts without an existing list?

jmdobbs, 2024-04-28:
Hi,

Doing some testing, it seems like there is no way to choose to individually select or deselect tilts (or tomostars) in the CLI, unless you already know exactly which of those to exclude via the input_data additional option. Is this correct?

Thanks for the great work!!
Joe.

closed 2024-04-29 by jmdobbs

alisterburt, 2024-04-29:
Hey @jmdobbs - thanks for opening this issue!

I'm glad you found the automated routines in `ts_import` and `ts_aretomo`

One route to excluding tilts that are bad in the way you describe is to run BoxNet and use `filter_quality` WarpTool to remove frame series based on the percentage of rubbish in the generated masks 

https://github.com/warpem/warp/blob/090a5a28c017a08eb2292e29219d36d5b98dbfed/WarpTools/Commands/General/FilterQuality.cs#L43-L45

This obviously relies on having a BoxNet model which works well for masking high contrast rubbish.

@dtegunov I had a quick poke around the internet and realised we don't provide a pre-trained model - do we have one somewhere? Should we retrain one?

jmdobbs, 2024-04-29:
Hi @alisterburt. I'd considered this possibility, but found that our usual boxnet was not working so well. Are there meant to be any issues with performance for boxnet models trained on windows but applied in linux? See attached for comparison of the same boxnet model on the same tilt.
[image]