warpem/warp #32: bug: BoxNet model trained on Windows not performing well on linux

alisterburt, 2024-04-29:
@jmdobbs reported in #31 a bad experience using a boxnet model trained on windows on Linux through WarpTools

[image]

alisterburt, 2024-04-29:
@jmdobbs apparently there are slight differences between the BoxNet models in 1.10 (current desktop release) and 2.0 (current code here). @dtegunov will train a new general model sometime over the next week or so

jmdobbs, 2024-04-29:
Thanks for the update, that makes sense!

alisterburt, 2024-05-17:
@jmdobbs small update here - @dtegunov has been retraining but ran into some issues that need a bit of debugging

rbs-sci, 2024-05-20:
Thanks for the info @alisterburt; logged in to ask about this. :)