warpem/warp #33: task: document how to manually select/deselect individual tilts

alisterburt, 2024-04-29:
In #31 @jmdobbs highlighted that the way to select/deselect individual tilts isn't obvious - let's document the process so that we can point people there in the future

dtegunov, 2024-04-29:
We currently have tilt exclusion mechanisms at 4 stages:
1. When a tilt series is imported – the rejected tilts will only remain in the MDOC, Warp won't know about them at all. 
2. Before a tilt stack is exported for external alignment, the UseTilt flags specified in the XML metadata are considered. These can be set manually (we could use some automation here). 
3. When importing tilt alignments, based on shift magnitude. This will affect the UseTilt flags (I think?).
4. Finally, all subsequent processing will respect the UseTilt flags. You can also remove a tilt entirely from the XML file, but this will take more manual effort.

alisterburt, 2024-04-29:
They can also be excluded with the `filter_quality` WarpTool based on the amount of rubbish in BoxNet masks

> Before a tilt stack is exported for external alignment, the UseTilt flags specified in the XML metadata are considered. These can be set manually (we could use some automation here).

@dtegunov what kind of automation were you thinking?

dtegunov, 2024-04-29:
Right, but not once you've produced the .tomostar. This has to happen before stage 1. 

I was thinking basically filter_quality, but on a per-tilt level once the series already exists.

alisterburt, 2024-04-29:
gotcha - yep we should add that

shahpnmlab, 2024-04-30:
I have tried to handle this in two ways (both are covered in Warp options) - But the way I am doing it for now is after building the stack I run my imod wrappers on the stacks, before performing actual alignments, i scan for dark tilts (which you also do) and remove them from the stack and log those frames, I then run a coarse alignment to determine badly tracking tilts (which you do too --FOV?)  which also get removed and logged to a text file. Alignments are then carried out on the decimated stack. Then before importing the xf files i rename the XML files of the aberrant frames.

alisterburt, 2024-04-30:
You're up late @shahpnmlab ! 

I think as you said we have what you're doing covered but rerunning tilt-series alignment without badly tracked tilts is a good callout and we're not currently doing this... the question is then: should we do this automatically? There are a few cases we've seen where patch tracking will work very well and aretomo will fail or vice versa

shahpnmlab, 2024-04-30:
A simple xcorr w/ or w/o cosine stretching should be relatively easy to run to find views w/ shifts larger shifts should then be thrown away immediately (this is how i do it in my wrapper https://github.com/shahpnmlab/automateImod/blob/d26d36596fd410064f54b2dc5e93e4b4a7b73b29/automateImod/utils.py#L11 for filtering).

alisterburt, 2024-05-18:
closed by #82 - see https://warpem.github.io/warp/reference/warptools/subsets/

closed 2024-05-18 by alisterburt