warpem/warp #37: QoL suggestion for EER handling

shahpnmlab, 2024-04-30:
I read the draft docs, and EER support conitnues to be rudimentary, which is great in this case as I what would very dearly like to have from a UX perspective is to provide the total dose (e.g. 40 e/Å^2 for frame series or 3.5e/Å^2 for a TS) and the dose per frame eg. 1e/Å^2, and have Warp handle the grouping in the background. Ofc, this is not an urgent request, and you guys have more important bugs to iron out, but if you could modify the eer inputs this way in the near future would be super fantastic!

alisterburt, 2024-04-30:
this feels like a good suggestion to me, thanks @shahpnmlab ! 

@dtegunov what do you think?

alisterburt, 2024-04-30:
Discussed briefly with @shahpnmlab as what he was saying wasn't obvious to me

- EER files have a large number of internal sections
- any grouping of those sections might not be evenly divisible, what do you do with leftover frames?

c.f. https://relion.readthedocs.io/en/release-3.1/Reference/MovieCompression.html#falcon4-eer

dtegunov, 2024-05-01:
Leftover frames are a non-issue for 2D data (who cares about it after 40 e/A^2), and negligible for tilt movies (you'd usually use finer fractionation there to correct the extreme drift at high tilt). We can check a single movie during settings creation and print the estimated information loss.

I'll add the exposure-based group size definition in addition to `--eer_ngroups`.

dtegunov, 2024-05-01:
Addressed in 5b7fc29703e2ba45c4b5359ac1244107628c86c8

closed 2024-05-01 by dtegunov