warpem/warp #38: cuFFT error during template matching

thorstenwagner, 2024-04-30:
I'm using the torch-2 branch and run into following exception during template matching:

```
No alternative input specified, will use input parameters from warp_tiltseries.settings
File search will be relative to /mnt/beegfs/home/sphire-devel/warptest/tomostar
5 files found
Parsing previous results for each item, if available...
5/5, previous metadata found for 5                             
Downloading map from EMDB: 100.00%                                                                               
Extracting downloaded map... Done
Setting --template_angpix to 0.7289999 based on template map
Using 1536 orientations for matching
Connecting to workers...
Connected to 4 workers
0/5terminate called after throwing an instance of 'std::runtime_error'
  what():  cuFFT error: CUFFT_INTERNAL_ERROR at /mnt/beegfs/software/em/warp/2.0beta/src/warp/NativeAcceleration/gtom/src/FFT/IFFT.cu:24

Failed to process /mnt/beegfs/home/sphire-devel/warptest/warp_tiltseries/TS_1.tomostar, marked as unselected     
Unhandled exception. System.AggregateException: One or more errors occurred. (Connection refused (localhost:37457))
 ---> System.Net.Http.HttpRequestException: Connection refused (localhost:37457)
 ---> System.Net.Sockets.SocketException (111): Connection refused
   at System.Net.Sockets.Socket.AwaitableSocketAsyncEventArgs.ThrowException(SocketError error, CancellationToken cancellationToken)
   at System.Net.Sockets.Socket.AwaitableSocketAsyncEventArgs.System.Threading.Tasks.Sources.IValueTaskSource.GetResult(Int16 token)
   at System.Net.Sockets.Socket.<ConnectAsync>g__WaitForConnectWithCancellation|285_0(AwaitableSocketAsyncEventArgs saea, ValueTask connectTask, CancellationToken cancellationToken)
   at System.Net.Http.HttpConnectionPool.ConnectToTcpHostAsync(String host, Int32 port, HttpRequestMessage initialRequest, Boolean async, CancellationToken cancellationToken)
   --- End of inner exception stack trace ---
   at System.Net.Http.HttpConnectionPool.ConnectToTcpHostAsync(String host, Int32 port, HttpRequestMessage initialRequest, Boolean async, CancellationToken cancellationToken)
   at System.Net.Http.HttpConnectionPool.ConnectAsync(HttpRequestMessage request, Boolean async, CancellationToken cancellationToken)
   at System.Net.Http.HttpConnectionPool.CreateHttp11ConnectionAsync(HttpRequestMessage request, Boolean async, CancellationToken cancellationToken)
   at System.Net.Http.HttpConnectionPool.AddHttp11ConnectionAsync(QueueItem queueItem)
   at System.Threading.Tasks.TaskCompletionSourceWithCancellation`1.WaitWithCancellationAsync(CancellationToken cancellationToken)
   at System.Net.Http.HttpConnectionPool.SendWithVersionDetectionAndRetryAsync(HttpRequestMessage request, Boolean async, Boolean doRequestAuth, CancellationToken cancellationToken)
   at System.Net.Http.RedirectHandler.SendAsync(HttpRequestMessage request, Boolean async, CancellationToken cancellationToken)
   at System.Net.Http.HttpClient.<SendAsync>g__Core|83_0(HttpRequestMessage request, HttpCompletionOption completionOption, CancellationTokenSource cts, Boolean disposeCts, CancellationTokenSource pendingRequestsCts, CancellationToken originalCancellationToken)
   --- End of inner exception stack trace ---
   at System.Threading.Tasks.Task.ThrowIfExceptional(Boolean includeTaskCanceledExceptions)
   at System.Threading.Tasks.Task`1.GetResultCore(Boolean waitCompletionNotification)
   at Warp.WorkerConsole.SetFileOutput(String path) in /mnt/beegfs/software/em/warp/2.0beta/src/warp/WarpLib/WorkerWrapper.cs:line 540
   at WarpTools.Commands.BaseCommand.<>c__DisplayClass1_0.<IterateOverItems>b__0(Int32 iitem, Int32 threadID) in /mnt/beegfs/software/em/warp/2.0beta/src/warp/WarpTools/Commands/BaseCommand.cs:line 103
   at Warp.Tools.Helper.<>c__DisplayClass50_0.<ForCPUGreedy>b__0(Object id) in /mnt/beegfs/software/em/warp/2.0beta/src/warp/WarpLib/Tools/Helper.cs:line 768
EMPIAR-10491_5TS_e2e.sh: line 131: 383355 Aborted                 WarpTools ts_template_match --settings warp_tiltseries.settings --tomo_angpix 10 --template_emdb 15854 --template_diameter 130 --symmetry O --perdevice 2
terminate called after throwing an instance of 'std::runtime_error'
  what():  cuFFT error: CUFFT_INTERNAL_ERROR at /mnt/beegfs/software/em/warp/2.0beta/src/warp/NativeAcceleration/gtom/src/FFT/IFFT.cu:24
```

Any idea what might be the problem?

btw, there is no logs folder ... so I cant provide any additional logs.

alisterburt, 2024-04-30:
Are you sure there is no logs folder? It should be inside your `warp_tiltseries` directory - I would be surprised if there was nothing as it is the worker process which calls things from cuFFT via `NativeAcceleration`

thorstenwagner, 2024-04-30:
Ah, I thought it needs be in the project folder.

Here we are:
[logs.tar.gz](https://github.com/warpem/warp/files/15160538/logs.tar.gz)

ThomasHoffmann77, 2024-04-30:
I had the same error when the GPU ran out of memory. For one of the tests I reduced the number of workers

thorstenwagner, 2024-04-30:
> I had the same error when the GPU ran out of memory. For one of the tests I reduced the number of workers

Thanks for the info!

I would expect that running it on RTX 6000 with 22GB should be sufficient?

thorstenwagner, 2024-04-30:
```
WarpTools·ts_template_match·\✂                                                   
 13 --settings·warp_tiltseries.settings·\✂                                           
 12 --tomo_angpix·10·\✂                                                              
 11 --template_emdb·15854·\✂                                                         
 10 --template_diameter·130·\✂                                                       
  9 --symmetry·O·\✂                                                                  
  8 --perdevice·2✂    
```

So simply reducing --perdevice to 1?

alisterburt, 2024-04-30:
I'm going to close this assuming that it was indeed an out of memory error - please reopen if that wasn't the case!

closed 2024-04-30 by alisterburt