warpem/warp #39: perf: local resolution estimation is very slow in `MCore`

alisterburt, 2024-04-30:
simplified log with paths removed below, local res takes ~5 mins for a 330^3 volume - looking at CPU utilisation it appears to be running in a single thread at 100% for the full duration

```text
2024-04-30 10:50:52.524 Received "MPAFinishSpecies", with 3 arguments, for GPU #0, 79391 MB free:
2024-04-30 10:50:54.352 Reconstructing half-map 1...
2024-04-30 10:51:44.872 Reconstructing half-map 2...
2024-04-30 10:52:35.321 Finalizing map...
2024-04-30 10:52:35.326 1/5: Global FSC
2024-04-30 10:52:47.306 Global resolution is 20.590
2024-04-30 10:52:47.307 2/5: Anisotropic FSC
2024-04-30 10:52:50.534 3/5: Local resolution
2024-04-30 10:57:39.931 4/5: Training denoising
2024-04-30 10:57:42.712 4/5: Training denoising: Preparing mask... 
2024-04-30 10:58:32.605 4/5: Training denoising: 300/300, 0:00:00 remaining, log(loss) = -2.2592
2024-04-30 10:58:32.672 5/5: Applying denoising
2024-04-30 10:58:43.275 Execution took 470.745 seconds
```

shahpnmlab, 2024-04-30:
@dtegunov i wonder if the Occupy local scale estimation could stand in as a substitute for this tedious calc. Robust local rec calcs can then be carried out at the very end of the refinement pipeline.

dtegunov, 2024-05-01:
It can get quite slow when you combine a large box with low resolution (the local window size is increased for low res maps). It's calculated on the GPU, so the CPU thread is just waiting for the kernels to execute. The original intent was to provide local filtering capabilities, but since n2n is much better, it only serves a visualization purpose now. I'll consider doing it on a sparser grid and interpolating instead.

alisterburt, 2024-05-01:
Makes sense! Not a big deal but good to keep in mind

closed 2024-05-01 by dtegunov