warpem/warp #40: bug: warp errors out after succesful aretomo runs

jmdobbs, 2024-04-30:
Hi,
From the testing I and @ThomasHoffmann77 have been doing, it appears that warp doesn't properly recognize successful aretomo jobs. After the job finishes running (and the files are written to the tiltstack directory), we get the error

2024-04-30 08:00:25.140 System.InvalidOperationException: Cannot mix synchronous and asynchronous operation on process stream.
2024-04-30 08:00:25.140    at System.Diagnostics.Process.get_StandardOutput()
2024-04-30 08:00:25.140    at WarpWorker.WarpWorker.EvaluateCommand(NamedSerializableObject Command)

See attached for job log
[gTS_003.log](https://github.com/warpem/warp/files/15168168/gTS_003.log)

alisterburt, 2024-04-30:
This one looks like a real bug, I'm not sure why you're running into it when we're not - @dtegunov  could this be caused by the recent  changes you made to logging?

dtegunov, 2024-05-01:
Correct, fixed in fc22b43f5cd826a0760a1d95dc174ab67804d93b

closed 2024-05-01 by dtegunov