warpem/warp #41: Aretomo warp compatibility flags possibly wrong in job submission?

jmdobbs, 2024-05-01:
Hi,
With the update from yesterday, aretomo is now running fine. However, warp is unable to find the .xf file. I think this may because the wrong --outimod flag is passed during the job submission. The job passes outimod 1, whereas the usual flag for compatibility with warp is --outimod 2
 --outxf is also set to 0 (should presumably be 1?)

024-05-01 07:28:48.217 Executing AreTomo2 in /struct/mahamid/jdobbs/warp2_testing/data_gseries/processing/tiltstack/gTS_003 with arguments: -InMrc gTS_003.st -AngFile gTS_003.rawtlt -VolZ 0 -OutBin 0 -TiltAxis 85.2 -1 -AlignZ 100 -TiltCor 1 -OutImod 1 -DarkTol 0 -OutMrc gTS_003_aligned.mrc -Gpu 0
2024-05-01 07:28:48.582 
2024-05-01 07:28:48.584 -InMrc      gTS_003.st
2024-05-01 07:28:48.586 -OutMrc     gTS_003_aligned.mrc
2024-05-01 07:28:48.588 -AlnFile    
2024-05-01 07:28:48.589 -AngFile    gTS_003.rawtlt
2024-05-01 07:28:48.591 -RoiFile    
2024-05-01 07:28:48.593 -TmpFile    
2024-05-01 07:28:48.594 -LogFile    
2024-05-01 07:28:48.596 -AlignZ     100
2024-05-01 07:28:48.598 -VolZ       0
2024-05-01 07:28:48.599 -OutBin     1.00
2024-05-01 07:28:48.601 -TiltRange  0.00  0.00
2024-05-01 07:28:48.603 -TiltAxis   85.20  -1.00
2024-05-01 07:28:48.604 -Gpu        0
2024-05-01 07:28:48.606 -TiltCor    1.00  0.00
2024-05-01 07:28:48.608 -ReconRange  -90.00  90.00
2024-05-01 07:28:48.610 -PixSize    0.00
2024-05-01 07:28:48.612 -ImgDose    0.00
2024-05-01 07:28:48.613 -Kv         0.00
2024-05-01 07:28:48.615 -Cs         0.00
2024-05-01 07:28:48.617 -AmpContrast  0.07
2024-05-01 07:28:48.619 -ExtPhase   0.00  0.00
2024-05-01 07:28:48.621 -FlipVol    0
2024-05-01 07:28:48.623 -FlipInt    0
2024-05-01 07:28:48.624 -Sart       20  5
2024-05-01 07:28:48.626 -Wbp        0
2024-05-01 07:28:48.628 -Patch      0  0
2024-05-01 07:28:48.630 -TiltScheme  0.00  0.0  0.0
**2024-05-01 07:28:48.632 -OutXF      0**
2024-05-01 07:28:48.633 -Align      1
2024-05-01 07:28:48.635 -CropVol    0  0
**2024-05-01 07:28:48.637 -OutImod    1**
2024-05-01 07:28:48.639 -DarkTol    0.00
2024-05-01 07:28:48.641 -Bft        500.0  500.0
2024-05-01 07:28:48.643 -IntpCor    1

jmdobbs, 2024-05-01:
Odd, it's AreTomo2 here but the flags should be the same. The AreTomo --help for 1.3, like for 2.0, says that --outImod 1 should only generate imod files for relion4, not warp

alisterburt, 2024-05-01:
Could you attempt with 1.3.4 and report back? If I don't hear back I'll close this in a few days as it appears Warp is working as expected here

dtegunov, 2024-05-01:
The flags were working as described in the docs until some version, then I updated to 1.3.4 and had to change them to get anything useful. It's makes sense that they fixed this bug in a later version. That's what we get by relying on buggy behavior.

However, I also saw another WarpTools bug when testing this yesterday where it complained about not finding the .xf when I specified --output_processing. Are you possibly running into this one? I'll look into it today.

jmdobbs, 2024-05-01:
That makes sense, I used 1.3.0 and 2.0 and got the same results. We don't have 1.3.4 installed.

The exact command I ran was 
WarpTools ts_aretomo --settings TS_settings.settings --angpix 20 --alignz 2000 --min_fov 0.5

jmdobbs, 2024-05-02:
from the testing @ThomasHoffmann77 and I have done: even with aretomo1.3.4 the warp flags produce the incorrect result, our aretomo installation (with suffix Cuda118) appears to behave according to the docs

jmdobbs, 2024-05-03:
Thomas was able to patch our module to use --outimod 2, and this did not change the behavior (or created files at all). However, this allowed us to recognize that it was the formatting of the generated files that was wrong. Running aretomo with these parameters (above) generated gTS_003_st.xf, where warp needs gTS_003.xf

alisterburt, 2024-05-06:
@jmdobbs I'm going to update our wrapper to work with AreTomo2 which should fix this - please hold 🙂

alisterburt, 2024-05-10:
closed by #57

closed 2024-05-10 by alisterburt