warpem/warp #42: Test script does not run through

thorstenwagner, 2024-05-02:
Hi,

warp some don't finish the script. I see some errors in the error files, but its hard to tell for me what went wrong. You fill find attached the log,stderr and stdout files of both runs:

[second_trial.tar.gz](https://github.com/warpem/warp/files/15184579/second_trial.tar.gz)
[first_trial.tar.gz](https://github.com/warpem/warp/files/15184580/first_trial.tar.gz)

**first_trial.tar.gz: First run with an older relion 5 version ( relion/cuda/ver5.0_20240412T0851070200 )**

**second_trial.tar.gz:  this time with the latest relion/cuda/ver5.0_GPU_20240430T1027350200**

You can also find the module file (2.0beta) we use attached.

[2.0beta.tar.gz](https://github.com/warpem/warp/files/15184585/2.0beta.tar.gz)

The second trial has crashed much earlier than the first one

Maybe you can tell me what went wrong :-)

Best,
Thorsten

dtegunov, 2024-05-02:
Hey! 

Are you sure first_trial is really the first? It looks like you ended up with 2 identical AF species in the same population, so there must have been some results left from a previous run. Overall, you managed to get to just below 3.0 A, which is close to the expected 2.9 A for the whole script – so everything went almost OK :-)

In second_trial, it looks like Relion failed to run Refine3D because of insufficient slots on that system: "There are not enough slots available in the system to satisfy the 3 slots that were requested by the application"

thorstenwagner, 2024-05-03:
Hi Dimitry!

Thanks for your response :-) 

first_trial was restarted a few times - so it was not really the first trial :-) I checked the output, but I did not find the "final" reconstruction:
[image]

I opened the d_half_apoferritin.mrc, but they were all super noisy. Our warp users told me there should be a filtered map etc? Where can I find that?

Anyway, I will start from scratch.

As for the second attempt:

I am using the exact same submission script (submitting to a single GPU node) as the first run.  Only the relion module has changed. Bit odd.

dtegunov, 2024-05-03:
Ah, sorry about that, the d_* stuff is debug output that I need to disable. The relevant species files are located in m/species/[species name].

thorstenwagner, 2024-05-06:
I've added the `--oversubscribe` option to the mpirun command in the script. Lets see if that helps. No matter what relion version I use it crashes now with the  `There are not enough slots available in the system to satisfy the 3 slots that were requested by the application`.  I submit with `--ntasks-per-node 20`. But at the time when mpirun is called, there seems to be not enough free cores available.

Could it be that some processes continue to run in the background? That would also explain why it almost worked when I restarted warp several times for the first trial, because the problematic processes might be skipped?

thorstenwagner, 2024-05-06:
Works now! Added an PR to make it work for others with a similar setup too :-)