warpem/warp #48: Export particles "no particles to write out"

asarnow, 2024-05-06:
It seems like I have everything working right up to the particle export step. The clean star files exist and all have particles, and all the rlnMicrographName entries match .tomostar files in the tomostar firectory as specified in the settings file.

```
(base) [asarnow@emnodeb37 WARP]$ WarpTools ts_export_particles --settings warp_tiltseries.settings --input_directory warp_tiltseries/matching --input_pattern "*_clean.star" --normalized_coords --output_star relion/matching.star --output_angpix 4.98 --box 72 --diameter 300 --relative_output_paths --2d --perdevice 1
Running command ts_export_particles with:
input_star = null
input_directory = warp_tiltseries/matching
input_pattern = *_clean.star
coords_angpix = null
normalized_coords = True
output_star = relion/matching.star
output_angpix = 4.98
box = 72
diameter = 300
relative_output_paths = True
2d = True
3d = False
dont_normalize_input = False
dont_normalize_3d = False
n_tilts = null
device_list = {  }
perdevice = 1
workers = {  }
settings = warp_tiltseries.settings
input_data = {  }
input_data_recursive = False
input_processing = null
output_processing = null

No alternative input specified, will use input parameters from warp_tiltseries.settings
File search will be relative to /home/asarnow/Projects/Misc/WARP/tomostar
375 files found
Parsing previous results for each item, if available...
375/375, previous metadata found for 375
Found 375 files in warp_tiltseries/matching matching *_clean.star;
Found 279191 particles in 375 tilt series
Connecting to workers...
Connected to 4 workers
375/375, 00:00 remaining
Finished processing in 00:00:00
Unhandled exception. System.InvalidOperationException: No particles to write out. Check that rlnMicrographName entries match tomostar file for each tilt series.
   at WarpTools.Commands.ExportParticlesTiltseries.WriteOutputStarFile(Dictionary`2 perTiltSeriesTables, String particleStarPath, Int32 outputDimensionality) in /home/asarnow/local/src/warp/WarpTools/Commands/Tiltseries/ExportParticlesTiltseries.cs:line 821
   at WarpTools.Commands.ExportParticlesTiltseries.Run(Object options) in /home/asarnow/local/src/warp/WarpTools/Commands/Tiltseries/ExportParticlesTiltseries.cs:line 299
   at WarpTools.WarpTools.Run(Object options) in /home/asarnow/local/src/warp/WarpTools/Program.cs:line 30
   at Warp.Tools.CommandLineParserHelper.ParseAndRun(String[] args, Func`2 run, Type[] verbs, String appName) in /home/asarnow/local/src/warp/WarpLib/Tools/CommandLineParserHelper.cs:line 26
   at WarpTools.WarpTools.Main(String[] args) in /home/asarnow/local/src/warp/WarpTools/Program.cs:line 17
   at WarpTools.WarpTools.<Main>(String[] args)
Aborted
```

alisterburt, 2024-05-06:
Hi @asarnow ! Sorry you're running into issues - it looks like you're trying to extract the particles after thresholding template matching results, can you confirm that you're running everything inside warp for now and haven't brought anything in from outside?

Could you also provide any logs in `<processing_dir>/logs`? 

I'll add some debug output to a branch for you to run now

alisterburt, 2024-05-06:
I've pushed a branch (df51156) which adds some useful debug printing and should help us figure out what's going on 

Would you mind building and running that?

example output:
```txt
5 files found
Parsing previous results for each item, if available...
5/5, previous metadata found for 5                                                                                                                 
Found 5 files in warp_tiltseries/matching matching *15854_clean.star;
TS: TS_32.tomostar   Particles: 248
TS: TS_11.tomostar   Particles: 310
TS: TS_17.tomostar   Particles: 142
TS: TS_1.tomostar   Particles: 250
TS: TS_23.tomostar   Particles: 222
Found 1172 particles in 5 tilt series
Connecting to workers...
Connected to 1 workers
0/5Processing TS_11.tomostar
310 particles for TS_11.tomostar
1/5, 00:25 remainingProcessing TS_1.tomostar                                                                                                       
250 particles for TS_1.tomostar
2/5, 00:18 remainingProcessing TS_17.tomostar                                                                                                      
142 particles for TS_17.tomostar
3/5, 00:11 remainingProcessing TS_32.tomostar                                                                                                      
248 particles for TS_32.tomostar
4/5, 00:05 remainingProcessing TS_23.tomostar                                                                                                      
222 particles for TS_23.tomostar
5/5, 00:00 remaining                                                                                                                               
Finished processing in 00:00:29
Saying goodbye to all workers... Done
```

alisterburt, 2024-05-06:
Just FYI I'm on Euro time right now and signing off - hopefully debug output is illuminating and there's a simple fix!

dtegunov, 2024-05-07:
(@alisterburt if you think this debugging code will be useful in the future, please make it conditional on WARP_DEBUG and merge into main later)

alisterburt, 2024-05-07:
now in main, @asarnow please rebuild from main and `export WARP_DEBUG=1` before running to see debug output

asarnow, 2024-05-07:
Thanks Alister! I confirm I'm not doing anything funny right now, just following the e2e script with my own data. The last entries in warp_tiltseries/logs is for the successful template matching step. There's no log directory in my top project directory, but stderr and stdout are captured by SLURM.

With `WARP_DEBUG=1` my output is:
```
No alternative input specified, will use input parameters from warp_tiltseries.settings
File search will be relative to /home/asarnow/Projects/Misc/WARP/tomostar
375 files found
Parsing previous results for each item, if available...
375/375, previous metadata found for 375
Found 375 files in warp_tiltseries/matching matching *OC43-S_72px_clean.star;
Found 279191 particles in 375 tilt series
Connecting to workers...
Connected to 4 workers
375/375, 00:00 remaining
Finished processing in 00:00:00
Unhandled exception. System.InvalidOperationException: No particles to write out. Check that rlnMicrographName entries match tomostar file for each tilt series.
   at WarpTools.Commands.ExportParticlesTiltseries.WriteOutputStarFile(Dictionary`2 perTiltSeriesTables, String particleStarPath, Int32 outputDimensionality) in /home/asarnow/local/src/warp/WarpTools/Commands/Tiltseries/ExportParticlesTiltseries.cs:line 821
   at WarpTools.Commands.ExportParticlesTiltseries.Run(Object options) in /home/asarnow/local/src/warp/WarpTools/Commands/Tiltseries/ExportParticlesTiltseries.cs:line 299
   at WarpTools.WarpTools.Run(Object options) in /home/asarnow/local/src/warp/WarpTools/Program.cs:line 30
   at Warp.Tools.CommandLineParserHelper.ParseAndRun(String[] args, Func`2 run, Type[] verbs, String appName)
   at WarpTools.WarpTools.Main(String[] args) in /home/asarnow/local/src/warp/WarpTools/Program.cs:line 17
   at WarpTools.WarpTools.<Main>(String[] args)
Aborted
```

asarnow, 2024-05-07:
I am stupid, didn't pull from git first. I think passing WARP_DEBUG on the command line is working?

```
$WARP_DEBUG=1 WarpTools ts_export_particles --settings warp_tiltseries.settings --input_directory warp_tiltseries/matching --input_pattern "*_clean.star" --normalized_coords --output_star relion/matching.star --output_angpix 4.98 --box 72 --diameter 300 --relative_output_paths --2d --perdevice 1 --device_list 0
Running command ts_export_particles with:
input_star = null
input_directory = warp_tiltseries/matching
input_pattern = *_clean.star
coords_angpix = null
normalized_coords = True
output_star = relion/matching.star
output_angpix = 4.98
box = 72
diameter = 300
relative_output_paths = True
2d = True
3d = False
dont_normalize_input = False
dont_normalize_3d = False
n_tilts = null
device_list = { 0 }
perdevice = 1
workers = {  }
settings = warp_tiltseries.settings
input_data = {  }
input_data_recursive = False
input_processing = null
output_processing = null

No alternative input specified, will use input parameters from warp_tiltseries.settings
File search will be relative to /home/user/Projects/Misc/WARP/tomostar
375 files found
Parsing previous results for each item, if available...
375/375, previous metadata found for 375
Found 375 files in warp_tiltseries/matching matching *_clean.star;
TS: DA23-3_pace2_p12_ts_021.tomostar   Particles: 545
TS: DA23-3_pace2_p12_ts_022.tomostar   Particles: 569
TS: DA23-3_pace2_p12_ts_023.tomostar   Particles: 632
TS: DA23-3_pace2_p12_ts_024.tomostar   Particles: 621
TS: DA23-3_pace2_p12_ts_025.tomostar   Particles: 567
TS: DA23-3_pace2_p11_ts_023.tomostar   Particles: 905
TS: DA23-3_pace2_p12_ts_001.tomostar   Particles: 559
TS: DA23-3_pace2_p12_ts_002.tomostar   Particles: 617
TS: DA23-3_pace2_p12_ts_003.tomostar   Particles: 590
TS: DA23-3_p01_ts_001.tomostar   Particles: 611
TS: DA23-3_pace2_p12_ts_006.tomostar   Particles: 635
TS: DA23-3_pace2_p12_ts_004.tomostar   Particles: 539
TS: DA23-3_pace2_p01_ts_020.tomostar   Particles: 580
TS: DA23-3_p01_ts_024.tomostar   Particles: 570
TS: DA23-3_pace2_p03_ts_016.tomostar   Particles: 564
TS: DA23-3_pace2_p12_ts_008.tomostar   Particles: 478
TS: DA23-3_p02_ts_022.tomostar   Particles: 469
TS: DA23-3_pace2_p04_ts_014.tomostar   Particles: 718
TS: DA23-3_pace2_p02_ts_018.tomostar   Particles: 376
TS: DA23-3_pace2_p10_ts_002.tomostar   Particles: 831
TS: DA23-3_pace2_p05_ts_012.tomostar   Particles: 947
TS: DA23-3_pace2_p08_ts_006.tomostar   Particles: 949
TS: DA23-3_pace2_p12_ts_009.tomostar   Particles: 580
TS: DA23-3_pace2_p06_ts_010.tomostar   Particles: 926
TS: DA23-3_ts_019.tomostar   Particles: 919
TS: DA23-3_pace2_p07_ts_008.tomostar   Particles: 416
TS: DA23-3_pace2_p12_ts_011.tomostar   Particles: 654
TS: DA23-3_pace2_p10_ts_025.tomostar   Particles: 921
TS: DA23-3_pace2_p09_ts_004.tomostar   Particles: 937
TS: DA23-3_pace2_p12_ts_012.tomostar   Particles: 609
TS: DA23-3_pace2_p12_ts_007.tomostar   Particles: 523
TS: DA23-3_pace2_p12_ts_005.tomostar   Particles: 677
TS: DA23-3_pace2_p12_ts_014.tomostar   Particles: 610
TS: DA23-3_pace2_p12_ts_015.tomostar   Particles: 417
TS: DA23-3_pace2_p12_ts_013.tomostar   Particles: 604
[... 1089 lines cut ...]
373/375, 00:00 remaining
Processing DA23-3_pace2_p07_ts_004.tomostar
374/375, 00:00 remaining
Processing DA23-3_pace2_p01_ts_012.tomostar
375/375, 00:00 remaining
Finished processing in 00:00:01
Unhandled exception. System.InvalidOperationException: No particles to write out. Check that rlnMicrographName entries match tomostar file for each tilt series.
   at WarpTools.Commands.ExportParticlesTiltseries.WriteOutputStarFile(Dictionary`2 perTiltSeriesTables, String particleStarPath, Int32 outputDimensionality) in /home/user/local/src/warp/WarpTools/Commands/Tiltseries/ExportParticlesTiltseries.cs:line 840
   at WarpTools.Commands.ExportParticlesTiltseries.Run(Object options) in /home/user/local/src/warp/WarpTools/Commands/Tiltseries/ExportParticlesTiltseries.cs:line 318
   at WarpTools.WarpTools.Run(Object options) in /home/user/local/src/warp/WarpTools/Program.cs:line 30
   at Warp.Tools.CommandLineParserHelper.ParseAndRun(String[] args, Func`2 run, Type[] verbs, String appName) in /home/asarnow/local/src/warp/WarpLib/Tools/CommandLineParserHelper.cs:line 26
   at WarpTools.WarpTools.Main(String[] args) in /home/user/local/src/warp/WarpTools/Program.cs:line 17
   at WarpTools.WarpTools.<Main>(String[] args)
Aborted
$ bash: line 1: 124234 Killed                  /home/user/local/src/warp/Release/linux-x64/publish/WarpWorker -d 0 -p 0 --pipe WarpWorkerPipe_1_124189_1774910816 > worker0.out 2> worker0.err

$ cat worker0.out
Running on GPU #0 (10851 MB free), port 41562

Received "LoadGainRef", with 5 arguments, for GPU #0, 10851 MB free:
Loaded gain reference and defect map: 5760, 4092, 1, normal, real, 1 A/px, ID = -1, False, True, False,
Execution took 0.573 seconds

10 seconds without heartbeat, exiting
```

alisterburt, 2024-05-08:
This is super strange, the names match perfectly. Can you confirm that CTF estimation on the tilt series level ran as expected? There is an early exit clause which until #56 (just now) was silent and might be the culprit here

https://github.com/warpem/warp/blob/bbce5a0825fb6535141cd1cf3eeac6f86726d71d/WarpTools/Commands/Tiltseries/ExportParticlesTiltseries.cs#L193-L197

I added some more debug output in #56 - running again should tell us if this is what's happening: if you don't have CTF info present the only thing I can think of is that you missed running `ts_ctf` or it crashed for some reason?

asarnow, 2024-05-08:
Ah, they do all have the CTF metadata error. I did run ts_ctf...I am re-running it. What should be in the tiltseries XML file if ts_ctf succeeds? Mine seem to have a defocus value but the resolution estimate is 0.

asarnow, 2024-05-09:
After rerunning `ts_ctf` and `ts_reconstruct` the extraction completed successfully. I suppose that previously `ts_ctf` ran out of memory but that didn't interfere with reconstruction, even with deconvolution, or template matching. Thanks for the help, Alister!

alisterburt, 2024-05-10:
glad you got things working @asarnow ! We got some useful debug info added out of this 🙂 

For anyone with the same issue: CTF info should look something like this:

```txt
<OptionsCTF>
		<Param Name="Window" Value="512" />
		<Param Name="DefectsPath" Value="" />
		<Param Name="GainTranspose" Value="False" />
		<Param Name="GainFlipY" Value="True" />
		<Param Name="GainFlipX" Value="False" />
		<Param Name="GainHash" Value="" />
		<Param Name="GainPath" Value="gain_ref.mrc" />
		<Param Name="EERGroupFrames" Value="0" />
		<Param Name="BinTimes" Value="0" />
		<Param Name="PixelSize" Value="0.7894" />
		<Param Name="DosePerAngstromFrame" Value="-2.64" />
		<Param Name="GridDims" Value="5,5,1" />
		<Param Name="ZMax" Value="8" />
		<Param Name="ZMin" Value="0.5" />
		<Param Name="UseMovieSum" Value="False" />
		<Param Name="DoPhase" Value="False" />
		<Param Name="Amplitude" Value="0.07" />
		<Param Name="Cc" Value="2.7" />
		<Param Name="Cs" Value="2.7" />
		<Param Name="Voltage" Value="300" />
		<Param Name="RangeMax" Value="0.2255428571428571428571428571" />
		<Param Name="RangeMin" Value="0.0526266666666666666666666667" />
		<Param Name="DefectsHash" Value="" />
		<Param Name="Dimensions" Value="0,0,0" />
	</OptionsCTF>
	<CTF>
		<Param Name="PixelSize" Value="0.7894" />
		<Param Name="ZernikeCoeffsOdd" Value="0.7868362;-10.223117;1.1946881;0.4046562;-5.59376;0.032335937;0;0;0;0;0;0" />
		<Param Name="BeamTilt2" Value="1,0,1" />
		<Param Name="BeamTilt" Value="0,0" />
		<Param Name="PhaseShift" Value="0" />
		<Param Name="Scale" Value="1.0" />
		<Param Name="BfactorAngle" Value="0" />
		<Param Name="BfactorDelta" Value="0" />
		<Param Name="Bfactor" Value="0" />
		<Param Name="ZernikeCoeffsEven" Value="0;0;0;0;0;0;0;0" />
		<Param Name="Amplitude" Value="0.07" />
		<Param Name="DefocusDelta" Value="0.00761689911503086" />
		<Param Name="Defocus" Value="2.552727" />
		<Param Name="Voltage" Value="300.0" />
		<Param Name="Cc" Value="2.7" />
		<Param Name="Cs" Value="2.70000004768372" />
		<Param Name="Distortion" Value="1, 0, 0, 1" />
		<Param Name="PixelSizeAngle" Value="0" />
		<Param Name="PixelSizeDeltaPercent" Value="0" />
		<Param Name="DefocusAngle" Value="23" />
		<Param Name="Thickness" Value="0" />
	</CTF>
	<GridCTF Width="1" Height="1" Depth="41" MarginX="0" MarginY="0" MarginZ="0">
		<Node X="0" Y="0" Z="0" Value="2.6927104" />
		<Node X="0" Y="0" Z="1" Value="2.6928964" />
		<Node X="0" Y="0" Z="2" Value="2.5619066" />
		<Node X="0" Y="0" Z="3" Value="2.7199652" />
		<Node X="0" Y="0" Z="4" Value="2.7546015" />
		<Node X="0" Y="0" Z="5" Value="2.6337123" />
		<Node X="0" Y="0" Z="6" Value="2.670344" />
		<Node X="0" Y="0" Z="7" Value="2.7170398" />
		<Node X="0" Y="0" Z="8" Value="2.6555932" />
		<Node X="0" Y="0" Z="9" Value="2.636291" />
		<Node X="0" Y="0" Z="10" Value="2.6272643" />
		<Node X="0" Y="0" Z="11" Value="2.6261575" />
		<Node X="0" Y="0" Z="12" Value="2.5920126" />
		<Node X="0" Y="0" Z="13" Value="2.6145253" />
		<Node X="0" Y="0" Z="14" Value="2.4135764" />
		<Node X="0" Y="0" Z="15" Value="2.6227539" />
		<Node X="0" Y="0" Z="16" Value="2.6113844" />
		<Node X="0" Y="0" Z="17" Value="2.608053" />
		<Node X="0" Y="0" Z="18" Value="2.579708" />
		<Node X="0" Y="0" Z="19" Value="2.5965438" />
		<Node X="0" Y="0" Z="20" Value="2.476012" />
		<Node X="0" Y="0" Z="21" Value="2.5555718" />
		<Node X="0" Y="0" Z="22" Value="2.5664902" />
		<Node X="0" Y="0" Z="23" Value="2.5881114" />
		<Node X="0" Y="0" Z="24" Value="2.5601254" />
		<Node X="0" Y="0" Z="25" Value="2.5527186" />
		<Node X="0" Y="0" Z="26" Value="2.529058" />
		<Node X="0" Y="0" Z="27" Value="2.530597" />
		<Node X="0" Y="0" Z="28" Value="2.5080166" />
		<Node X="0" Y="0" Z="29" Value="2.523884" />
		<Node X="0" Y="0" Z="30" Value="2.5056248" />
		<Node X="0" Y="0" Z="31" Value="2.4666986" />
		<Node X="0" Y="0" Z="32" Value="2.5176775" />
		<Node X="0" Y="0" Z="33" Value="2.482773" />
		<Node X="0" Y="0" Z="34" Value="2.4424853" />
		<Node X="0" Y="0" Z="35" Value="2.4394794" />
		<Node X="0" Y="0" Z="36" Value="2.538938" />
		<Node X="0" Y="0" Z="37" Value="2.4643047" />
		<Node X="0" Y="0" Z="38" Value="2.4686089" />
		<Node X="0" Y="0" Z="39" Value="2.2161703" />
		<Node X="0" Y="0" Z="40" Value="2.48394" />
	</GridCTF>

```

closed 2024-05-10 by alisterburt