warpem/warp #49: Running MCore from subdirectory hangs infinitely on "Loading population..."

rbs-sci, 2024-05-07:
Minor quality of life request/pseudo-bug.

Certainly non-critical.

I `cd`'d into the`m` directory then accidentally ran `MCore` (for a refinement) from there rather than the project directory; rather than crash or complain it couldn't find what I'd asked it to work on, it just sat on "Loading population..." until I killed it (five or six minutes, my eyes needed a break) - it would be nice for it to throw a message up suggesting the data couldn't be found, as until I checked `top` I thought it was actually doing something.

alisterburt, 2024-05-07:
thanks for the report @rbs-sci ! Agree this would be nice 🙂

dtegunov, 2024-05-11:
Check added. 

(It was a larger bug where any exceptions thrown in DoProcessing were swallowed silently.)

closed 2024-05-11 by dtegunov