warpem/warp #50: compiling with cuda/12.1 gcc-12.2 torch-2.2.2 fails

schloegl, 2024-05-07:
I'm trying to install warp on Debian12 (aka bookworm).  Cuda/11.7 fails to compile with gcc-12.2 (which is the default on debian12); therefore, I'm trying to upgrade the software stack to cuda/12.x which is compatible with gcc-12. 

Initially, the installation works fine, until building "NativeAccceleration", the compilation fails with this error 

```
(20240507) schloegl@gpu136:/dev/shm/schloegl-src-warp-20240507-IFF27k7G/warp/NativeAcceleration/build$ make 
[  0%] Building CUDA object CMakeFiles/NativeAcceleration.dir/gtom/src/Masking/IrregularSphereMask.cu.o
/dev/shm/schloegl-src-warp-20240507-IFF27k7G/warp/NativeAcceleration/gtom/src/Masking/IrregularSphereMask.cu(19): error: texture is not a template
   texture<tfloat, 2, cudaReadModeElementType> texIrregularSphereRadius2d;
   ^

/dev/shm/schloegl-src-warp-20240507-IFF27k7G/warp/NativeAcceleration/gtom/src/Masking/IrregularSphereMask.cu(50): error: identifier "cudaBindTexture2D" is undefined
    cudaBindTexture2D(
    ^

/dev/shm/schloegl-src-warp-20240507-IFF27k7G/warp/NativeAcceleration/gtom/src/Masking/IrregularSphereMask.cu(67): error: identifier "cudaUnbindTexture" is undefined
    cudaUnbindTexture(texIrregularSphereRadius2d);
    ^

/dev/shm/schloegl-src-warp-20240507-IFF27k7G/warp/NativeAcceleration/gtom/src/Masking/IrregularSphereMask.cu(109): error: no instance of overloaded function "tex2D" matches the argument list
            argument types are: (<error-type>, float, float)
     tfloat radius = tex2D(texIrregularSphereRadius2d, phi, theta);
                     ^

4 errors detected in the compilation of "/dev/shm/schloegl-src-warp-20240507-IFF27k7G/warp/NativeAcceleration/gtom/src/Masking/IrregularSphereMask.cu".
make[2]: *** [CMakeFiles/NativeAcceleration.dir/build.make:1122: CMakeFiles/NativeAcceleration.dir/gtom/src/Masking/IrregularSphereMask.cu.o] Error 2
make[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/NativeAcceleration.dir/all] Error 2
make: *** [Makefile:91: all] Error 2

```

I tried with release 2.0.0dev0 as well as with the lasts sources from the git repo (commit 06dd67caf7c554e2feb7d633a61778b6656a1b03) . Both result in the same error.  Can you recommend a way how to overcome this ?

alisterburt, 2024-05-07:
Hey! The code isn't compatible with CUDA 12 and will require some updates before that will work - is there a reason you're trying to build against your own CUDA/conpiler toolchain rather than following the suggested conda based install?

alisterburt, 2024-05-07:
@schloegl given that this isn't supposed to work I'm closing as a non-issue for now! Do let us know if you have any issues with the outlined installation procedure

closed 2024-05-07 by alisterburt

schloegl, 2024-05-07:
Can confirm that I could install it with the suggested conda-based installation procudure. Thanks.