warpem/warp #54: bug: MCore unable to run in fresh env due to `LD_LIBRARY_PATH` being set incorrectly

alisterburt, 2024-05-07:
Trying to run MCore I ran into:

```txt
2024-05-07 18:21:07.301 /lib64/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /home/burta2/software/warptools-dev/repo/Release/linux-x64/publish/libLibTorchSharp.so)
```

Which was weird because `ldd /home/burta2/software/warptools-dev/repo/Release/linux-x64/publish/libLibTorchSharp.so` gave

```txt
libstdc++.so.6 => /home/burta2/mambaforge/envs/warp_build/lib/libstdc++.so.6 (0x0000146aa6b5d000)
```

@dtegunov tracked it down to my `LD_LIBRARY_PATH` pointing at `/home/burta2/mambaforge/envs/warp_build/lib64`

checking that folder revealed...
```txt
(warp_build) [burta2@sc1ns006is08 EMPIAR-10491-5TS]$ ls /home/burta2/mambaforge/envs/warp_build/lib64
ls: cannot access '/home/burta2/mambaforge/envs/warp_build/lib64': No such file or directory`
```

however `/home/burta2/mambaforge/envs/warp_build/lib` was full of nice shared libraries - adding this directory to my `LD_LIBRARY_PATH` fixed the issue and `MCore` runs as expected

Something meant that the wrong thing got added to `LD_LIBRARY_PATH` during environment creation - not sure why

asarnow, 2024-05-14:
Possibly useful, this is what I have in my working module file.

```
if { ![is-loaded cuda/11.7] } {module load cuda/11.7}
if { ![is-loaded imod/4.12.62] } {module load imod/4.12.62}

set warp_root "/home/$USER/local/src/warp/Release/linux-x64/publish"

prepend-path PATH $warp_root
prepend-path LD_LIBRARY_PATH "/home/$USER/local/src/warp/NativeAcceleration/build/lib"
prepend-path LD_LIBRARY_PATH "/home/$USER/local/opt/miniforge3/envs/warp_build/lib"
prepend-path LD_LIBRARY_PATH "/home/$USER/local/opt/miniforge3/envs/warp_build/lib64"
setenv "RELION_EXTERNAL_RECONSTRUCT_EXECUTABLE" $warp_root/"Noise2Half"

unset warp_root
```

alisterburt, 2024-05-21:
turns out its normal that conda doesn't add everything to LD_LIBRARY_PATH, instead shared libraries are correctly linked with a relative path inside the conda env in RPATH at build time... setting LD_LIBARARY_PATH for things we build in the conda env is thus our responsibility and this is expected 🤷🏼‍♂️

closed 2024-05-21 by alisterburt