warpem/warp #55: Problems about MRC file format

Angeevillor, 2024-05-08:
Hello，
It seems that the linux version of Warp writes MRC files in a format problem that causes software like EMAN2,IMOD,Dynamo, etc. to not properly read or display all of Warp's MRC or MRCS formatted image files.
How can I use other software or programs to process these MRC files? Can you give me some advice?
Thanks a lot!

alisterburt, 2024-05-08:
Can you give some concrete examples of errors you see? 

I haven't come across this problem but it's been a while since I tried to do something like this

dtegunov, 2024-05-08:
This is probably due to 16 bit MRCs being the default in Warp now, and old versions of those packages not recognizing the format. What versions are you using? 

If it's a widespread issue, we should introduce an environment variable to make Warp programs less compressive-aggressive.

asarnow, 2024-05-08:
This is the IMOD error. Current IMOD supports Mode 12 though.

```
$ header warp_tiltseries/reconstruction/DA23-3_ts_018_9.96Apx.mrc
ERROR: mrc_head_read - bad file mode 12.
ERROR: iiOpen - warp_tiltseries/reconstruction/DA23-3_ts_018_9.96Apx.mrc has unknown format.

ERROR: iiuOpen - Could not open 'warp_tiltseries/reconstruction/DA23-3_ts_018_9.96Apx.mrc'
```

alisterburt, 2024-05-10:
Okay, so recent versions of IMOD work fine but old versions won't work... I'll add a note to the docs

closed 2024-05-10 by alisterburt

alisterburt, 2024-05-10:
closing as now documented, will open an issue to track whether we gate writing 32 bit behind an environment variable

alisterburt, 2024-05-10:
added option to write out 32 bit files in #63

asarnow, 2024-05-10:
The IMOD changelog was a bit confusing - Mode 12 was added a couple years ago, but only to the beta. You need a 4.12.x version of IMOD, whereas latest stable version 4.11.25 doesn't work. (And current beta 4.12.62 confirmed working).

alisterburt, 2024-05-10:
Good to know - I'm happy with pushing people to IMOD beta releases :)

PR #62 (2024-05-10): add warning in tilt series quick start guide about mrc16b output
[image]

closes #55