warpem/warp #59: Deconvolved tomograms all slices identical

asarnow, 2024-05-09:
The raw tomograms look normal, but the ones in `warp_tiltseries/reconstruction/deconv` appear to be composed entirely of one (central?) slice. They have the expected dimension, but all the slices look the same.

alisterburt, 2024-05-10:
Thanks for the report @asarnow - I'm unable to reproduce right now but this is because of problems in my own dev environment, will have to fix and retry later

asarnow, 2024-05-11:
I think there were some issues in my project, AreTomo2 was used without the compatibility patches before. I'm repeating ts_aretomo through ts_reconstruct now.

dtegunov, 2024-05-11:
Did it end up keeping only the 0 tilt?

asarnow, 2024-05-14:
I've re-run with the AreTomo2 patches and my regular tomograms look good, but the deconvolved ones are still just a single image repeated across all slices.

dtegunov, 2024-05-18:
Fixed in 08510e6

closed 2024-05-18 by dtegunov