warpem/warp #71: ts_export_particles fails with missing particle .star files

asarnow, 2024-05-10:
For example, if one had run every step up to extracting particles, and then provides particles for a subset of tiltseries using `--input_pattern`  (whether by wildcard or removing some matching .star files). The worker fails to connect, there's no error output even with WARP_DEBUG=1. In this case I had moved matching.star files from WARP, for some tiltseries.

stdout

```
TS: DA23-3_ts_021.tomostar   Particles: 910
TS: DA23-3_ts_022.tomostar   Particles: 934
TS: DA23-3_ts_023.tomostar   Particles: 899
TS: DA23-3_ts_024.tomostar   Particles: 909
TS: DA23-3_ts_025.tomostar   Particles: 913
Found 194806 particles in 229 tilt series
Connecting to workers...
```

stderr

```
Unhandled exception. System.TimeoutException: Worker process did not connect within the allotted time.
   at Warp.WorkerWrapper.ListenForPort(String pipeName, Int32 timeoutMilliseconds) in /home/user/local/src/warp/WarpLib/WorkerWrapper.cs:line 158
   at Warp.WorkerWrapper..ctor(Int32 deviceID, Boolean silent, Boolean attachDebugger) in /home/user/local/src/warp/WarpLib/WorkerWrapper.cs:line 90
   at WarpTools.Commands.DistributedOptions.GetWorkers(Boolean attachDebugger) in /home/user/local/src/warp/WarpTools/Commands/DistributedOptions.cs:line 57
   at WarpTools.Commands.ExportParticlesTiltseries.Run(Object options) in /home/user/local/src/warp/WarpTools/Commands/Tiltseries/ExportParticlesTiltseries.cs:line 184
   at WarpTools.WarpTools.Run(Object options) in /home/user/local/src/warp/WarpTools/Program.cs:line 30
   at Warp.Tools.CommandLineParserHelper.ParseAndRun(String[] args, Func`2 run, Type[] verbs, String appName) in /home/user/local/src/warp/WarpLib/Tools/CommandLineParserHelper.cs:line 26
   at WarpTools.WarpTools.Main(String[] args) in /home/user/local/src/warp/WarpTools/Program.cs:line 17
   at WarpTools.WarpTools.<Main>(String[] args)
/var/spool/slurm/d/job1145690/slurm_script: line 46: 12406 Aborted                 WARP_DEBUG=1 WarpTools ts_export_particles --settings warp_tiltseries.settings --input_directory warp_tiltseries/matching --input_pattern "*_clean.star" --normalized_coords --output_star relion/matching.star --output_angpix 4.98 --box 72 --diameter 300 --relative_output_paths --2d
```

alisterburt, 2024-05-10:
looking into this - thanks for the report!

alisterburt, 2024-05-10:
huh - I just removed a single STAR file and everything runs fine:
```sh
WarpTools ts_export_particles \
--settings warp_tiltseries.settings \
--input_directory warp_tiltseries/matching \
--input_pattern "*15854_clean.star" \
--normalized_coords \
--output_star particle_export_test/matching.star \
--output_processing particle_export_test \
--output_angpix 4 \
--box 96 \
--diameter 130 \
--relative_output_paths \
--2d 
```
runs and gives:
```txt
Found 4 files in warp_tiltseries/matching matching *15854_clean.star;
Found 922 particles in 4 tilt series
Connecting to workers...
Connected to 1 workers
1/5, 00:29 remainingno particles found in TS_1.tomostar, skipping...                                                                               
5/5, 00:00 remaining                                                                                                                               
Finished processing in 00:00:26
Saying goodbye to all workers... Done
```

Am I doing something different to your situation?

asarnow, 2024-05-14:
This turned out to be a side effect of tilt series alignment being incomplete/not run. I can use my good TS list at this stage now.

alisterburt, 2024-05-14:
Thanks for coming back! This is good news - I'll close here

closed 2024-05-14 by alisterburt