warpem/warp #73: Nondeterministic failure to run

asarnow, 2024-05-11:
Running ts_ctf and ts_reconstruct on RTX A4000s with 16GB of memory I'm encountering an apparent nondeterminism where it immediately crashes ("worker failed to connect") but runs successfully after several attempts. K3 data with 12x downsampling. Possibly related to debug mode, so far when this happens I've added `export WARP_DEBUG=1` to my SLURM submission script, and then it's worked on the 2nd attempt with debug on. (So > 3 total). I'll do some further testing later and see if I can see any other correlates on the node.

asarnow, 2024-05-11:
I think this is caused by another process using a small amount of GPU memory, which was allowed to run due to some SLURM problem. Not sure if there is ever legitimately too little memory or if there is some kind of hangup. (How long is the timeout for the workers?)

dtegunov, 2024-05-11:
Hi Daniel,

There is a 10 second timeout for the first handshake defined here ("10_000"): https://github.com/warpem/warp/blob/b1cda64f48f25a56cd3a44497bd063b15aa9ccab/WarpLib/WorkerWrapper.cs#L90

Could you please give it a try with higher values? If it solves the problem, I'll make the change permanent.

mbeeby, 2024-05-15:
Hi Dimitry, Daniel,

I confirm I get the same sporadic error running diverse WarpTools commands on SLURM on linux on a 3090 card.

Not sure where to find the 10_000 timeout to try adjusting in my conda install?

dtegunov, 2024-05-15:
Morgan, are you using the pre-built Conda package? You may be one of the first people to do so since it only became available last weekend :-) In that case, you can't modify the code.

Since this is a change with no obvious negative effects, I just went ahead and increased the timeout (a393f22). This should be available as v2.0.0dev6 from Conda in a few minutes, or just `git pull` and rerun `scripts/publish-unix.sh` if you're building from source.

mbeeby, 2024-05-16:
Dimitry - yes, I saw Alister at CCP-EM two weeks ago and he excited me about having a Linux version. So yes, I'm using the Conda package, though I have very little experience with Conda. Should really build myself.

I've not had recurrence of the failure to connect this morning after a few runs.

dtegunov, 2024-05-18:
@asarnow are you still experiencing this issue?

asarnow, 2024-05-19:
I don't think so, it seems like it became less frequent after my SLURM queue issue was fixed, and it doesn't seem to happen at all now. OTOH I haven't re-run the whole pipeline, but I still say close the issue for now.

closed 2024-05-19 by dtegunov