warpem/warp #74: apoferritin map has the wrong handedness

dmichalak, 2024-05-13:
I'm running through the provided script and currently at the steps involving refinement in M. Comparing the map to EMD-11603 and PDB: 5N27, it looks like my map has opposite handedness. This is also the case for the initial refined map output from RELION5.

Does a convention change between RELION versions?

alisterburt, 2024-05-13:
Hey @dmichalak - I just checked and I also got inverted handedness, thanks for flagging this

I don't think this is to do with RELION conventions, it's an error in either
- tilt axis angle (off by 180)
- some 2D image flip somewhere

dtegunov, 2024-05-13:
We should update the script not to flip the tilt angles during import.

PR #75 (2024-05-14): invert tilt angles on import in ts processing example
closes #74
closed 2024-05-14 by alisterburt