warpem/warp #76: Template matching rerun fails

asarnow, 2024-05-16:
I'm trying to re-run template matching and all tomostar files are failing with messages like:

```
Connected to 4 workers
0/375 Failed to process /home/user/Projects/Misc/WARP/warp_tiltseries/DA23-3_pace2_p12_ts_022.tomostar, marked as unselected
1/375, 1 failed, 03:52 remaining
```

Worker crash files are created containing:

```
System.NullReferenceException: Object reference not set to an instance of an object.
   at Warp.TiltSeries.MatchFull(ProcessingOptionsTomoFullMatch options, Image template, Func`4 progressCallback) in /home/asarnow/local/src/warp/WarpLib/TiltSeries.cs:line 1895
   at WarpWorker.WarpWorker.EvaluateCommand(NamedSerializableObject Command) in /home/user/local/src/warp/WarpWorker/WarpWorker.cs:line 625
```

I just rebuilt so I'll go back a few commits, see if it's easy to isolate.

asarnow, 2024-05-16:
I went back to the 10th, seems to be an issue with `--reuse_results`?

alisterburt, 2024-05-17:
I'm able to reproduce @asarnow - running 

```sh
WarpTools ts_template_match \
--settings warp_tiltseries.settings \
--tomo_angpix 10 \
--template_emdb 15854 \
--template_diameter 130 \
--symmetry O \
--perdevice 2 \
--reuse_results 
```

leads to crashes the same as you see which do not happen if `--reuse_results` isn't used

I'll take a look later 🙂

dtegunov, 2024-05-18:
Fixed in c8b254e.

closed 2024-05-18 by dtegunov