warpem/warp #77: cant make template picker to run

rajanprabu, 2024-05-17:
Hi there, 

Greetings from Martinsried.  I have ben trying to get through the EMPIAR-10491_5TS_e2e.sh. It runs well until the template picking. Unfortunately it fails with the following error 

```
Parsing previous results for each item, if available...
5/5, previous metadata found for 5                                                                                                                             
EMD-15854 already exists in /fs/gpfs41/lv11/fileset05/pool/pool-schulman-soft/hpcl8/Warp/test2/warp_tiltseries/template/emd_15854.mrc, skipping download
Setting --template_angpix to 0.7289999 based on template map
Using 1536 orientations for matching
Connecting to workers...
Connected to 2 workers
0/5terminate called after throwing an instance of 'std::runtime_error'
  what():  cuFFT error: CUFFT_INTERNAL_ERROR at /fs/pool/pool-schulman-soft/hpcl8/Warp/warp/NativeAcceleration/gtom/src/FFT/IFFT.cu:24

terminate called after throwing an instance of 'std::runtime_error'
  what():  cuFFT error: CUFFT_INTERNAL_ERROR at /fs/pool/pool-schulman-soft/hpcl8/Warp/warp/NativeAcceleration/gtom/src/FFT/IFFT.cu:24

Failed to process /fs/gpfs41/lv11/fileset05/pool/pool-schulman-soft/hpcl8/Warp/test2/warp_tiltseries/TS_1.tomostar, marked as unselected                       
Failed to process /fs/gpfs41/lv11/fileset05/pool/pool-schulman-soft/hpcl8/Warp/test2/warp_tiltseries/TS_11.tomostar, marked as unselected                      
Unhandled exception. Unhandled exception. System.AggregateException: One or more errors occurred. (Connection refused (localhost:34267))
 ---> System.Net.Http.HttpRequestException: Connection refused (localhost:34267)
 ---> System.Net.Sockets.SocketException (111): Connection refused
   at System.Net.Sockets.Socket.AwaitableSocketAsyncEventArgs.ThrowException(SocketError error, CancellationToken cancellationToken)
   at System.Net.Sockets.Socket.AwaitableSocketAsyncEventArgs.System.Threading.Tasks.Sources.IValueTaskSource.GetResult(Int16 token)
   at System.Net.Sockets.Socket.<ConnectAsync>g__WaitForConnectWithCancellation|285_0(AwaitableSocketAsyncEventArgs saea, ValueTask connectTask, CancellationToken cancellationToken)
   at System.Net.Http.HttpConnectionPool.ConnectToTcpHostAsync(String host, Int32 port, HttpRequestMessage initialRequest, Boolean async, CancellationToken cancellationToken)
   --- End of inner exception stack trace ---
   at System.Net.Http.HttpConnectionPool.ConnectToTcpHostAsync(String host, Int32 port, HttpRequestMessage initialRequest, Boolean async, CancellationToken cancellationToken)
   at System.Net.Http.HttpConnectionPool.ConnectAsync(HttpRequestMessage request, Boolean async, CancellationToken cancellationToken)
   at System.Net.Http.HttpConnectionPool.CreateHttp11ConnectionAsync(HttpRequestMessage request, Boolean async, CancellationToken cancellationToken)
   at System.Net.Http.HttpConnectionPool.AddHttp11ConnectionAsync(QueueItem queueItem)
   at System.Threading.Tasks.TaskCompletionSourceWithCancellation`1.WaitWithCancellationAsync(CancellationToken cancellationToken)
   at System.Net.Http.HttpConnectionPool.SendWithVersionDetectionAndRetryAsync(HttpRequestMessage request, Boolean async, Boolean doRequestAuth, CancellationToken cancellationToken)
   at System.Net.Http.RedirectHandler.SendAsync(HttpRequestMessage request, Boolean async, CancellationToken cancellationToken)
   at System.Net.Http.HttpClient.<SendAsync>g__Core|83_0(HttpRequestMessage request, HttpCompletionOption completionOption, CancellationTokenSource cts, Boolean disposeCts, CancellationTokenSource pendingRequestsCts, CancellationToken originalCancellationToken)
   --- End of inner exception stack trace ---
   at System.Threading.Tasks.Task.ThrowIfExceptional(Boolean includeTaskCanceledExceptions)
   at System.Threading.Tasks.Task`1.GetResultCore(Boolean waitCompletionNotification)
   at Warp.WorkerConsole.SetFileOutput(String path) in /fs/gpfs41/lv11/fileset05/pool/pool-schulman-soft/hpcl8/Warp/warp/WarpLib/WorkerWrapper.cs:line 541
   at WarpTools.Commands.BaseCommand.<>c__DisplayClass1_0.<IterateOverItems>b__0(Int32 iitem, Int32 threadID) in /fs/gpfs41/lv11/fileset05/pool/pool-schulman-soft/hpcl8/Warp/warp/WarpTools/Commands/BaseCommand.cs:line 103
   at Warp.Tools.Helper.<>c__DisplayClass50_0.<ForCPUGreedy>b__0(Object id) in /fs/gpfs41/lv11/fileset05/pool/pool-schulman-soft/hpcl8/Warp/warp/WarpLib/Tools/Helper.cs:line 768
Aborted (core dumped)

```

I have compiled the binary based on the instructions. But result is the same even if I use the condo provided binary. Any help to surpass this step would be very helpful. I am using Quadro RTX 5000 with coda 11.7 as recommended in the instructions. 

Also it while be very helpful to provide the outputs produced from the EMPIAR-10491_5TS_e2e.sh script to compare as well as to skip the steps if necessary like here. Thanks for reading.

alisterburt, 2024-05-17:
@rajanprabu I suspect this is a memory error and that will be more clear in the worker logs in `warp_tiltseries/logs`

You can try `--perdevice 1` to get things working which will limit to one process per GPU and limit memory requirements

alisterburt, 2024-05-17:
> Also it while be very helpful to provide the outputs produced from the EMPIAR-10491_5TS_e2e.sh script to compare as well as to skip the steps if necessary like here. Thanks for reading.

@dtegunov what do you think? Would be simple enough to upload the results to zenodo...

dtegunov, 2024-05-17:
> You can try `--perdevice 1` to get things working which will limit to one process per GPU and limit memory requirements

I'll update the script to --perdevice 1. Template matching already has ~perfect GPU occupancy and running more than 1 on a GPU in parallel leads to lower speed.

> @dtegunov what do you think? Would be simple enough to upload the results to zenodo...

Let's do that!

alisterburt, 2024-05-17:
> I'll update the script to --perdevice 1. Template matching already has ~perfect GPU occupancy and running more than 1 on a GPU in parallel leads to lower speed.

good to know!

> Let's do that!
I'll do a new clean run over the weekend and upload that as there have been a few changes since I last ran it

rajanprabu, 2024-05-17:
@dtegunov  @alisterburt  Thanks for answering. As I mentioned I did run with --perdevice 1.  How big of memory does it need. I ran with Quadro RTX 5000 that has about 16GB memory.

alisterburt, 2024-05-17:
Hi @rajanprabu - you did not mention this, you edited your original comment to add this information then told me you had already mentioned it... your logs mentioned 2 workers and the script uses two workers so I assumed you had followed the script.

As mentioned, you can check the logs in the directory to see if it is indeed a memory error. If it is something else the logs may also be useful

dtegunov, 2024-05-18:
The memory consumption for the test data set was indeed 18 GB per process 😵 aa63b22 reduces it below 8 GB, and adds a --batch_angles parameter to ts_template_match to regulate it.

closed 2024-05-18 by dtegunov

rajanprabu, 2024-05-18:
@dtegunov  Thanks a lot and  it works fine now.  @alisterburt  It was not my intention. I had edited my question twice immediately after I posted it and not after.  Nevertheless I am really sorry for the inconveniences if there is any. 

I will continue to test and get back here.