warpem/warp #78: ts_aretomo poor performance compared to regular aretomo

jmdobbs, 2024-05-17:
Hi, doing testing we've noticed that the warp aretomo performs notably worse at alignment than using our own scripts to deploy aretomo. I suspect this may be to do with the patches: control over this parameter isn't given in warp, and from the logs it appears that it deploys aretomo with 0 x 0 patches. With aretomo we've seen that # patches can have a big impact, even if you're not using the local alignments.

Would it be possible to specify the number of patches in warp? This might improve the alignments substantially. Or it's something else, but this is really the only different thing than our usual workflow (+ using more binned stacks, but even at 6Apx stacks the alignments are bad)

alisterburt, 2024-05-17:
Hi @jmdobbs - thanks for the feedback! This is very interesting, I wasn't aware that the use of patches was/could be decoupled from the local alignment process and am not sure @dtegunov was either.

I'll play with the parameters now and if I see any improvements will expose it ASAP 🙂

dtegunov, 2024-05-17:
Multi-patch being faster and more accurate (without actually doing local alignments) is counterintuitive to me. You're not running it in axis angle search mode, right?

jmdobbs, 2024-05-17:
Multi-patch is -in our experience- certainly slower, but seems to have given much better alignments in the same data. I'll put our scripts below, but the two key differences are: 1. we used unbinned tilts, and 2. we use patches

We have run it both in axis angle search mode and without, either way it comes to the exactly correct value.

> 
> #TOMOLIST=( ${IMODDIR}/TS_011.mrc )  # Single tomogram
> TILTANG=85.2
> PATCHX=6
> PATCHY=8
> ZTHICKNESS=2000
> BINFACTOR=8
> ORIGPXSIZE=1.631
> 
> 
> INPUTLIST=$(pwd)"/input_"$(date +"%Y%m%d%H%M%S")".list"
> ls -d "${TOMOLIST[@]}" | awk -F'/' 'BEGIN{OFS="/"} {if($NF == "") {NF--; print $0,$NF} else {print $0,$NF}}' > $INPUTLIST
> 
> NUMTOMO=$(sed '/^ *$/ d' $INPUTLIST | wc -l)
> OUTPXSIZE=$(echo | awk -v a=$BINFACTOR -v b=$ORIGPXSIZE '{print a*b}')
> 
> mkdir logs
> sbatch --array=0-$(($NUMTOMO-1))%5 --export=INPUTLIST=$INPUTLIST,TILTANG=$TILTANG,PATCHX=$PATCHX,PATCHY=$PATCHY,ZTHICKNESS=$ZTHICKNESS,BINFACTOR=$BINFACTOR,OUTPXSIZE=$OUTPXSIZE /struct/mahamid/Processing/AreTomo/aretomo_array_template_ex.sh
> 
>

alisterburt, 2024-05-17:
I've added patch support in a branch and am testing now - running with patches runs local alignments but we can choose not to use them for downstream processing. The source of the better results might be that it uses the 'smart grid' from MotionCor to ignore bad areas during alignment...

alisterburt, 2024-05-17:
thanks Joe - I'm running the comparison now and the code with a new `--patches` option is in a branch here `at-patch-test` if you'd like to build/run

alisterburt, 2024-05-17:
closed by #79 - will be built/deployed to conda via https://github.com/warpem/warp/actions/runs/9131796028

alisterburt, 2024-05-17:
@jmdobbs I'm signing off for now but please do report back if this changes things for you as expected?

I noticed another difference in the comparison above: you have fiducials in one and not the other - presumably because of boxnet issues - not sure if relevant just noting

jmdobbs, 2024-05-17:
@alisterburt will do. The fiducials are, as you suspected, visible due to boxnet issues, but should not have affected any difference in alignments as aretomo used the tilt stack written out by warp (where masking is not applied, so the fiducials were seen by aretomo in both cases).

dtegunov, 2024-05-17:
Reopening since this is a major issue worth a deeper investigation.

alisterburt, 2024-05-21:
@jmdobbs - could you send the raw data for a single TS which is causing issues across? I'll take a look today 🙂

jmdobbs, 2024-05-21:
@alisterburt I've sent you an email with the link now. Appreciate it, thanks!

alisterburt, 2024-05-21:
You're a star, ta!

alisterburt, 2024-05-21:
@jmdobbs I ran your data myself with the following commands

<details>
<summary>WarpTools commands</summary>
<br>

```sh
# phase plate, pixel size 1.631, gainref flipY

WarpTools create_settings \
--folder_data frames \
--folder_processing warp_frameseries \
--output warp_frameseries.settings \
--extension "*.tif" \

WarpTools create_settings \
--output warp_tiltseries.settings \
--folder_processing warp_tiltseries \
--folder_data tomostar \
--extension "*.tomostar" \
--angpix 1.631 \
--gain_path gain.mrc \
--gain_flip_y \
--exposure 2.99
--tomo_dimensions 4400x6000x1000

WarpTools fs_motion_and_ctf \
--settings warp_frameseries.settings \
--m_grid 1x1x3 \
--c_grid 2x2x1 \
--c_range_max 7 \
--c_defocus_max 8 \
--c_use_sum \
--c_fit_phase \
--out_averages \
--perdevice 4 

WarpTools ts_import \
--mdocs mdoc \
--frameseries warp_frameseries \
--tilt_exposure 2.99 \
--min_intensity 0.3 \
--output tomostar

WarpTools ts_aretomo \
--settings warp_tiltseries.settings \
--alignz 2000 \
--angpix 10 

WarpTools ts_reconstruct \
--settings warp_tiltseries.settings \
--angpix 15 
```
</details>

I get the following reconstruction

https://github.com/warpem/warp/assets/7307488/82006640-327c-4532-adcf-7f68405d0bb7

<details>
<summary>The tomostar looks like this</summary>
<br>
```txt

data_

loop_
_wrpMovieName #1
_wrpAngleTilt #2
_wrpAxisAngle #3
[... 489 lines cut ...]
	</GridAngleY>
	<GridAngleZ Width="1" Height="1" Depth="1" MarginX="0" MarginY="0" MarginZ="0">
		<Node X="0" Y="0" Z="0" Value="0" />
	</GridAngleZ>
	<GridDoseBfacs Width="1" Height="1" Depth="1" MarginX="0" MarginY="0" MarginZ="0">
		<Node X="0" Y="0" Z="0" Value="0" />
	</GridDoseBfacs>
	<GridDoseBfacsDelta Width="1" Height="1" Depth="1" MarginX="0" MarginY="0" MarginZ="0">
		<Node X="0" Y="0" Z="0" Value="0" />
	</GridDoseBfacsDelta>
	<GridDoseBfacsAngle Width="1" Height="1" Depth="1" MarginX="0" MarginY="0" MarginZ="0">
		<Node X="0" Y="0" Z="0" Value="0" />
	</GridDoseBfacsAngle>
	<GridDoseWeights Width="1" Height="1" Depth="1" MarginX="0" MarginY="0" MarginZ="0">
		<Node X="0" Y="0" Z="0" Value="1" />
	</GridDoseWeights>
	<GridLocationBfacs Width="1" Height="1" Depth="1" MarginX="0" MarginY="0" MarginZ="0">
		<Node X="0" Y="0" Z="0" Value="0" />
	</GridLocationBfacs>
	<GridLocationWeights Width="1" Height="1" Depth="1" MarginX="0" MarginY="0" MarginZ="0">
		<Node X="0" Y="0" Z="0" Value="1" />
	</GridLocationWeights>
</TiltSeries>
```
</details>

alisterburt, 2024-05-21:
the reconstruction looks good - I'm not sure what's going wrong in your case. Could you compare files and see if there are obvious differences? Are you still running the patched WarpTools @ThomasHoffmann77 put together which changed the flags to AreTomo?

ThomasHoffmann77, 2024-05-21:
@jmdobbs, for any dev-Tag we have both, the module installed from source and the one based on conda. The latter has a -conda suffix.
Am Di., Mai 21, 2024 21:02 schrieb alisterburt : 
 the reconstruction looks good - I'm not sure what's going wrong in your case. Could you compare files and see if there are obvious differences? Are you still running the patched WarpTools @ThomasHoffmann77 (https://github.com/ThomasHoffmann77) put together which changed the flags to AreTomo?

jmdobbs, 2024-05-21:
@alisterburt passing these parameters exactly still gave me poor alignments using the source built module (with the --OutImod2 warp compatibility patch)

[image]

running the conda module (with --OutImod 1), it worked :)

[image]

I'm very sorry that this was the result of our alteration to the parameters (my bad idea for sure). I should been more careful and recognized that even though outimod 2 is the warp compatibility flag, this is not the same warp!

alisterburt, 2024-05-21:
No worries @jmdobbs - these things happen and I'm glad we got to the bottom of it! Happy processing 🙂

closed 2024-05-21 by alisterburt

dtegunov, 2024-05-22:
Not being able to use the flag the manual says is meant for Warp is also very strange to us 😉

CleoShen, 2025-01-10:
Hi all, I wonder if I've used a standalone AreTomo2 to reconstruct the tomograms with 4x4 patches. Is it possible to use the AreTomo generated .xf files for "WarpTools ts_reconstruct" to make good reconstructions?

jmdobbs, 2025-01-10:
Hi @CleoShen, you can use WarpTools ts_import_alignments to bring them into warp. However, in practice, the patches won't help as the local alignments they provide won't be used. It might be quicker and easier just to rerun aretomo through warp.

CleoShen, 2025-01-10:
Hi @jmdobbs would you mind sharing your one example of using the ts_aretomo? I just ran one time, but only outputs a stack instead of a reconstruction.

jmdobbs, 2025-01-10:
@CleoShen An example command is below, but what you should expect is the following with ts_aretomo:
-it will create the stack aretomo runs on
-it will then run your aretomo executable, aligning the stack and producing an xf file within the tiltstack subfolder for this tiltseries
(and will also produce an aligned stack)
-warp will then import the alignments from the xf, holding them in the xml file

You would not expect that this will directly reconstruct the tomogram, it just does the aligning so that warp can reconstruct later (see the tutorial for an example of this process, but using etomo instead of aretomo).

Example command (which I ran yesterday successfully): WarpTools ts_aretomo --settings settings_tomos.settings --alignz 2000 --angpix 10 --min_fov 0.9

alisterburt, 2025-01-10:
@jmdobbs thanks for providing support here!