warpem/warp #80: Deselect tilt series (not individual tilts)

asarnow, 2024-05-17:
What is the best way to deselect tilt series? I've tried both setting UnselectFilter="True" in the XML files, and moving the XML files to another directory, and it's still counting to the original number of tilt series in the progress info.

alisterburt, 2024-05-17:
Hi Daniel, it's a bit clunky but you can specify one or multiple files with one of the command line flags (can't remember off the top of my head and on mobile) in the warptool for modifying selections

This will only work if done before starting a template matching run, once it is going it will go through each item one by one - the list of items is not updated on the fly in warptools programs

asarnow, 2024-05-17:
You mean the `--input_data` argument for the template matching itself? What should file names be for that? The basenames or the XML or tomostar files?

alisterburt, 2024-05-17:
Either to the template matching or to the WarpTools change_selection program :)Names should be full tomostar namesSent from mobile - apologies for brevity

asarnow, 2024-05-17:
OK got it, thanks! For all the tools, `--input_data` takes e.g. `tomostar/tiltseries.tomostar` and same goes for the .txt file list input. This works (alters the UnselectManual attribute not UnselectFilter).

I think the output is bugged though, running just `--deselect` (so all the files) the output says
[image]
but then running `--select --input_data good_ts.txt` gives
[image]
which is a bit confusing.

asarnow, 2024-05-17:
Also not 100% related but is it safe to run multiple template matching jobs on different subsets of tilt series?

dtegunov, 2024-05-18:
Yes, this should be safe.

alisterburt, 2024-05-18:
closed by #82 - see docs at https://warpem.github.io/warp/reference/warptools/subsets/

jmdobbs, 2024-08-11:
Was there ever any indication about whether there is any bug with UnselectManual? I have noticed recently that it doesn't appear to be respected during processes e.g. tomogram reconstruction, and I see that a few people have also noticed similar behavior. Of course, it could very well be something we're doing wrong!

jowagner91, 2024-08-21:
I ran into the same problem. Change selection will alter the flag UnselectManual in the XML file for a tomo but this will not be considered when running for example ts_reconstruct. Adding input data to every job seems like a somewhat viable workaround.

alisterburt, 2024-08-23:
will check this out properly when I get back from the EMBO workshop! Thanks for raising :-)