warpem/warp #81: Inactivating Tilt series and Tilt Images

frozenfas, 2024-05-18:
First, sorry, as I know this has been asked a few times, I still don't understand the correct way to deactivate a tilt series and/or individual tilt images.

I have written a small script to change the "UnselectManual" attribute and the "UseTilt" tag in the warp_tiltseries/*.xml files but it does not seem to be respected completely.  For example, if I run WarpTools ts_aretomo it runs on all TS regardless if they are marked 'True' in the "UnselectManual" attribute.  If they do run, then they fail to import the results from TS, where I have marked UseTilt to False: 

```
56 active tilts in series, but 44 lines in /XXX/warpem/warp_tiltseries/tiltstack/ts-68/ts-68_Imod/ts-68_st.xf
1/23, 53:02 remaining 
```   
For the error message it seems that it does take into account the "UseTilt" tag when setting up the aretomo job (44 are active and there are 44 in the *.st) but after when parsing the xf file it looks for the 56 tilts.

Do the tomostar files also need to be changed in some way to match the XML files?

I understand from the other messages there are built-in tools for inactivating images. Could you just paste a generic command for using WarpTools to deselect tilt images and I will try to us it?

Thanks for your help and sorry I could not figure it out from the other questions

alisterburt, 2024-05-18:
Hi @frozenfas ! Don't worry about asking questions if things aren't clear - I agree the selection mechanisms are not obvious  enough and this will help us to clarify/document them properly 🙂 

I'll add these to the docs now

Deselecting a tilt image:
```sh
WarpTools change_selection --settings warp_frameseries.settings --deselect --input_data frames/2Dvs3D_53-1_00001_-0.0_Jul31_10.36.03.tif
```

```txt
Running command change_selection with:
select = False
deselect = True
null = False
invert = False
settings = warp_frameseries.settings
input_data = { frames/2Dvs3D_53-1_00001_-0.0_Jul31_10.36.03.tif }
input_data_recursive = False
input_processing = null
output_processing = null

Using alternative input specified by --input_data
File search will be relative to /projects/site/gred/cryoem/burta2/cryo_tomo_test_data/EMPIAR-10491-5TS
1 files found
Parsing previous results for each item, if available...
1/1, previous metadata found for 1                                                                                                                                          
1/1 processed                                                                                                                                                               
Before change: 0 deselected, 0 selected, 1 null
After change: 1 deselected, 0 selected, 0 null
```

Deselecting a tilt series:
```sh
WarpTools change_selection --settings warp_tiltseries.settings --deselect --input_data tomostar/TS_1.tomostar
```

```txt
Running command change_selection with:
select = False
deselect = True
null = False
invert = False
settings = warp_tiltseries.settings
input_data = { tomostar/TS_1.tomostar }
input_data_recursive = False
input_processing = null
output_processing = null

Using alternative input specified by --input_data
File search will be relative to /projects/site/gred/cryoem/burta2/cryo_tomo_test_data/EMPIAR-10491-5TS
1 files found
Parsing previous results for each item, if available...
1/1, previous metadata found for 1                                                                                                                                          
1/1 processed                                                                                                                                                               
Before change: 0 deselected, 1 selected, 0 null
After change: 1 deselected, 0 selected, 0 null
```

alisterburt, 2024-05-18:
The first two lines of the tilt series XML then look like this (key part being `UnselectManual="True"`

```xml
<?xml version="1.0" encoding="utf-8"?>
<TiltSeries DataDirectory="/projects/site/gred/cryoem/burta2/cryo_tomo_test_data/EMPIAR-10491-5TS/tomostar" AreAnglesInverted="True" PlaneNormal="-0.13401411, 0.0035971305, 0.9909729" Bfactor="-7.300099" Weight="1" MagnificationCorrection="1, 0, 0, 1" UnselectFilter="False" UnselectManual="True" CTFResolutionEstimate="4.3">
```

It's possible that there is a bug when trying to import alignment results after deselecting tilt images - I'll try to reproduce your issues there but that'll have to wait until next week

closed 2024-05-18 by alisterburt

alisterburt, 2024-05-18:
Glad @frozenfas ! 🙂

PR #82 (2024-05-18): add docs describing subset selection mechanisms in WarpTools
closes #81 #80 #33