warpem/warp #99: In ts_template_match, specify using deconv tomogram?

dmichalak, 2024-05-21:
Hello,

Is there a way to force `ts_template_match` to use deconvolved reconstructions? I didn't find an option in the command and I don't quite know how to parse the parameters in `warp_tiltseries.settings` yet. 

The output png files from `threshold_picks` shows particle positions with the original tomogram. This may or may not indicate that matching is not being done in the deconvolved tomograms.

Dennis

alisterburt, 2024-05-21:
Hi Dennis, there is no way to specify this as it would not provide any benefit - template matching applies the correct CTF to the template for each chunk (subvolume) of the tomogram. This CTF is the analytically correct one for the center of the subvolume, the same one used during the reconstruction of the original tomograms. 

Running on the deconvolved tomograms would introduce a mismatch between the CTFs and would make the matching strictly worse...

Real space methods like CNNs may perform better on deconvolved tomograms though

dmichalak, 2024-05-21:
Hi Alister,

Ah yes, I was assuming just as enhancing the contrast can improve CNN performance, then it might with template matching. I now see in the paper that template matching is done in Fourier space so it's not quite the same. 

Thanks! 
Dennis