warpem/warp #101: MCore is stuck ?

rajanprabu, 2024-05-23:
Hi there, 

Greetings from Martinsried. I have been testing the EMPIAR script EMPIAR-10491_5TS_e2e.sh provided along with the repo. Everything is going fine until I reach the MCore refinement step. 

```
(warp_build) hpcl8004 Warp/empiar_test_1 $ MCore --population m/10491.population --perdevice_refine 1 --iter 0
``` 

Even when I run the MCore with 0 iteration, this is simply not progressing further and stuck on the command line itself. Not using GPU or CPU. logs folder inside m/refinement_temp/ is empty. I did try many things like checking paths, Cuda etc etc  but nothing is helped so far. 

Please upload the results that you get from  your EMPIAR-10491_5TS_e2e.sh script test run. It would be helpful to compare and find out whats missing.

dtegunov, 2024-05-23:
Hi Rajan,

Is your code up to date? There was a bug in MCore that prevented it from properly printing error messages, which got fixed in c2835b5 2 weeks ago.

@alisterburt that's a +1 for your Zenodo deposition plan!

rajanprabu, 2024-05-24:
Ok.. That fixed it. Thanks @dtegunov

dtegunov, 2024-05-28:
This seems unrelated to the original issue, but let's track it here. How much memory does your GPU have?

alisterburt, 2024-05-29:
I've reproduced these issues on `2.0.0dev8` from conda

1. when launching `MCore` the process appears stuck, `Ctrl + C` appears to release whatever is stuck and it progresses with the following messages 

```txt
Loading population... Done
Creating directories... Done
Spawning workers... 

Done
Preparing for refinement – this will take a few minutes per species
Preparing refinement requisites...
0/1

1/1                                                                                                                            
Performing refinement
Preparing population for data source 10491...Done
Loading gain reference for 10491... Done
Refining all series in data source...
5/5                                                                                                                            
Commiting changes in 10491...Done
Saving intermediate refinement results for 10491...Done
Finishing refinement
Gathering intermediate results, then reconstructing and filtering...
0/1
```

2.  refinement hangs on `Gathering intermediate results, then reconstructing and filtering...`

Investigating...

alisterburt, 2024-05-29:
Same behaviour on a fresh build of 319532f9b3e25f167194551469be02ecbde3f7ee

alisterburt, 2024-05-29:
Went back to `2.0.0dev2` (e1166c00a5faccd2cb414312449b7bb5d5949f02) and I'm not getting the 'hanging' when `MCore` starts - seeing if it runs to completion now then will start debugging properly

alisterburt, 2024-05-29:
confirmed working on `2.0.0dev2` @rajanprabu if you install this version you should be able to move forward until we get a fix in place

alisterburt, 2024-05-29:
c2835b5f0603bdce7add4459091b25ae0c0aa2d7 is the first bad commit

alisterburt, 2024-05-29:
closed by #109 and deploying to conda as `2.0.0dev9` 
c.f. https://github.com/warpem/warp/actions/runs/9293392100

closed 2024-05-29 by alisterburt

xingzhe1129, 2024-06-05:
I updated to the latest version. But my jobs still failed on the Gathering step with the same reason. Do you know why it happened? Thanks.

alisterburt, 2024-06-05:
How much RAM do you have? It's possible you're running out of memory and that's causing the crash

PR #109 (2024-05-29): Revert "Don't miss exceptions from processing"
This reverts commit c2835b5f0603bdce7add4459091b25ae0c0aa2d7 and closes #101 .

Tested and working

```sh
MCore --population m/10491.population --perdevice_refine 4 --iter 0
```
```
Loading population... Done
Creating directories... Done
Spawning workers... Done
Preparing for refinement – this will take a few minutes per species
Preparing refinement requisites...
1/1                                                                                                                            
Performing refinement
Preparing population for data source  [... cut]