warpem/warp #119: Warp(linux version) crashed when processing motion and ctf

Angeevillor, 2024-06-05:
Hello,

I tried to use warp to process the Sars-Cov-2 data from empiar(EMPIAR-10453). I loaded the data and set the settings of warp:
WarpTools create_settings \
--folder_data ${FRAME_SERIES_DATA_DIR} \
--output warp_frameseries.settings \
--folder_processing warp_frameseries \
--extension "*.mrc" \
--angpix 1.329

The data was successfully loaded. And then I conduct the motion correction and ctf estimation of the frames by:
WarpTools fs_motion_and_ctf \
--settings warp_frameseries.settings \
--m_grid 1x1x8 \
--c_grid 2x2x1 \
--c_range_max 7 \
--c_defocus_max 8 \
--c_use_sum \
--out_averages \
--out_average_halves \
--perdevice 4

**Then warp was crashed and I got the error reports just like this:**

System.ArithmeticException: Function does not accept floating point Not-a-Number values.
   at System.Math.Sign(Single value)
   at Warp.Cubic1D.GetPCHIPSlopes(float2[] data, Single[] del) in (warp_path)/warp/WarpLib/Cubic1D.cs:line 153
   at Warp.Cubic1D..ctor(float2[] data) in (warp_path)/warp/WarpLib/Cubic1D.cs:line 34
   at Warp.Cubic1D.Fit(float2[] data, Int32 numnodes) in (warp_path)/warp/WarpLib/Cubic1D.cs:line 186
   at Warp.Movie.ProcessCTF(Image originalStack, ProcessingOptionsMovieCTF options) in /home/taining/programs/warp/WarpLib/Movie.cs:line 1525
   at WarpWorker.WarpWorker.EvaluateCommand(NamedSerializableObject Command) in /home/taining/programs/warp/WarpWorker/WarpWorker.cs:line 263 

Was there something wrong with what I did earlier, or did I miss some step? Can you give me some advice?

Thanks a lot!

Angeevillor, 2024-06-06:
In fact, it starts with the first error, and then continues until it has read all the files. Let's say I have 205 files and it says:
Failed 1/205;
Failed 2/205;
...
Instead of displaying the error on the terminal, it generates a file containing error information such as "worker_xxx_crash.txt".

alisterburt, 2024-06-06:
@Angeevillor I just took a look at this EMPIAR entry and the images are tilt series (stacks of motion corrected 2D images) not multi-frame micrographs (movies).

Warp/WarpTools are currently not compatible with tilt series as inputs so this is the reason your motion correction and ctf estimation is failing.

If you want to process tilt series inputs you will have to 
- split the tilt series into one image per tilt, same name as in mdoc files
- process these tilt images as 1 frame movies using WarpTools

I'm not sure if any images have been removed from the tilt series compared to the mdoc files. Ideally you would set the filenames to be the same as in the mdoc files when splitting

Good luck if you attempt processing this way! I would recommend the Python packages [mrcfile](https://github.com/ccpem/mrcfile) and [mdocfile](https://github.com/teamtomo/mdocfile) to help with scripting the split of tilt series into fake movies

closed 2024-06-06 by alisterburt

alisterburt, 2024-06-06:
@Angeevillor I wrote the Python script for splitting the data in EMPIAR-10453

```python
from pathlib import Path
import mrcfile
import mdocfile

OUTPUT_DIRECTORY = 'movies'
TILT_SERIES_DIRECTORY = 'tiltseries'
MDOC_DIRECTORY = 'mdoc'

def process_tilt_series(mdoc_file: Path, output_directory: Path):
    # load mdoc into pandas dataframe
    df = mdocfile.read(mdoc_file)

    # load tilt series into numpy array
    tilt_series_file = Path(TILT_SERIES_DIRECTORY) / mdoc_file.stem
    tilt_series = mrcfile.read(tilt_series_file)

    # check for same number of tilts
    if len(df) != len(tilt_series):
        raise RuntimeError(f'more tilts in mdoc than tilt series for {tilt_series_file}')

    # split tilt series into individual tilts
    for movie_path, image in zip(df['SubFramePath'], tilt_series):
        output_movie_file = output_directory / Path(movie_path.name)
        mrcfile.write(output_movie_file, image)

if __name__ == '__main__':
    mdocs = list(Path(MDOC_DIRECTORY).glob('*.mdoc'))
    Path(OUTPUT_DIRECTORY).mkdir(exist_ok=True, parents=True)
    for mdoc in mdocs:
        process_tilt_series(mdoc_file=mdoc, output_directory=Path(OUTPUT_DIRECTORY))
```

tree output

```txt
.
├── mdoc
│   └── TS_005.mrc.mdoc
├── movies
│   ├── TS_005_001_-0.0.tif
│   ├── TS_005_002_3.0.tif
│   ├── TS_005_003_6.0.tif
│   ├── TS_005_004_-3.0.tif
│   ├── TS_005_005_-6.0.tif
│   ├── TS_005_006_9.0.tif
│   ├── TS_005_007_12.0.tif
│   ├── TS_005_008_-9.0.tif
│   ├── TS_005_009_-12.0.tif
│   ├── TS_005_010_15.0.tif
│   ├── TS_005_011_18.0.tif
│   ├── TS_005_012_-15.0.tif
│   ├── TS_005_013_-18.0.tif
│   ├── TS_005_014_21.0.tif
│   ├── TS_005_015_24.0.tif
│   ├── TS_005_016_-21.0.tif
│   ├── TS_005_017_-24.0.tif
│   ├── TS_005_018_27.0.tif
│   ├── TS_005_019_30.0.tif
│   ├── TS_005_020_-27.0.tif
│   ├── TS_005_021_-30.0.tif
│   ├── TS_005_022_33.0.tif
│   ├── TS_005_023_36.0.tif
│   ├── TS_005_024_-33.0.tif
│   ├── TS_005_025_-36.0.tif
│   ├── TS_005_026_39.0.tif
│   ├── TS_005_027_42.0.tif
│   ├── TS_005_028_-39.0.tif
│   ├── TS_005_029_-42.0.tif
│   ├── TS_005_030_45.0.tif
│   ├── TS_005_031_48.0.tif
│   ├── TS_005_032_-45.0.tif
│   ├── TS_005_033_-48.0.tif
│   ├── TS_005_034_51.0.tif
│   ├── TS_005_035_54.0.tif
│   ├── TS_005_036_-51.0.tif
│   ├── TS_005_037_-54.0.tif
│   ├── TS_005_038_57.0.tif
│   ├── TS_005_039_60.0.tif
│   ├── TS_005_040_-57.0.tif
│   └── TS_005_041_-60.0.tif
├── tiltseries
│   └── TS_005.mrc
└── warp_splitter.py

```

alisterburt, 2024-06-06:
@Angeevillor I added some documentation in #127 which is now available online at https://warpem.github.io/warp/reference/warptools/processing_tilt_series_stacks/

Angeevillor, 2024-06-06:
Thank you so much for your advice!

I tried to split the dataset by this script, and continue my work, but it seems like I met a new problem juest like before: 

System.Exception: Could not successfully read (data_path)frames/TS_005_033_-48.0.tif within the specified number of attempts.
   at Warp.Headers.MapHeader.ReadFromFilePatient(Int32 attempts, Int32 mswait, String path, int2 headerlessSliceDims, Int64 headerlessOffset, Type headerlessType) in (warp_path)/WarpLib/Headers/Headers.cs:line 127
   at WarpWorker.WarpWorker.LoadAndPrepareStack(String path, Decimal scaleFactor, Int32 maxThreads) in  (warp_path)/warp/WarpWorker/WarpWorker.cs:line 809
   at WarpWorker.WarpWorker.EvaluateCommand(NamedSerializableObject Command) in  (warp_path)/warp/WarpWorker/WarpWorker.cs:line 226

alisterburt, 2024-06-06:
My bad, I wrote the files as MRC but the file names as tif, that's the issue

alisterburt, 2024-06-06:
You'll need to either modify the script to write tif files or modify the image file extensions and the mdoc files

Angeevillor, 2024-06-07:
Yes, I tried to modify the script and write the image as .mrc format just like this:

(project_path)/SARS_warp/warp_frameseries/TS_005_033_-48.0.mrc
(project_path)/SARS_warp/warp_frameseries/TS_005_033_-39.0.mrc
...
Then the data could be opened by imod. So I continued my work, and modified the warp settings:

**WarpTools create_settings --folder_data ${FRAME_SERIES_DATA_DIR} --output warp_frameseries.settings --folder_processing warp_frameseries --extension "*.mrc" --angpix 1.329**

And then conduct the motion and ctf:

**WarpTools fs_motion_and_ctf --settings warp_frameseries.settings --m_grid 1x1x8 --c_grid 2x2x1 --c_range_max 7 --c_defocus_max 8 --c_use_sum --out_averages --out_average_halves --perdevice 4**

And it says:

No alternative input specified, will use input parameters from warp_frameseries.settings
File search will be relative to (project_path)/SARS_warp/frames
10701 files found
Parsing previous results for each item, if available...
10701/10701, previous metadata found for 32
Connecting to workers...
Connected to 8 workers
**Failed to process (project_path)/SARS_warp/warp_frameseries/TS_005_033_-48.0.mrc, marked as unselected
Failed to process (project_path)SARS_warp/warp_frameseries/TS_005_028_-39.0.mrc, marked as unselected**
...
Then warp was crashed just like before , and I got a new error like this:
**System.NullReferenceException: Object reference not set to an instance of an object.
   at Warp.Tools.MathHelper.Diff(float2[] data) in (warp_path)/warp/WarpLib/Tools/MathHelper.cs:line 638
   at Warp.Movie.ProcessShift(Image originalStack, ProcessingOptionsMovieMovement options) in  (warp_path)/warp/WarpLib/Movie.cs:line 2450
   at WarpWorker.WarpWorker.EvaluateCommand(NamedSerializableObject Command) in  (warp_path)/warp/WarpWorker/WarpWorker.cs:line 275**

Angeevillor, 2024-06-07:
Yes, I have completed all the processes of the tutorial data without any errors. Finally, I got a 2.9Å structure , so I want to try another set of data.

alisterburt, 2024-06-07:
Okay good to know - processing tilt series stacks isn't something we natively support so I'd recommend trying something else for now 🙂

Angeevillor, 2024-06-07:
OK, I see. Thanks a lot!