warpem/warp #125: Overwritten logs

rahelwoldeyes, 2024-06-05:
Hi,

When running WarpTools functions, log files in the `warp_tiltseries/logs/` directory are overwritten. Is this intended behavior (a feature) or a bug? Ideally, I'd like the logs to be appended to instead of overwritten. Is there a way to achieve this?

alisterburt, 2024-06-05:
Hi @rahelwoldeyes - nice to see you here! This is the current intended behavior and there is no way for you to configure WarpTools to get the behavior you want... I've opened #126 with a potential solution, will wait and see if Dimitry wants this change before merging

rahelwoldeyes, 2024-06-05:
Hi @alisterburt - Thank you for looking into it!

alisterburt, 2024-06-06:
@rahelwoldeyes after discussing with Dimitry we are planning to
- append to the logs as you suggested
- add additional information about the command being executed (name + parameters)

no timeline on this as I'll be at a workshop next week

rahelwoldeyes, 2024-06-06:
@alisterburt That is great! Thank you!

JadisInka, 2026-09-14:
Hi @alisterburt I was wondering if this was still on your radar? It would be a really useful feature.