warpem/warp #133: Z Dimension Error

LucasDillard, 2024-06-10:
Set Z Dimensions do not match

2024-06-10 18:52:50.809 Accord.DimensionMismatchException: Tomogram dimensions (348, 474, 8) don't match expectation (348, 474, 78) (Parameter 'Array dimensions must match.')
2024-06-10 18:52:50.809    at Warp.TiltSeries.MatchFull(ProcessingOptionsTomoFullMatch options, Image template, Func`4 progressCallback) in /sw/warp/2.0.0dev11/WarpLib/TiltSeries.cs:line 1546
2024-06-10 18:52:50.809    at WarpWorker.WarpWorker.EvaluateCommand(NamedSerializableObject Command) in /sw/warp/2.0.0dev11/WarpWorker/WarpWorker.cs:line 631

alisterburt, 2024-06-11:
Thanks @LucasDillard !

This came up in a workshop at the University of Michigan today. Tomograms were accidentally reconstructed with a way too small z dimension and this led to a failure during template matching

Proposed fix: add a warning if reconstructed tomogram has strangely small number of elements in any dimension

alisterburt, 2024-06-17:
agree, let's warn during settings creation - will add now

PR #150 (2024-06-17): Warn if tomogram dimensions appear small
closes #133
closed 2024-06-17 by alisterburt