warpem/warp #135: Core dump error during tilt alignement

RamteenS, 2024-06-11:
Dear all,

When trying to complete the tutorial, I encounter a core dump error at the tilt alignment step. The error message is as follows:

System.FormatException: The input string '-44.00' was not in a correct format.
   at System.Number.ThrowFormatException[TChar](ReadOnlySpan`1 value)
   at System.Single.Parse(String s)
   at Warp.TiltSeries.InitializeFromTomoStar(Star table) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1717610830978/work/WarpLib/TiltSeries.cs:line 304
   at Warp.TiltSeries..ctor(String path, String dataDirectoryName) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1717610830978/work/WarpLib/TiltSeries.cs:line 286
   at WarpTools.Commands.BaseOptions.<>c__DisplayClass32_0.<Evaluate>b__1(Int32 i, Int32 threadID) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1717610830978/work/WarpTools/Commands/BaseOptions.cs:line 170
   at Warp.Tools.Helper.<>c__DisplayClass49_0.<ForCPU>b__0(Object id) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1717610830978/work/WarpLib/Tools/Helper.cs:line 717

Any ideas on where the problem might be?

Many thanks for your help 

Best 

Ramteen

alisterburt, 2024-06-11:
Hi @RamteenS - could you share your tomostar file here?

RamteenS, 2024-06-11:
Hi alister,
Many thanks for your reply. Here is my tomostar file: 

data_

loop_
_wrpMovieName #1
_wrpAngleTilt #2
_wrpAxisAngle #3
_wrpDose #4
_wrpAverageIntensity #5
_wrpMaskedFraction #6
  ../warp_frameseries/2Dvs3D_53-1_00041_-40.0_Jul31_11.05.12.tif  -38.00  -93.500  105.600006  3.721  0.000
  ../warp_frameseries/2Dvs3D_53-1_00040_-38.0_Jul31_11.04.30.tif  -35.99  -93.500   102.96001  3.693  0.000
  ../warp_frameseries/2Dvs3D_53-1_00037_-36.0_Jul31_11.02.04.tif  -34.00  -93.500       95.04  3.574  0.000
  ../warp_frameseries/2Dvs3D_53-1_00036_-34.0_Jul31_11.01.23.tif  -32.00  -93.500        92.4  3.750  0.000
  ../warp_frameseries/2Dvs3D_53-1_00033_-32.0_Jul31_10.59.02.tif  -30.00  -93.500       84.48  3.754  0.000
  ../warp_frameseries/2Dvs3D_53-1_00032_-30.0_Jul31_10.58.12.tif  -28.00  -93.500   81.840004  3.779  0.000
  ../warp_frameseries/2Dvs3D_53-1_00029_-28.0_Jul31_10.56.01.tif  -26.00  -93.500   73.920006  3.764  0.000
  ../warp_frameseries/2Dvs3D_53-1_00028_-26.0_Jul31_10.55.20.tif  -24.00  -93.500    71.28001  3.553  0.000
  ../warp_frameseries/2Dvs3D_53-1_00025_-24.0_Jul31_10.53.10.tif  -22.00  -93.500       63.36  3.791  0.000
  ../warp_frameseries/2Dvs3D_53-1_00024_-22.0_Jul31_10.52.29.tif  -20.00  -93.500       60.72  3.850  0.000
  ../warp_frameseries/2Dvs3D_53-1_00021_-20.0_Jul31_10.50.20.tif  -18.00  -93.500   52.800003  3.787  0.000
  ../warp_frameseries/2Dvs3D_53-1_00020_-18.0_Jul31_10.49.38.tif  -16.00  -93.500   50.160004  3.742  0.000
  ../warp_frameseries/2Dvs3D_53-1_00017_-16.0_Jul31_10.47.29.tif  -14.00  -93.500       42.24  3.793  0.000
  ../warp_frameseries/2Dvs3D_53-1_00016_-14.0_Jul31_10.46.47.tif  -12.01  -93.500   39.600002  3.801  0.000
  ../warp_frameseries/2Dvs3D_53-1_00013_-12.0_Jul31_10.44.21.tif  -10.01  -93.500       31.68  3.609  0.000
  ../warp_frameseries/2Dvs3D_53-1_00012_-10.0_Jul31_10.43.39.tif   -8.00  -93.500       29.04  3.617  0.000
   ../warp_frameseries/2Dvs3D_53-1_00009_-8.0_Jul31_10.41.34.tif   -6.00  -93.500       21.12  3.812  0.000
   ../warp_frameseries/2Dvs3D_53-1_00008_-6.0_Jul31_10.40.51.tif   -4.00  -93.500   18.480001  3.812  0.000
   ../warp_frameseries/2Dvs3D_53-1_00005_-4.0_Jul31_10.38.47.tif   -2.00  -93.500       10.56  3.730  0.000
   ../warp_frameseries/2Dvs3D_53-1_00004_-2.0_Jul31_10.38.06.tif    0.00  -93.500        7.92  3.836  0.000
   ../warp_frameseries/2Dvs3D_53-1_00001_-0.0_Jul31_10.36.03.tif    1.99  -93.500           0  3.809  0.000
    ../warp_frameseries/2Dvs3D_53-1_00002_2.0_Jul31_10.36.43.tif    4.00  -93.500        2.64  3.807  0.000
    ../warp_frameseries/2Dvs3D_53-1_00003_4.0_Jul31_10.37.23.tif    6.00  -93.500        5.28  3.789  0.000
    ../warp_frameseries/2Dvs3D_53-1_00006_6.0_Jul31_10.39.28.tif    7.99  -93.500   13.200001  3.736  0.000
    ../warp_frameseries/2Dvs3D_53-1_00007_8.0_Jul31_10.40.08.tif   10.00  -93.500       15.84  3.799  0.000
   ../warp_frameseries/2Dvs3D_53-1_00010_10.0_Jul31_10.42.16.tif   11.99  -93.500       23.76  3.818  0.000
   ../warp_frameseries/2Dvs3D_53-1_00011_12.0_Jul31_10.42.56.tif   13.99  -93.500   26.400002  3.764  0.000
   ../warp_frameseries/2Dvs3D_53-1_00014_14.0_Jul31_10.45.03.tif   15.99  -93.500       34.32  3.678  0.000
   ../warp_frameseries/2Dvs3D_53-1_00015_16.0_Jul31_10.45.51.tif   17.99  -93.500   36.960003  3.777  0.000
   ../warp_frameseries/2Dvs3D_53-1_00018_18.0_Jul31_10.48.13.tif   19.99  -93.500       44.88  3.734  0.000
   ../warp_frameseries/2Dvs3D_53-1_00019_20.0_Jul31_10.48.53.tif   21.99  -93.500       47.52  3.781  0.000
   ../warp_frameseries/2Dvs3D_53-1_00022_22.0_Jul31_10.51.04.tif   23.99  -93.500   55.440002  3.559  0.000
   ../warp_frameseries/2Dvs3D_53-1_00023_24.0_Jul31_10.51.44.tif   25.99  -93.500       58.08  3.705  0.000
   ../warp_frameseries/2Dvs3D_53-1_00026_26.0_Jul31_10.53.54.tif   27.99  -93.500          66  3.664  0.000
   ../warp_frameseries/2Dvs3D_53-1_00027_28.0_Jul31_10.54.34.tif   30.00  -93.500       68.64  3.664  0.000
   ../warp_frameseries/2Dvs3D_53-1_00030_30.0_Jul31_10.56.46.tif   32.00  -93.500   76.560005  3.707  0.000
   ../warp_frameseries/2Dvs3D_53-1_00031_32.0_Jul31_10.57.26.tif   33.99  -93.500   79.200005  3.621  0.000
   ../warp_frameseries/2Dvs3D_53-1_00034_34.0_Jul31_10.59.47.tif   35.99  -93.500       87.12  3.713  0.000
   ../warp_frameseries/2Dvs3D_53-1_00035_36.0_Jul31_11.00.36.tif   37.99  -93.500       89.76  3.664  0.000
   ../warp_frameseries/2Dvs3D_53-1_00038_38.0_Jul31_11.03.02.tif   39.99  -93.500       97.68  3.598  0.000
   ../warp_frameseries/2Dvs3D_53-1_00039_40.0_Jul31_11.03.42.tif   41.99  -93.500   100.32001  3.627  0.000

Thanks a lot again for your hep 

Cheers

Ramteen

RamteenS, 2024-06-11:
@alisterburt 
I get this error for several tomostar file and various angles

alisterburt, 2024-06-11:
thanks for the report @RamteenS - we weren't handling locale specific floating point numbers correctly. I assume you're somewhere where the decimal separator used is a comma?

this should be fixed by #136 which will be available on conda as `v2.0.0dev12` once this deployment workflow finishes https://github.com/warpem/warp/actions/runs/9467369167/job/26081236563

RamteenS, 2024-06-12:
Dear Alister,
Thanks again for your help. I still get the following message when running tilt alignment: 
angpix = 10
mask = False
min_fov = 0
initial_axis = -85,6
do_axis_search = False
patch_size = 500
delete_intermediate = False
device_list = {  }
perdevice = 1
workers = {  }
settings = warp_tiltseries.settings
input_data = {  }
input_data_recursive = False
input_processing = null
output_processing = null

No alternative input specified, will use input parameters from warp_tiltseries.settings
File search will be relative to /data/ramteen/subtomo_test_tutorial/Warp_M/tomostar
5 files found
Parsing previous results for each item, if available...
5/5, previous metadata found for 0                                                                                                                                                                                                                
Connecting to workers...
Connected to 4 workers
Performing patch tracking alignment on all tilt-series...
0/5
Failed to import alignments:
The input string ' -40.01' was not in a correct format.
1/5, 00:28 remaining                                                                                                                                                                                                                              
Failed to import alignments:
The input string ' -40.01' was not in a correct format.
2/5, 00:21 remaining                                                                                                                                                                                                                              
Failed to import alignments:
The input string ' -40.01' was not in a correct format.
3/5, 00:14 remaining                                                                                                                                                                                                                              
Failed to import alignments:
The input string ' -40.01' was not in a correct format.
4/5, 00:07 remaining                                                                                                                                                                                                                              
Failed to import alignments:
The input string ' -40.01' was not in a correct format.
5/5, 00:00 remaining                                                                                                                                                                                                                              
Finished processing in 00:00:56
Saying goodbye to all workers... Done

I think there is still an issue ...

Thanks again for your help 

Best,

Ramteen

alisterburt, 2024-06-12:
@RamteenS thanks for getting back here - a few things:

I see a comma in your 'initial axis' entry which is either populated by your settings file or your command line arguments - this is not right, could you provide the commands you used to create settings and run tilt series alignments?

Could you also please post the full traceback to help me track down the new source of the error?

RamteenS, 2024-06-12:
Hi Allister,
Really thanks for your time and help. the command line that i use is:
WarpTools ts_etomo_patches --settings warp_tiltseries.settings --angpix 10 --patch_size 500 --initial_axis -85.6
But when it runs it shows -85,6 as you noticed (i do not have this argument in my settings file). 
Here is the full traceback: 
(warp) [ramteen@022-031-LNX Warp_M]$ WarpTools ts_etomo_patches --settings warp_tiltseries.settings --angpix 10 --patch_size 500 --initial_axis -85.6
Running command ts_etomo_patches with:
angpix = 10
mask = False
min_fov = 0
initial_axis = -85,6
do_axis_search = False
patch_size = 500
delete_intermediate = False
device_list = {  }
perdevice = 1
workers = {  }
settings = warp_tiltseries.settings
input_data = {  }
input_data_recursive = False
input_processing = null
output_processing = null

No alternative input specified, will use input parameters from warp_tiltseries.settings
File search will be relative to /data/ramteen/subtomo_test_tutorial/Warp_M/tomostar
5 files found
Parsing previous results for each item, if available...
5/5, previous metadata found for 5                                                                                                                                                                                                                
Connecting to workers...
Connected to 4 workers
Performing patch tracking alignment on all tilt-series...
0/5Possible XF file paths:
/data/ramteen/subtomo_test_tutorial/Warp_M/warp_tiltseries/tiltstack/TS_11/TS_11.xf
/data/ramteen/subtomo_test_tutorial/Warp_M/warp_tiltseries/tiltstack/TS_11/TS_11.xf
/data/ramteen/subtomo_test_tutorial/Warp_M/warp_tiltseries/tiltstack/TS_11/TS_11_st.xf
/data/ramteen/subtomo_test_tutorial/Warp_M/warp_tiltseries/tiltstack/TS_11/TS_11_st.xf
/data/ramteen/subtomo_test_tutorial/Warp_M/warp_tiltseries/tiltstack/TS_11.xf
/data/ramteen/subtomo_test_tutorial/Warp_M/warp_tiltseries/tiltstack/TS_11.xf
/data/ramteen/subtomo_test_tutorial/Warp_M/warp_tiltseries/tiltstack/TS_11_st.xf
/data/ramteen/subtomo_test_tutorial/Warp_M/warp_tiltseries/tiltstack/TS_11_st.xf
/data/ramteen/subtomo_test_tutorial/Warp_M/warp_tiltseries/tiltstack/TS_11/TS_11_Imod/TS_11.xf
/data/ramteen/subtomo_test_tutorial/Warp_M/warp_tiltseries/tiltstack/TS_11/TS_11_Imod/TS_11.xf
/data/ramteen/subtomo_test_tutorial/Warp_M/warp_tiltseries/tiltstack/TS_11/TS_11_Imod/TS_11_st.xf
/data/ramteen/subtomo_test_tutorial/Warp_M/warp_tiltseries/tiltstack/TS_11/TS_11_Imod/TS_11_st.xf
/data/ramteen/subtomo_test_tutorial/Warp_M/warp_tiltseries/tiltstack/TS_11/TS_11_aligned_Imod/TS_11.xf
/data/ramteen/subtomo_test_tutorial/Warp_M/warp_tiltseries/tiltstack/TS_11/TS_11_aligned_Imod/TS_11.xf
/data/ramteen/subtomo_test_tutorial/Warp_M/warp_tiltseries/tiltstack/TS_11/TS_11_aligned_Imod/TS_11_st.xf
/data/ramteen/subtomo_test_tutorial/Warp_M/warp_tiltseries/tiltstack/TS_11/TS_11_aligned_Imod/TS_11_st.xf
/data/ramteen/subtomo_test_tutorial/Warp_M/warp_tiltseries/tiltstack/TS_11_Imod/TS_11.xf
/data/ramteen/subtomo_test_tutorial/Warp_M/warp_tiltseries/tiltstack/TS_11_Imod/TS_11.xf
/data/ramteen/subtomo_test_tutorial/Warp_M/warp_tiltseries/tiltstack/TS_11_Imod/TS_11_st.xf
/data/ramteen/subtomo_test_tutorial/Warp_M/warp_tiltseries/tiltstack/TS_11_Imod/TS_11_st.xf

Importing 2D transforms from /data/ramteen/subtomo_test_tutorial/Warp_M/warp_tiltseries/tiltstack/TS_11/TS_11.xf

Failed to import alignments:
The input string '  40.01' was not in a correct format.
1/5, 00:30 remainingPossible XF file paths:                                                                                                                                                                                                       
/data/ramteen/subtomo_test_tutorial/Warp_M/warp_tiltseries/tiltstack/TS_1/TS_1.xf
/data/ramteen/subtomo_test_tutorial/Warp_M/warp_tiltseries/tiltstack/TS_1/TS_1.xf
/data/ramteen/subtomo_test_tutorial/Warp_M/warp_tiltseries/tiltstack/TS_1/TS_1_st.xf
/data/ramteen/subtomo_test_tutorial/Warp_M/warp_tiltseries/tiltstack/TS_1/TS_1_st.xf
/data/ramteen/subtomo_test_tutorial/Warp_M/warp_tiltseries/tiltstack/TS_1.xf
/data/ramteen/subtomo_test_tutorial/Warp_M/warp_tiltseries/tiltstack/TS_1.xf
/data/ramteen/subtomo_test_tutorial/Warp_M/warp_tiltseries/tiltstack/TS_1_st.xf
/data/ramteen/subtomo_test_tutorial/Warp_M/warp_tiltseries/tiltstack/TS_1_st.xf
/data/ramteen/subtomo_test_tutorial/Warp_M/warp_tiltseries/tiltstack/TS_1/TS_1_Imod/TS_1.xf
/data/ramteen/subtomo_test_tutorial/Warp_M/warp_tiltseries/tiltstack/TS_1/TS_1_Imod/TS_1.xf
/data/ramteen/subtomo_test_tutorial/Warp_M/warp_tiltseries/tiltstack/TS_1/TS_1_Imod/TS_1_st.xf
/data/ramteen/subtomo_test_tutorial/Warp_M/warp_tiltseries/tiltstack/TS_1/TS_1_Imod/TS_1_st.xf
[... 77 lines cut ...]
/data/ramteen/subtomo_test_tutorial/Warp_M/warp_tiltseries/tiltstack/TS_17/TS_17_Imod/TS_17_st.xf
/data/ramteen/subtomo_test_tutorial/Warp_M/warp_tiltseries/tiltstack/TS_17/TS_17_aligned_Imod/TS_17.xf
/data/ramteen/subtomo_test_tutorial/Warp_M/warp_tiltseries/tiltstack/TS_17/TS_17_aligned_Imod/TS_17.xf
/data/ramteen/subtomo_test_tutorial/Warp_M/warp_tiltseries/tiltstack/TS_17/TS_17_aligned_Imod/TS_17_st.xf
/data/ramteen/subtomo_test_tutorial/Warp_M/warp_tiltseries/tiltstack/TS_17/TS_17_aligned_Imod/TS_17_st.xf
/data/ramteen/subtomo_test_tutorial/Warp_M/warp_tiltseries/tiltstack/TS_17_Imod/TS_17.xf
/data/ramteen/subtomo_test_tutorial/Warp_M/warp_tiltseries/tiltstack/TS_17_Imod/TS_17.xf
[... cut ...]
bash : ligne 1 : 1235015 Killed                  /home/ramteen/.conda/envs/warp/bin/WarpWorker -d 1 -p 0 --pipe WarpWorkerPipe_2_1234831_1952147956 > worker0.out 2> worker0.err
bash : ligne 1 : 1235184 Killed                  /home/ramteen/.conda/envs/warp/bin/WarpWorker -d 2 -p 0 --pipe WarpWorkerPipe_3_1234831_835336495 > worker0.out 2> worker0.err
 Done
bash : ligne 1 : 1235340 Killed                  /home/ramteen/.conda/envs/warp/bin/WarpWorker -d 3 -p 0 --pipe WarpWorkerPipe_4_1234831_708987416 > worker0.out 2> worker0.err

Thanks a lot again 

Ramteen

alisterburt, 2024-06-13:
@RamteenS should be fixed by #144 which is deploying as v2.0.0dev13 now
https://github.com/warpem/warp/actions/runs/9507281986

closed 2024-06-13 by alisterburt

alisterburt, 2024-06-14:
Hi @RamteenS 

These issues are separate from the last one, in future please open a new issue if something else comes up

Two things:
- Your tilt series alignment has completely failed - your tomograms have no useful contents in them
- you should not have to mess with --input_processing in the way you have, you've done something

I'd recommend starting all tilt series processing again from scratch, if you still run into issues don't hesitate to open a new issue

Please also see

https://warpem.github.io/warp/reference/warptools/advanced_data_flow/?h=ad

RamteenS, 2024-06-14:
Thank a lot Alister, i will reprocess again from scratch and if it does not work i will open an new issue. 

Thanks again for all your help

Best 

Ramteen

alisterburt, 2024-06-14:
Thanks @RamteenS - it's likely an issue caused by the settings creation issue you mentioned which I will fix so don't be surprised/disappointed if it doesn't work out - we've got bugs to fix 🤠

RamteenS, 2024-06-14:
No problem, I am pretty sure it will work smoothly in the end :)

Best,

Ramteen

PR #136 (2024-06-11): parse wrpAngleTilt and wrpDose without respecting locale
fixes #135
PR #144 (2024-06-13): add missing invariant culture spec in float parsing
closes #135