warpem/warp #140: Error during tilt alignement

RamteenS, 2024-06-12:
Dear Alister,
Thanks again for your help. I still get the following message when running tilt alignment:
angpix = 10
mask = False
min_fov = 0
initial_axis = -85,6
do_axis_search = False
patch_size = 500
delete_intermediate = False
device_list = { }
perdevice = 1
workers = { }
settings = warp_tiltseries.settings
input_data = { }
input_data_recursive = False
input_processing = null
output_processing = null

No alternative input specified, will use input parameters from warp_tiltseries.settings
File search will be relative to /data/ramteen/subtomo_test_tutorial/Warp_M/tomostar
5 files found
Parsing previous results for each item, if available...
5/5, previous metadata found for 0
Connecting to workers...
Connected to 4 workers
Performing patch tracking alignment on all tilt-series...
0/5
Failed to import alignments:
The input string ' -40.01' was not in a correct format.
1/5, 00:28 remaining
Failed to import alignments:
The input string ' -40.01' was not in a correct format.
2/5, 00:21 remaining
Failed to import alignments:
The input string ' -40.01' was not in a correct format.
3/5, 00:14 remaining
Failed to import alignments:
The input string ' -40.01' was not in a correct format.
4/5, 00:07 remaining
Failed to import alignments:
The input string ' -40.01' was not in a correct format.
5/5, 00:00 remaining
Finished processing in 00:00:56
Saying goodbye to all workers... Done

I think there is still an issue ... When doing the reconstruction, the output file does not contain the tomogramm

Thanks again for your help

Best,

Ramteen

RamteenS, 2024-06-12:
Here is what i get after tomogram reconstruction: 
[image]

Cheers 

Ramteen

alisterburt, 2024-06-12:
Hi @RamteenS - I've reopened #135 to continue discussions

It's normal that your tomograms don't reconstruct properly if your alignments could not be imported 🙂 

We'll figure this out, appreciate the patience! You seem to be the first user who has commas as decimal separators on Linux and it's likely this will cause a few more little bugs we will have to fix

alisterburt, 2024-06-12:
Closing here to continue discussion in #135

closed 2024-06-12 by alisterburt