warpem/warp #143: problem with filename extraction from SubFramePath in `ts_import`

alisterburt, 2024-06-13:
Someone called Pankaj on the mailing list has found a bug that I've tracked down to a likely issue with parsing file names from `SubFramePath` entries

https://github.com/warpem/warp/blob/0549286e3c4389136aed0f790d66420714b7c6a8/WarpTools/Commands/Tiltseries/ImportTiltseries.cs#L195-L197

context:
error:
```txt
Failed to parse TS_001.mrc.mdoc: At least one of the referenced frame series could not be found:
```

generated here:
https://github.com/warpem/warp/blob/0549286e3c4389136aed0f790d66420714b7c6a8/WarpTools/Commands/Tiltseries/ImportTiltseries.cs#L226-L227

the fact that no entry name was printed implies the parsing failed

alisterburt, 2024-06-13:
test program runs as expected so problem is likely elsewhere
```txt
using System;
					
public class Program
{
	public static void Main()
	{
		string line = "SubFramePath = ../frames/2024-04-14_17.57.51_TS_001_-51.0.tif";
		string[] parts = line.Split(new[] { " = " }, StringSplitOptions.RemoveEmptyEntries);
		foreach (string p in parts)
			Console.WriteLine(p);
	}
}
```

closed 2024-06-13 by alisterburt

pankajumich, 2024-06-14:
Hi Alister,
here is the mdoc file 👍:

PixelSpacing = 1.667
Voltage = 300
Version = SerialEM Version 4.1.2 64-bit,  built Jan 12 2024  04:15:29
ImageFile = 2024-04-14_17.57.51_TS_001_.tif
ImageSize = 5760 4092
DataMode = 1

[T = SerialEM: Titan Krios D3877 at LSI UMich                14-Apr-24  17:57:58    ]

[T =     Tilt axis angle = 84.8, binning = 1  spot = 5  camera = 1 dosym = 12.0]

[ZValue = 0]
MinMaxMean = -678 2564 36.8651
TiltAngle = -51.0102
StagePosition = -509.592 -2.79361
StageZ = 121.733
Magnification = 53000
Intensity = 0.190106
ExposureDose = 3.31076
DoseRate = 8.86912
PixelSpacing = 1.667
SpotSize = 5
ProbeMode = 0
Defocus = -1.47981
ImageShift = -0.321156 0.557904
RotationAngle = 174.78
ExposureTime = 0.30759
Binning = 1
CameraIndex = 1
DividedBy2 = 1
OperatingMode = 1
UsingCDS = 0
MagIndex = 28
LowDoseConSet = 4
CountsPerElectron = 16
TargetDefocus = -3.5
PriorRecordDose = 112.566
SubFramePath = ../frames/2024-04-14_17.57.51_TS_001_-51.0.tif
NumSubFrames = 11
FrameDosesAndNumber = 0.30098 11
DateTime = 14-Apr-2024  17:57:58
NavigatorLabel = 53
FilterSlitAndLoss = 20 0
UncroppedSize = -5760 -4092
RotationAndFlip = 4
TimeStamp = 135277078

[ZValue = 1]
MinMaxMean = -761 2802 40.6728
TiltAngle = -48.0091
StagePosition = -509.59 -2.79151
StageZ = 121.733
Magnification = 53000
Intensity = 0.190106
ExposureDose = 3.31076
DoseRate = 9.87624
PixelSpacing = 1.667
SpotSize = 5
ProbeMode = 0
Defocus = -1.57265
ImageShift = -0.301678 0.54859
RotationAngle = 174.78
ExposureTime = 0.30759
Binning = 1
CameraIndex = 1
DividedBy2 = 1
OperatingMode = 1
[... 1178 lines cut ...]
SpotSize = 5
ProbeMode = 0
Defocus = -2.26772
ImageShift = 0.128038 0.0634959
RotationAngle = 174.78
ExposureTime = 0.30759
Binning = 1
CameraIndex = 1
DividedBy2 = 1
OperatingMode = 1
UsingCDS = 0
MagIndex = 28
LowDoseConSet = 4
CountsPerElectron = 16
TargetDefocus = -3.5
PriorRecordDose = 82.7691
SubFramePath = ../frames/2024-04-14_17.54.17_TS_001_51.0.tif
NumSubFrames = 11
FrameDosesAndNumber = 0.30098 11
DateTime = 14-Apr-2024  17:54:24
NavigatorLabel = 53
FilterSlitAndLoss = 20 0
UncroppedSize = -5760 -4092
RotationAndFlip = 4
TimeStamp = 135276864

alisterburt, 2024-06-14:
@pankajumich all looks good there - could you please share with me all of the frames from a tilt series and the corresponding mdoc so I can debug? If you don't have an easy way to share something like wetransfer should work well

Fengyun0101, 2024-06-14:
Hi Alister @alisterburt ,

I came up with the similar issue of  "Failed to parse Position_1.mdoc: At least one of the referenced frame series could not be found: Position_1_001_0.00_20230927_180658_Fractions.mrc". I uploaded a dataset for your debug. Thanks
https://www.dropbox.com/t/nKL70f33sKtMlL6o

Best,
Feng

alisterburt, 2024-06-14:
thanks @Fengyun0101 - downloading!

pankajumich, 2024-06-14:
> @pankajumich all looks good there - could you please share with me all of the frames from a tilt series and the corresponding mdoc so I can debug? If you don't have an easy way to share something like wetransfer should work well

Hi @alisterburt here is the link for the data;[
](https://www.dropbox.com/scl/fo/0cdkl7oqw6de0ics8oc1r/ABhRFBUy6JudONAjZfzOTMk?rlkey=8e5uknrwd4szeis2uv8lj4tgt&dl=0)

pankajumich, 2024-06-14:
> @pankajumich all looks good there - could you please share with me all of the frames from a tilt series and the corresponding mdoc so I can debug? If you don't have an easy way to share something like wetransfer should work well

Hi @alisterburt, here is the link to the data:
https://www.dropbox.com/scl/fo/0cdkl7oqw6de0ics8oc1r/ABhRFBUy6JudONAjZfzOTMk?rlkey=8e5uknrwd4szeis2uv8lj4tgt&dl=0

alisterburt, 2024-06-14:
thanks @pankajumich - downloading

alisterburt, 2024-06-17:
@Fengyun0101 I just checked your data - the filenames in your mdoc don't match the filenames in your frames directory

```
Failed to parse Position_1.mdoc: At least one of the referenced frame series could not be found: Position_1_001_0.00_20230927_180658_Fractions.mrc
```

```
$ ls frames/Position_1_001_0.00_20230927_180658_Fractions.mrc
ls: cannot access 'frames/Position_1_001_0.00_20230927_180658_Fractions.mrc': No such file or directory
```

The frames directory contains files where `Fractions` has a lowercase F
```
Position_1_001_0.00_20230927_180658_fractions.mrc
```

You can correct the names in your mdoc file with the following bash script (assuming they are all in a folder called `mdoc`)

```bash
mkdir new_mdoc
for f in mdoc/*.mdoc; do cat $f | sed 's|_Fractions|_fractions|g' > new_mdoc/${f##*/}; done
```

alisterburt, 2024-06-17:
Hi @pankajumich - the timestamps in filenames in your frames directory don't match the timestamps in filenames in your mdoc files

from your mdoc `2024-04-14_17.57.51_TS_001_-48.0.tif` vs the movie file `frames/2024-04-14_17.57.27_TS_001_-48.0.tif`
you'll have to modify your mdocs so the filenames match

pankajumich, 2024-06-19:
Thanks @Alister Burt ***@***.***>