warpem/warp #145: Handedness flip/unflipping has no effect (?)

shahpnmlab, 2024-06-14:
I am report ing this issue whilst using WarpTool binaries built in commit  57c0b3cd.  I am observing a rather curious pathology. After running TS ctf with --auto_hand 10 (there are 10 TS in the projDir), Warp detects that the defocus handedness is flipped, unflips the hand and re-estimates the 3D CTF params. _However_  when I ran ```ts_defocus_hand``` --check, the sub-program still reports an inverted defocus handedness. When I then rerun the command with --set_flip and check again, the defocus handedness continues to be reported with a negative correlation score. This is quite confusing and I am not sure if the case is that the internally warp is doing the flipping and if there is just a bug in the STDOUT message printing logic that does not take into account if the hand has been flipped and continues to report summat stale results. Below is output for the ``ts_ctf`` and handedness checking outputs.

```
⁠ Running command ts_ctf with:
window = 512
range_low = 30
range_high = 7
defocus_min = 0.5
defocus_max = 10
voltage = 300
cs = 2.7
amplitude = 0.07
fit_phase = False
auto_hand = 10
device_list = {  }
perdevice = 1
workers = {  }
settings = warp_tiltseries.settings
input_data = {  }
input_data_recursive = False
input_processing = null
output_processing = null

No alternative input specified, will use input parameters from warp_tiltseries.settings
File search will be relative to /gpfs3/well/stuart/users/vho505/Research_projects/RotavirusGenome/WARP_sub/tomostar
10 files found
Parsing previous results for each item, if available...
10/10, previous metadata found for 0                                                                                                            
Connecting to workers...
Connected to 4 workers
Estimating the CTF for 10 tilt series to check handedness...
10/10, 00:00 remaining                                                                                                                          
Finished processing in 00:01:02
Checking defocus handedness for all tilt series...
10/10, -0.783                                                                                                                                   
Average correlation: -0.783
The average correlation is negative, which means that the defocus handedness needs to be flipped compared to its current state
Flipping the defocus handedness for all tilt series...
10/10                                                                                                                                           
Now running CTF estimation for all tilt series with correct defocus handedness...
10/10, 00:00 remaining                                                                                                                          
Finished processing in 00:00:59
Saying goodbye to all workers... Done
Saving settings... Done

Running command ts_defocus_hand with:
check = True
set_auto = False
set_flip = False
set_noflip = False
set_switch = False
settings = warp_tiltseries.settings
input_data = {  }
input_data_recursive = False
input_processing = null
output_processing = null

No alternative input specified, will use input parameters from warp_tiltseries.settings
File search will be relative to /gpfs3/well/stuart/users/vho505/Research_projects/RotavirusGenome/WARP_sub/tomostar
10 files found
Parsing previous results for each item, if available...
10/10, previous metadata found for 10                                                                                                           
Checking defocus handedness for all tilt series...
10/10, -0.783                                                                                                                                   
Average correlation: -0.783
The average correlation is negative, which means that the defocus handedness should be set to 'flip ⁠
```
Let me know if this has been fixed in later commits and I will update and close the issue.

dtegunov, 2024-06-14:
Hi Pranav, 

This looks like the intended behavior. ts_defocus_hand checks what the absolute state of the flip flag should be, not whether it should be flipped relative to its current state. So previous flipping will have no effect on what ts_defocus_hand reports, and manually setting it to flip will have no effect because it's already set to flip by the previous operation.

alisterburt, 2024-06-14:
I think what I'm saying is correct but will check when I get to work

the mechanism is different from warp desktop, it reports the correlation you get without taking into account the flag and tells you which way to set the flag accordingly

So what you are seeing isn't unexpected, does that make sense?

closed 2024-06-14 by alisterburt

shahpnmlab, 2024-06-14:
Thanks for the clarification. I must admit though that it can be a bit misleading, perhaps a message can be printed, in the case when the invert_hand field and the corelation values are anti-correlated,  so that users arent left with scratching their heads?