warpem/warp #146: Locale issues when creating settings

alisterburt, 2024-06-14:
Reported by @RamteenS

> I noticed by the way that, at the beginning of the processing when creating settings files, i need to put coma for the pixel size otherwise it does not create the settings file:

alisterburt, 2024-06-14:
I can't see any remaining `float.Parse()` calls which do not have `CultureInfo.InvariantCulture` specified so I can't see why your settings creation would fail. @RamteenS could you paste the traceback?

```py
In [9]: for f in files:
   ...:     print(f'looking at {f}')
   ...:     f = open(f, mode='r')
   ...:     lines = f.readlines()
   ...:     for idx, line in enumerate(lines):
   ...:         if ('float.Parse' in line) and not ('CultureInfo.InvariantCulture' in line):
   ...:             print(f'{idx}: {line}')
   ...:     f.close()

```

```txt
looking at MrcConverter/Options.cs
looking at MrcConverter/MrcConverter.cs
looking at MrcConverter/obj/Debug/.NETCoreApp,Version=v8.0.AssemblyAttributes.cs
looking at MrcConverter/obj/Debug/MrcConverter.AssemblyInfo.cs
looking at Noise2Half/Noise2Half.cs
looking at Noise2Half/obj/Debug/.NETCoreApp,Version=v8.0.AssemblyAttributes.cs
looking at Noise2Half/obj/Debug/Noise2Half.AssemblyInfo.cs
looking at WarpWorker/REST.cs
looking at WarpWorker/OptionsCLI.cs
looking at WarpWorker/WarpWorker.cs
looking at WarpWorker/obj/Debug/.NETCoreApp,Version=v8.0.AssemblyAttributes.cs
looking at WarpWorker/obj/Debug/WarpWorker.AssemblyInfo.cs
looking at WarpWorker/API/Service.cs
looking at WarpWorker/API/Console.cs
looking at MCore/OptionsCLI.cs
looking at MCore/MCore.cs
looking at MCore/obj/Debug/.NETCoreApp,Version=v8.0.AssemblyAttributes.cs
looking at MCore/obj/Debug/MCore.AssemblyInfo.cs
looking at Noise2Map/Options.cs
looking at Noise2Map/Noise2Map.cs
looking at Noise2Map/obj/Debug/.NETCoreApp,Version=v8.0.AssemblyAttributes.cs
looking at Noise2Map/obj/Debug/Noise2Map.AssemblyInfo.cs
looking at WarpLib/TiltSeries.cs
looking at WarpLib/WorkerWrapper.cs
looking at WarpLib/CPU.cs
looking at WarpLib/Star.cs
779:             return (name1 == null ? GetColumn(0) : GetColumn(name1)).Select(v => float.Parse(v.Replace("inf", "Infinity").

looking at WarpLib/OptionsWarp.cs
looking at WarpLib/SoftMap.cs
looking at WarpLib/LinearGrid4D.cs
looking at WarpLib/Cubic1D.cs
looking at WarpLib/CTF.cs
looking at WarpLib/WarpBase.cs
looking at WarpLib/GPU.cs
looking at WarpLib/FFTW.cs
looking at WarpLib/ProcessingOptions.cs
looking at WarpLib/Image.cs
looking at WarpLib/CubicGrid.cs
looking at WarpLib/Cubic1DShort.cs
looking at WarpLib/Projector.cs
looking at WarpLib/NMAMap.cs
201:                             float3 AtomPos = new float3(float.Parse(Line.Substring(30, 8)),

202:                                                         float.Parse(Line.Substring(38, 8)),

203:                                                         float.Parse(Line.Substring(46, 8))) / pixelSize;

looking at WarpLib/Movie.cs
looking at WarpLib/FSC.cs
looking at WarpLib/obj/Debug/.NETCoreApp,Version=v8.0.AssemblyAttributes.cs
looking at WarpLib/obj/Debug/WarpLib.AssemblyInfo.cs
looking at WarpLib/NNModels/C2DNet.cs
looking at WarpLib/NNModels/TomoNet.cs
looking at WarpLib/NNModels/TildeNet.cs
[... 264 lines cut ...]
looking at MTools/obj/Debug/MTools.AssemblyInfo.cs
looking at MTools/Commands/ListSpecies.cs
looking at MTools/Commands/ResampleParticleTrajectories.cs
looking at MTools/Commands/RemoveSpecies.cs
looking at MTools/Commands/UpdateMask.cs
looking at MTools/Commands/CreatePopulation.cs
looking at MTools/Commands/CreateSpecies.cs
414:                         float DetectorPixel = float.Parse(TableRelion.GetRowValue(0, "rlnDetectorPixelSize")) * 1e4f;

415:                         float Mag = float.Parse(TableRelion.GetRowValue(0, "rlnMagnification"));

428:                     AngPixCoords = float.Parse(TableRelion.GetRowValue(0, "rlnImagePixelSize"));

432:                     AngPixCoords = float.Parse(TableOptics.GetRowValue(0, "rlnImagePixelSize"));

looking at MTools/Commands/ShiftSpecies.cs
looking at MTools/Commands/AddSource.cs
looking at MTools/Commands/ListSources.cs
looking at MTools/Commands/BaseCommand.cs
looking at MTools/Commands/CreateSource.cs
looking at MTools/Commands/RotateSpecies.cs
looking at MTools/Commands/RemoveSource.cs
looking at MTools/Commands/ExpandSymmetry.cs

```

RamteenS, 2024-06-17:
Hi Alister,
When creating the settings files (same for both frameseries and tiltseries) i get the following core dump error if i put a dot  instead of coma for the pixel size: 
(warp) [ramteen@022-031-LNX Warp_M]$ WarpTools create_settings \
> --folder_data frames \
> --folder_processing warp_frameseries \
> --output warp_frameseries.settings \
> --extension "*.tif" \
> --angpix 0.7894 \
> --gain_path gain_ref.mrc \
> --gain_flip_y \
> --exposure 2.64
Running command create_settings with:
output = warp_frameseries.settings
folder_processing = warp_frameseries
folder_data = frames
recursive = False
extension = *.tif
angpix = 0.7894
bin = null
bin_angpix = null
gain_path = gain_ref.mrc
defects_path = null
gain_flip_x = False
gain_flip_y = True
gain_transpose = False
exposure = 2,64
eer_ngroups = 40
eer_groupexposure = null
tomo_dimensions = null

Pixel size is not a number, trying to read from file instead...
Unhandled exception. System.Exception: No files found matching 0.7894
   at WarpTools.Commands.CreateSettings.Run(Object options) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1718315900454/work/WarpTools/Commands/CreateSettings.cs:line 117
   at WarpTools.WarpTools.Run(Object options) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1718315900454/work/WarpTools/Program.cs:line 30
   at Warp.Tools.CommandLineParserHelper.ParseAndRun(String[] args, Func`2 run, Type[] verbs, String appName) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1718315900454/work/WarpLib/Tools/CommandLineParserHelper.cs:line 26
   at WarpTools.WarpTools.Main(String[] args) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1718315900454/work/WarpTools/Program.cs:line 17
   at WarpTools.WarpTools.<Main>(String[] args)
Aborted (core dumped)

Thanks for your help again 

Best 

Ramteen

dtegunov, 2024-06-17:
It was hiding in a decimal.Parse rather than float.Parse. I went through the whole solution and fixed all instances of decimal and double parsing. Should be fixed in e237148.

closed 2024-06-17 by dtegunov

alisterburt, 2024-06-17:
pushed as `v2.0.0dev14`
https://github.com/warpem/warp/actions/runs/9554134582/job/26334528289

RamteenS, 2024-06-18:
Thanks a lot, both of you. This issue has been fixed. Nevertheless, I have another problem now during tomogram reconstruction. I will open a new issue!

Thank again 

Best 

Ramteen