warpem/warp #148: NotImplementedException in M refinement

asarnow, 2024-06-16:
I'm seeing this NotImplementedException when I try to refine a new population/species created from a Relion run_data.star. I'm not sure what I'm missing, it was working before...

```
Unhandled exception. Unhandled exception. Unhandled exception. Unhandled exception. System.NotImplementedException: The method or operation is not implemented.
   at MCore.MCore.WorkerDied(Object sender, EventArgs e) in /home/user/local/src/warp/MCore/MCore.cs:line 597
   at Warp.WorkerWrapper.ReportDeath() in /home/user/local/src/warp/WarpLib/WorkerWrapper.cs:line 246
   at Warp.WorkerWrapper.<StartHeartbeat>b__14_0() in /home/user/local/src/warp/WarpLib/WorkerWrapper.cs:line 180
System.NotImplementedException: The method or operation is not implemented.
```

My commands:

```
MTools create_population \
    --directory m \
    --name $pop

MTools create_source \
    --name $pop \
    --population m/${pop}.population \
    --processing_settings warp_tiltseries.settings

MTools create_species \
    --population m/${pop}.population \
    --name $spec \
    --diameter 300 \
    --sym C3 \
    --temporal_samples 1 \
    --half1 relion/Refine3D/job006/run_half1_class001_unfil.mrc \
    --half2 relion/Refine3D/job006/run_half2_class001_unfil.mrc \
    --mask m/mask_4apx_e5.mrc \
    --particles_relion relion/Refine3D/job006/run_data.star \
    --angpix_resample 1.245 \
    --lowpass 15

MCore \
    --population m/${pop}.population \
    --refine_imagewarp 6x4 \
    --refine_particles \
    --ctf_defocus \
    --ctf_defocusexhaustive \
    --perdevice_refine 2
```

The particles for the Refine3D were created by ts_export_particles. Refinement output before the error:

```
Loading population... Done
Creating directories... Done
Spawning workers... Done
Preparing for refinement – this will take a few minutes per species
Preparing refinement requisites...
0/1                                                                              1/1
Performing refinement
Preparing population for data source oc43-s-m20...Done
Loading gain reference for oc43-s-m20... Done
Refining all series in data source..
... runs though all TS ...
Commiting changes in oc43-s-m20...
```

dtegunov, 2024-06-16:
Hi Daniel,

My guess would be the worker running out of memory in the final step. What GPU is it running on? The box size is quite large at 420 px. If a larger pixel size would be enough to not run into Nyquist, you can try removing the existing species and creating one with a larger pixel size. It's ok if it doesn't match the data – it will be automatically rescaled for refinement.

asarnow, 2024-06-16:
I'm using RTX A6000s (16GB). I am running into Nyquist at 2.5Å but there are duplicate particles that are inflating the estimate. I was re-running after removing duplicates but now I'm just using my original job form the Refine3D output. The box size ends up being 482, but I was able to refine these particles previously. I can't do it at the original pixel size of 0.83 Å, 1.245 is a 1.5x bin compromise.

asarnow, 2024-06-16:
It proceeds farther with 1 worker per device (it's still running), so this is another memory error indeed. Thanks Dimitry!

closed 2024-06-17 by alisterburt