warpem/warp #149: how to export particles after M refinement in linux version

linhua-tai, 2024-06-17:
Dear all,

I have a population containing two data sources, and I would like to export particles after M refinement to further classify. I've checked MCore, MTools and WarpTools, and couldn't find a easy one-button way to export particles after M refinement as what we used to have in Warp 1 windows GUI. I'm wondering if someone know how to do this?

Thank you very much!
Best,
Linhua

alisterburt, 2024-06-17:
Hey @linhua-tai great to see you here!

You're right, this tool doesn't exist and for now you will have to export particles from each species in WarpTools and combine star files

I was considering whether all particle export should go through WarpTools now but was totally forgetting about the multi-data-source capabilities... this should be added 😀

linhua-tai, 2024-06-17:
Hi Alister, thank you for your prompt reply! Always good to see you too!
In addition to your reply, since the particles.star file that M produces has columns naming as "_wrpXXXXX" and WarpTools doesn't recognize it, should I just change the corresponding columns to "_rlnXXXXX". For instance, "_wrpSourceName" to "_rlnMicrographName"? 
Thank you very much!

alisterburt, 2024-06-17:
that's a good point! Yes, you will have to change that too - if you have temporal poses in that STAR file they won't be taken into account unfortunately

sagarbiophysics, 2024-11-05:
Hi @alisterburt, 

I had a follow up on this! btw So far TOMOMAN-STOPGAP-Warptools is working great! 

So, if I get it right, it should technically extract refined particles from M, but my suspicion is that it doesn't. Am I missing something? Is there a way to force it to use a right hashed version of the data source? 

Also currently I am testing how to modify (whether to separate shifts in angstroms) M refined star to the one used for war-tool re-extraction. 

Cheers, 
Sagar

JuhaHuiskonen, 2024-12-08:
Hi, I have the same issue.  Any updates?  @sagarbiophysics, were you able to export particles after modifying the header of the STAR file?

alisterburt, 2024-12-09:
note: depending on how you did your refinements it's not a simple modification of the header

alisterburt, 2024-12-09:
@JuhaHuiskonen what was your export command? You likely specified the pixel size wrong - you can also look at the `_average.mrc` files produced for each tilt series when exporting, you should see a white blob at the center if your export was successful.

It is not a simple modification of the header if you refined particle poses over time as part of your M refinement.

The 2021 paper used the desktop software M which did have particle export - we haven't added this yet here, it is one of the last few things we still need for feature parity.

JuhaHuiskonen, 2024-12-09:
Hi, thanks @alisterburt. My export command was as follows:

WarpTools ts_export_particles \
--settings warp_tiltseries.settings \
--input_star m2/species/ribosome_6467b3ed/ribosome_particles_relion.star \
--output_processing relion_ribosome_3d_round2 \
--output_star relion_ribosome_3d_round2/particles.star \
--coords_angpix 2.405 \
--output_angpix 2.405 \
--box 300 \
--diameter 350 \
--relative_output_paths \
--3d

alisterburt, 2024-12-09:
From memory M particle coordinates are in angstroms rather than px so I assume this is the source of your error

JuhaHuiskonen, 2024-12-09:
Yes, the coordinates in the STAR file are in angstrom. That's why I give --coords_angpix 2.405. Could you specify what is the error here?

alisterburt, 2024-12-09:
if the coordinates in the file are in angstroms rather than pixels and you specify that the coordinates are at a pixel size of 2.405 they will be incorrectly scaled, you should specify `--coords_angpix 1`

JuhaHuiskonen, 2024-12-10:
Thanks - in the meantime, I divided the coordinates by the pixel size, and then further divided by the tomogram dimensions.  This worked, when using option --normalized_coords - but only after I set the same box size (250 pix) that the MCrore map uses.  A larger box size for RELION (300 pix) did not work.

alisterburt, 2024-12-10:
You've introduced a possibility for error - did you divide through by the tomogram dimensions or the lengths of each dimension? (Number of pixels - 1)

JuhaHuiskonen, 2024-12-10:
My tomogram size is 4096 x 4096 x 2048.  Pixel size is 2.405 A/pix.  This is what I did:

rlnCoordinateX = (wrpCoordinateX1 / 2.405) / 4096
rlnCoordinateY = (wrpCoordinateY1 / 2.405) / 4096
rlnCoordinateZ = (wrpCoordinateZ1 / 2.405) / 2048

JuhaHuiskonen, 2024-12-10:
Do you mean it should be:

rlnCoordinateX = (wrpCoordinateX1 / 2.405) / (4096 - 1)
rlnCoordinateY = (wrpCoordinateY1 / 2.405) / (4096 - 1)
rlnCoordinateZ = (wrpCoordinateZ1 / 2.405) / (2048 - 1)

alisterburt, 2024-12-10:
Please think about it carefully and come to the conclusion for yourself, I feel like I have been very clear.

dtegunov, 2024-12-10:
Yes, this should be correct. Could you please elaborate on 250px boxes working, but 300px boxes failing? Are you exporting using the 3D or 2D option?

JuhaHuiskonen, 2024-12-10:
Thank you for your patience @alisterburt! I posted the pseudocode to benefit others who might have the same issue and might find this thread.

@dtegunov I am using the 3D option.

dtegunov, 2024-12-11:
This might be an issue with insufficient GPU memory, especially when the defocus is high. There used to be an option for using CPU memory for some steps, but it looks like we lost it in WT. Regardless, I would recommend using 2D particles, especially with such large boxes – this will save Relion a lot of calculations.