warpem/warp #154: Possible bug: ts_stack not working

jmdobbs, 2024-06-21:
Hi,
I wanted to report a possible bug with ts_stack being nonfunctional for a couple of us (on both our optimized warp module and the conda latest version). In our hands, ts_stack doesn't work at all, deselecting all tilts as it fails to create the stack. Possibly we are doing something wrong, but see the following below for a minimal example

[dobbs@srv-mahamid-01: /struct/mahamid/jdobbs/warp2_testing/test2] $  WarpTools ts_stack --settings frameSettings.settings --angpix 20
Running command ts_stack with:
angpix = 20
mask = False
device_list = {  }
perdevice = 1
workers = {  }
settings = frameSettings.settings
input_data = {  }
input_data_recursive = False
input_processing = null
output_processing = null

No alternative input specified, will use input parameters from frameSettings.settings
File search will be relative to /struct/mahamid/jdobbs/warp2_testing/test2
18 files found
Parsing previous results for each item, if available...
18/18, previous metadata found for 18                                                                                                                       
Connecting to workers...
Connected to 2 workers
Failed to process /struct/mahamid/jdobbs/warp2_testing/test2/processing/gTS_003_00012_-15.0.tif, marked as unselected                                       
Failed to process /struct/mahamid/jdobbs/warp2_testing/test2/processing/gTS_003_00011_-12.0.tif, marked as unselected                                       
Failed to process /struct/mahamid/jdobbs/warp2_testing/test2/processing/gTS_003_00013_-18.0.tif, marked as unselected                                       
Failed to process /struct/mahamid/jdobbs/warp2_testing/test2/processing/gTS_003_00014_21.0.tif, marked as unselected                                        
Failed to process /struct/mahamid/jdobbs/warp2_testing/test2/processing/gTS_003_00015_24.0.tif, marked as unselected                                        
Failed to process /struct/mahamid/jdobbs/warp2_testing/test2/processing/gTS_003_00016_27.0.tif, marked as unselected                                        
Failed to process /struct/mahamid/jdobbs/warp2_testing/test2/processing/gTS_003_00017_-21.0.tif, marked as unselected                                       
Failed to process /struct/mahamid/jdobbs/warp2_testing/test2/processing/gTS_003_00018_-24.0.tif, marked as unselected                                       
Failed to process /struct/mahamid/jdobbs/warp2_testing/test2/processing/gTS_003_00001_-0.0.tif, marked as unselected                                        
Failed to process /struct/mahamid/jdobbs/warp2_testing/test2/processing/gTS_003_00002_3.0.tif, marked as unselected                                         
Failed to process /struct/mahamid/jdobbs/warp2_testing/test2/processing/gTS_003_00004_9.0.tif, marked as unselected                                         
Failed to process /struct/mahamid/jdobbs/warp2_testing/test2/processing/gTS_003_00003_6.0.tif, marked as unselected                                         
Failed to process /struct/mahamid/jdobbs/warp2_testing/test2/processing/gTS_003_00005_-3.0.tif, marked as unselected                                        
Failed to process /struct/mahamid/jdobbs/warp2_testing/test2/processing/gTS_003_00006_-6.0.tif, marked as unselected                                        
Failed to process /struct/mahamid/jdobbs/warp2_testing/test2/processing/gTS_003_00007_-9.0.tif, marked as unselected                                        
Failed to process /struct/mahamid/jdobbs/warp2_testing/test2/processing/gTS_003_00008_12.0.tif, marked as unselected                                        
Failed to process /struct/mahamid/jdobbs/warp2_testing/test2/processing/gTS_003_00009_15.0.tif, marked as unselected                                        
Failed to process /struct/mahamid/jdobbs/warp2_testing/test2/processing/gTS_003_00010_18.0.tif, marked as unselected                                        
18/18, 18 failed, 00:00 remaining                                                                                                                           
Finished processing in 00:00:12
Saying goodbye to all workers... Done

alisterburt, 2024-06-21:
Hi @jmdobbs - ts_stack should be run on tilt series settings rather than frame series settings, this should fix your problem 🙂

alisterburt, 2024-06-21:
Closing, please reopen if you feel this is still an issue!

closed 2024-06-21 by alisterburt

jmdobbs, 2024-06-21:
Thanks Alister, appreciate it! Maybe that could be documented somewhere, eg. in the ts_stack options. Currently it only asks for _a_ settings file, not that one in particular, and to my mind it makes equal sense to prepare the stack from the frame settings (since you're making a stack of the frames) as to prepare it from the ts settings

alisterburt, 2024-06-21:
Agreed it's not obvious, I don't love the two settings files pattern generally 

As a general rule, if it starts with ts it's a tilt series program 

Will have a think about where to document this