warpem/warp #156: MCore invalid loss value during reconstruction, but only when using multiple species

jmdobbs, 2024-06-22:
Hi,

I've been trying multispecies M refinement and invariably get a crash when trying to reconstruct and filter the particles. This occurs when I'm doing --iter 0 (running the check after species import), but also occurs during "real" refinements and occurs on several gpu types (+/- with cuda unified memory on, in case it is memory related). I've tried it on our optimized module (dev14 warp) and on the conda build (dev12, there was a slight issue with our dev14).
Error message:

System.Exception: The loss function has reached an invalid value because something went wrong during training.
   at Warp.NoiseNet3DTorch.TrainOnVolumes(NoiseNet3DTorch network, Image[] halves1, Image[] halves2, Image[] masks, Single angpix, Single lowpass, Single upsample, Boolean dontFlatten, Boolean performTraining, Int32 niterations, Single startFrom, Int32 batchsize, Int32 gpuprocess, Action`1 progressCallback)
   at Warp.Sociology.Species.CalculateResolutionAndFilter(Single fixedResolution, Action`1 progressCallback, Int32 gpuID)
   at Warp.Sociology.Species.FinishRefinement(Int32 gpuID)
   at WarpWorker.WarpWorker.EvaluateCommand(NamedSerializableObject Command)

What I've tried:
Original attempt: multispecies, 3 particle types, --iter 0: above error (command MCore --population m1/all.population --perdevice_refine 1 --iter 0)
Then:
1. removing 2/3 species from the population file, so that I'm only running the --iter 0 run with a single species: this works with no problems. I did this with each of my three species successfully
2. After doing the species individually with --iter 0, I tried multispecies with 2 particle types. It works (with --iter 0)
3. Multispecies --iter 0 with a 3rd particle type again: It doesn't work, and I get the same error as above
4. After the run with the 3rd species, running with 2 species (--iter 0) again doesn't seem to work anymore: same error
5. (ran the species individually with --iter 0, then) multispecies with 2 particles but actually doing M refinement (image warp and poses): same error
6. ran one species individually with image warp and poses: it worked

So there may be some issue during the gathering results/reconstruction/filtering step during multispecies refinement specifically. Alternatively, it could be something I've done incorrectly.

alisterburt, 2024-06-22:
thanks for the report and comprehensive test! Sorry things aren't working for you - I'll see if I can reproduce an error with multispecies refinement

dtegunov, 2024-06-25:
This should be fixed in be93c32.

The issue was specific to multi-species + multi-GPU scenarios. Denoiser training had GPU #0 hard-coded for training data generation, which didn't work well with everything else being on a different GPU.

jmdobbs, 2024-06-26:
Thanks for the fix. And for the new version of warp, it's a really big improvement!

closed 2024-06-27 by alisterburt