warpem/warp #161: ts_reconstruct failed

RamteenS, 2024-06-26:
Hi Alister,
I have issues running ts_reconstruct within Warp. Below is my command and output: 
WarpTools ts_reconstruct --settings warp_tiltseries.settings --angpix 10
Running command ts_reconstruct with:
angpix = 10
halfmap_frames = False
halfmap_tilts = False
deconv = False
deconv_strength = 1
deconv_falloff = 1
deconv_highpass = 300
keep_full_voxels = False
dont_invert = False
dont_normalize = False
dont_mask = False
dont_overwrite = False
subvolume_size = 64
subvolume_padding = 3
device_list = {  }
perdevice = 1
workers = {  }
settings = warp_tiltseries.settings
input_data = {  }
input_data_recursive = False
input_processing = null
output_processing = null

No alternative input specified, will use input parameters from warp_tiltseries.settings
File search will be relative to /data/ramteen/subtomo_test_tutorial/Warp_M2/tomostar
5 files found
Parsing previous results for each item, if available...
5/5, previous metadata found for 5                                                                                                                                                                               
Connecting to workers...
Connected to 4 workers
5/5, 00:00 remaining                                                                                                                                                                                             
Finished processing in 00:00:11
Saying goodbye to all workers...bash : ligne 1 : 2633985 Killed                  /home/ramteen/.conda/envs/warp/bin/WarpWorker -d 0 -p 0 --pipe WarpWorkerPipe_1_2633953_9240358 > worker0.out 2> worker0.err
bash : ligne 1 : 2634162 Killed                  /home/ramteen/.conda/envs/warp/bin/WarpWorker -d 1 -p 0 --pipe WarpWorkerPipe_2_2633953_534155781 > worker0.out 2> worker0.err
bash : ligne 1 : 2634318 Killed                  /home/ramteen/.conda/envs/warp/bin/WarpWorker -d 2 -p 0 --pipe WarpWorkerPipe_3_2633953_67218779 > worker0.out 2> worker0.err
 Done
bash : ligne 1 : 2634474 Killed                  /home/ramteen/.conda/envs/warp/bin/WarpWorker -d 3 -p 0 --pipe WarpWorkerPipe_4_2633953_1616009821 > worker0.out 2> worker0.err

When i look at reconstructed tomograms in warp_tiltseries/recontruction folder, i get wrong/failed reconstruction. I attach a picture below: 
[image]

The reconstruction is in Float32

I would appreciate your help for this issue

Many thanks again for all your help

Best 

Ramteen

alisterburt, 2024-06-26:
Hi @RamteenS - did tilt series alignment run successfully? 

There isn't much to go off here, ts_reconstruct ran successfully but something about your earlier processing is off

alisterburt, 2024-07-03:
closing as user has not come back, feel free to reopen if you hit the same issue

closed 2024-07-03 by alisterburt