warpem/warp #173: z-flip when reconstructing tomograms with imod meta data

kristyrochon, 2024-07-03:
Hello,

I'm processing a data set which was aligned in IMOD. I imported the meta data (.xf and .tlt) files. When I compare the warp reconstruction with the imod reconstruction, the z coordinates are inverted (1/250 in the warp reconstruction is 250/250 in our imod reconstruction). The xy coordinates are correct. Any advice would be helpful. We are planning to use coordinate files from a dynamo alignment from particles picked in the imod reconstruction.

Respectfully,
Kristy

alisterburt, 2024-07-03:
Hi @kristyrochon,

Thanks for opening the issue - there's a difference in tilt angle convention (i.e. what is meant by a positive or negative tilt angle) between IMOD and Warp, typically this is accounted for by flipping the angles automatically during `ts_import`, there is a flag for turning this behavior off

```
--dont_invert      Don't invert tilt angles compared to IMOD's convention (inversion is usually needed to match IMOD's geometric h
                   andedness). This will flip the geometric handedness
```

Typically I would recommend picking in tomograms reconstructed by the package you intend to use for downstream analysis. I understand this can be frustrating if you already have picks - just be aware that fixing all possible differences in the transformations can be nontrivial and repicking might be easier 🙂 

Going to close but please reopen if things don't work out!

kristyrochon, 2024-07-05:
Hello, 

I've tried a few things and the z slices remain inverted. I started a new project in case some settings weren't being overwritten and the resulting reconstruction (without the --dont_invert), the z slice direction have remained the same, inverted from the imod reconstruction. I checked my results from the windows installation of warp and the reconstructions were correct (compared to the imod reconstruction) with the same tomogram. 

I checked the tilt stack and the tilts are in the same position comparing the imod and warp results. 

Respectfully,
Kristy

alisterburt, 2024-07-05:
Hi Kirsty,

Setting/unsetting `--dont_invert` definitely changes the physical handedness of the tomogram by flipping the z-axis. I confirmed this recently when processing apoferritin test data so I think something has gone wrong in your test.

It's also important to specify exactly what you mean when you say 'the IMOD reconstruction': do you mean a specific IMOD reconstruction you already have or one from the IMOD metadata in the tiltstack directory? It's possible that the tlt angles are output differently on desktop warp compared to WarpTools - we only use IMOD's alignments and don't care about the handedness of its reconstruction so this is okay for us.

kristyrochon, 2024-07-05:
Got it. Let me try again with a fresh project. 

To clarify, I have a reconstruction from the windows warp installation using imod meta data, a reconstruction from imod which we used gold bead fiducial alignment to generate the meta data, and now I have a reconstruction from this CLI version of warp using the exact same imod meta data. In the first two, when navigating up and down in the z-direction of the volume, the z coordinates are the same. 

I will note that when I used the --dont_invert the CTF handedness was positive. When I removed it, it was negative and required a flip. All of my testing with the windows version required a flip. Not sure if that's helpful, but it lead me to believe removing the --dont_invert had changed the settings. 

Let me retry. 

Respectfully,
Kristy

alisterburt, 2024-07-05:
Oh! Then the reason for your issue is clear

Your tilt angles are being set to whatever is in the `.tlt` from IMOD when you import the alignments from there, undoing any changes made by the presence/absence of the `--dont_invert` flag. In desktop warp, the inversion was applied at the moment of importing alignment results - in WarpTools it is applied upon importing the mdocs.

I would recommend manually flipping the tilt angles in your IMOD metadata if you absolutely need the Warp tomogram to match the IMOD tomogram. Please rerun `ts_defocus_hand` after this to ensure correct defocus handedness.

Closing again, as before please reopen if necessary

closed 2024-07-05 by alisterburt