warpem/warp #176: ts_export_particles doesn't check for rlnTomoName

asarnow, 2024-07-08:
The tool throws an error if rlnMicrographName isn't in the input .star file. Might be nice to also check for/accept rlnTomoName. Not a huge deal to change of course but one does need to remember if using particle sources (in this case PickYOLO and my own scripts) other than WARP.

alisterburt, 2024-07-08:
right, the RELION data model changed... will handle this particular thing but no guarantees for compatibility with all RELION-isms 🙂

asarnow, 2024-07-08:
Thanks Alister. Just the ones that are likely to be in shared data model for software with Relion export (e.g. PYP) seems like reasonable fence to me. Dunno what else that would be besides rlnTomoName.

asarnow, 2024-07-08:
Oh, in a similar vein with `--3d` the rlnTomoSubTomosAre2DStacks is still 1.

alisterburt, 2024-07-08:
thanks @asarnow will take a look at that!

alisterburt, 2024-07-08:
@asarnow I just ran 

```sh
WarpTools ts_export_particles \
--settings warp_ts_5ts.settings \
--input_directory warp_ts_5ts/matching \
--output_processing 3dout \
--input_pattern "*emd_15854_flipx_clean.star" \
--normalized_coords \
--output_star relion/matching3d.star \
--output_angpix 4 \
--box 64 \
--diameter 130 \
--relative_output_paths \
--3d
```

and the output doesn't have any `rlnSubtomosAre2DStacks`

```

data_

loop_
_rlnCoordinateX #1
_rlnCoordinateY #2
_rlnCoordinateZ #3
_rlnAngleRot #4
_rlnAngleTilt #5
_rlnAnglePsi #6
_rlnMicrographName #7
_rlnMagnification #8
_rlnDetectorPixelSize #9
_rlnCtfMaxResolution #10
_rlnImageName #11
_rlnCtfImage #12
_rlnPixelSize #13
_rlnVoltage #14
_rlnSphericalAberration #15
   285.473    52.587   49.908          45  17.612438  209.99998  ApoF_ANT_p02_ts_009_unsorted.tomostar  10000.0  4.00000  5.6  ../3dout/subtomo/ApoF_ANT_p02_ts_009_unsorted/ApoF_ANT_p02_ts_009_unsorted_0000000_4.00A.mrc  ../3dout/subtomo/ApoF_ANT_p02_ts_009_unsorted/ApoF_ANT_p02_ts_009_unsorted_0000000_ctf_4.00A.mrc  4.00000  300.000  2.700
   257.928   127.712   59.889   62.999996  29.565557  217.49998  ApoF_ANT_p02_ts_009_unsorted.tomostar  10000.0  4.00000  5.6  ../3dout/subtomo/ApoF_ANT_p02_ts_009_unsorted/ApoF_ANT_p02_ts_009_unsorted_0000001_4.00A.mrc  ../3dout/subtomo/ApoF_ANT_p02_ts_009_unsorted/ApoF_ANT_p02_ts_009_unsorted_0000001_ctf_4.00A.mrc  4.00000  300.000  2.700
   
```

asarnow, 2024-07-10:
I confirm as well, my mistake.

alisterburt, 2024-07-10:
No problem, thanks for confirming! I have your tomoname changes in a branch, just confirming I didn't break anything before merging it in

PR #180 (2024-07-10): add support for particle export from RELION 5 style STAR files
no support for centered coords (yet?)

took the opportunity to tidy up the program a bit

closes #176
closed 2024-07-10 by alisterburt